pascallanger/DIY-Multiprotocol-TX-Module #403: Supporting OMP Hobby M2 Native Protocol

richardclli, 2020-07-27:
Just want to see if anybody interested in working out the native protocol for OMP Hobby M2 helicopter.  I do not have much information yet.  And not sure if anybody working on this protocol.

pascallanger, 2020-07-27:
I don't have access to a TX/RX so nothing will happen on my end.

richardclli, 2020-07-27:
Well, anyway, I just want to see if anybody interested.  Since, it seems the native protocol have telemetry of voltage feedback as well.

pascallanger, 2020-07-29:
I don't think anyone will jump here since they are using the forum instead.
If you have the original remote why don't you go ahead and open it up so we can see what RF chip is used and may be start to do some dumps?

richardclli, 2020-08-02:
I will try to work out some details on this protocol once I have got the radio.

richardclli, 2020-08-03:
Just got the radio hardware, the wireless chip is XN297LCU, seems this chip is not present in the 4in1 or 5in1 module.......

[image]

pascallanger, 2020-08-03:
It is supported by the 4in1.
Start doing dumps.

richardclli, 2020-08-08:
Here is my first dump.  The dump starts before switch on the radio, and ends
[OMP-SwitchOn.xlsx](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5044621/OMP-SwitchOn.xlsx)

richardclli, 2020-08-08:
This the the dump during binding: 
  1. Make the Heli in binding mode
  2. Start capture
  3. Switch on radio
  4. End capture when bind success.

[OMP-Bind.xlsx](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5044628/OMP-Bind.xlsx)

richardclli, 2020-08-08:
The radio has telemetry as well, only voltage feedback.

richardclli, 2020-08-08:
Capture for some stick movement.

[OMP-Sticks.xlsx](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5044652/OMP-Sticks.xlsx)

pascallanger, 2020-08-08:
The excel files you are sharing are not exploitable....
You need to share the logic files.

pascallanger, 2020-08-08:
I can't do anything with these files something is wrong... I need the logic analyzer files to see what you've done...

richardclli, 2020-08-10:
These files are decoded SPI data.  I used the setup as per XN297L datasheet, CS low activate, SCK rise edge, data MSB first.  I also added coloring to separate the communication steps by long idle timing.

I can attach a raw dump here, but not sure if you can read it.  I use DSView and it can save raw timing files that can store the waveform captured.  Will work this out when I have some spare time later.

[image]

pascallanger, 2020-08-10:
Sorry but these files are shitty. Please share the original logic files.

richardclli, 2020-08-10:
Alright, here you are, see if it helps.
[DSLogic-la-200810-190101.dsl.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5050437/DSLogic-la-200810-190101.dsl.zip)

richardclli, 2020-08-10:
Seems some commands are not recognized properly.  I have updated the nrf24l01 decoder for xn297 chip.  You may put this in the decoder folder to replace the old one.

[nrf24l01.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5053354/nrf24l01.zip)

pascallanger, 2020-08-11:
Using the dsview file I can fully decode the SPI:
 - Bitrate: 250Kbps
 - Scramble: enabled
 - Enhanced: enabled
 - auto ack: enabled
 - bind address: "FLPBD" not sure what it means...
 - normal address: 0x4E, 0x72, 0x8E, 0x70, 0x62 which is part of the bind packet
 - ...

pascallanger, 2020-08-11:
Bind:
- RF channel 0x23=35
- Address: 46 4C 50 42 44 = "FLPBD" BD=BIND?
- Period: 4410µs
- P(16) = 42 4E 44 4E 72 8E 70 62 0F 14 19 1E 23 28 2D 32
  - 42 4E 44 : unknown
  - 4E 72 8E 70 62 : normal packet address
  - 0F 14 19 1E 23 28 2D 32 : 8 RF channels used for normal packets
- RX
  - 12 times: P(16) = 46 4C 50 53 4B 0 0 0 0 0 0 0 0 0 0 1
    - nearly the bind address: 46 4C 50 53 4B = "FLPSK" SK=??
  -  4 times: P(16) = 42 4E 44 4E 72 8E 70 62 F 14 19 1E 23 28 2D 32
    - sends back what the TX is sending

pascallanger, 2020-08-11:
For the normal packets, I would need separate dsview file for:
 - each stick Aileron, Elevator, Throttle, Rudder with what has been done example: AIL_MID_LEFT_RIGHT
 - each feature button like flight modes, rates...

pascallanger, 2020-08-11:
Normal:
- RF hopping table: 0F 14 19 1E 23 28 2D 32, note this is (i+3)*5
- Period : 4995µs
- Address: 4E 72 8E 70 62
- P(16)= 00 08 40 00 00 00 00 00 A8 A1 FC 20 08 00 00 04
  - 00 : index of current hopping 0..7, |0x40 every 7 requesting telemetry from the RX
  - 08 : unknown switch to 28 after the first 10 packets
  - 40 00 00 00 00 00 : unknown and not moving in the single dump...
  - A8 A1 FC 20 : contains the ele/ail stick info but unknown since I don't know what has moved when...
  - 08 00 00 04 : unknown and not moving in the single dump...
Can't do anything more without the additional requested dumps...

richardclli, 2020-08-12:
Will try to do dumps of each channel when I am available.

> 
> 
> For the normal packets, I would need separate dsview file for:
> 
>     * each stick Aileron, Elevator, Throttle, Rudder with what has been done example: AIL_MID_LEFT_RIGHT
> 
>     * each feature button like flight modes, rates...

richardclli, 2020-08-13:
I have done your requested dumps, the first dump is the default.dsl with the following settings and I change one switch/stick to worked out another dump:

IDLE=0
MODE=Alt
D/R=70%
HOLD=Release
Thr=down
Rud=center
Ali=center
Ele=center

And I have worked out the telemetry feedback of 3 different battery voltages as well, please see if this is good to you for analysis.

[OMP-M2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5069523/OMP-M2.zip)

pascallanger, 2020-08-13:
Hummm... From the filename I understand that you have moved elev up took a dump, then elev dow another dump,... Is that it?
This is not really what I've asked. But let's see what I can figure out from this.

pascallanger, 2020-08-13:
As I was thinking these dumps are not usefull at all. I spent 1 hour looking at them for nothing...

richardclli, 2020-08-14:
I think I may figured out something meaningful:
- P(16) = 05084000000000002862FC2E08000004
  - P[7-11] = 002862FC2E
  - P[7] is THR, THR 0% => 0, THR 100% => FF
  - P[11] is RUD, RUD center => 2E, RUD left =>94, RUD right => 0
  - P[8-10] I guess these may control 3 servos independently, that is why it is not straight forward for AIL and ELE movement, but need more study to verify.

Can we start code something for testing?  So we can test out how the fields affect the heli movement.

richardclli, 2020-08-14:
Just figured out some more about RUD, maybe P[12] is related to RUD as well, need more test to verify.

pascallanger, 2020-08-14:
The channels are most likely on 11 or 12 bits that's why you can't see it clearly. There is nothing to code right now. When I will have good dumps to work on then I'll look at it again.

richardclli, 2020-08-14:
You are so amazing, I am finally be able to decode the 4 channels: thr, ail, ele, and rud because of your insight given. P[7-12] are encoded by 11-bit for each channel from P[7] to P[12] as LSB to MSB, trying to work out some more about the switches.  From various experiments, pitch is not send to receiver, only 4 channels with some switches.

richardclli, 2020-08-15:
P[2]P[1] => 16 bits B[16..0]
B[5] = 0 Release, 1 Hold
B[11,10] = mode: 0, 1 alt mode; 2 3d mode
B[13,6] = idle: 0, 1, 2 (0 no -ve pitch, 1, 2, +ve and -ve pitch)

richardclli, 2020-08-15:
Here are some dumps for you to verify:
1. ele down/ail left -> ail mid -> ail right -> ail mid -> ail left -> ele mid -> ele up -> ele mid -> ele down
2. rud left/thr down -> rud mid -> rud right -> rud mid -> rud left -> thr mid -> thr up -> thr mid -> thr down

[1-ail-ele.dsl.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5077927/1-ail-ele.dsl.zip)
[2-rud-thr.dsl.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5077928/2-rud-thr.dsl.zip)

richardclli, 2020-08-15:
For switches:
HOLD 0, Mode 0, Idle 0 -> HOLD 1 -> HOLD 0 -> HOLD 1 -> Mode 1 -> Mode 2 -> Mode 1 -> Mode 0 -> Idle 1 -> Idle 2 -> Idle 1 -> Idle 0

[switches.dsl.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5077938/switches.dsl.zip)

richardclli, 2020-08-15:
And for Voltage telemetry:
Voltage = 12.6 -> start dump -> gradulate reduce voltage step by step to 10V

[tele.dsl.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5077959/tele.dsl.zip)

richardclli, 2020-08-15:
For telemetry, the received packet R(16), R[3R[2] seems contains the battery voltage information.

pascallanger, 2020-08-15:
OMP protocol available in Multi v1.3.1.56.
- I haven't tested anything so please test and report but make sure to remove the heli blades...
- For now it's using your current ID, if it works comment out #define FORCE_OMP_ORIGINAL_ID and try again.
- I haven't forced the bind since I don't know if the heli remembers the radio ID or not... To test this power on the radio first, skip the bind and then power on the heli, if you have control it means that the ID/freqs are stored and therefore the current code is good.

richardclli, 2020-08-15:
Thank you!!  I will test it out and report to you if this is good or not.

richardclli, 2020-08-16:
I tried the followings and not able to bind:
1. Check out code and build bin
2. Flash the Multi 5-in-1
3. Switch on radio
4. Power up OMP M2

It seems not to bind, then I tried to tune the Freq, but still cannot get the bind.  Anything missing?

pascallanger, 2020-08-16:
I'll have a look but can you answer my previous question about the bind?

pascallanger, 2020-08-16:
Try v1.3.1.57, I've fixed something

richardclli, 2020-08-16:
- bind successfully with #define FORCE_OMP_ORIGINAL_ID
- remarked #define FORCE_OMP_ORIGINAL_ID can bind successfully with a bind procedure
So binding is good now.

After binding, even the stick is moving, the servos will jiggle.  Trying to fine tune the freq did not help.  However, the stick movements and switches seems functional.

richardclli, 2020-08-16:
I figure out the problem in your code.  Attached is a fixed version.  It looks usable, will try to test more and have a flight this weekend.

[OMP_nrf24l01.ino.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5081494/OMP_nrf24l01.ino.txt)

richardclli, 2020-08-16:
And I figured out the Ail and Rud are reversed, I revised the code again as follows:

[OMP_nrf24l01.ino.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5081504/OMP_nrf24l01.ino.txt)

richardclli, 2020-08-16:
No telemetry yet?  It has battery voltage feedback, I think we can add it in using HUB telemetry with only voltage feedback.

pascallanger, 2020-08-17:
Try v1.3.1.58.
Telemetry won't work with the current emulation layer. In fact, I'm not even sure it's possible... I would need access to the heli to see what can be done.

richardclli, 2020-08-17:
The modified code is good to go, it works.

For telemetry, I read the code of the emulation layer, so only the sending is implemented, but not receiving.  But by theory, is it possible to implement the receiving as well?  I can refine the code if I have some prototype to start with.

pascallanger, 2020-08-17:
The problem here is that they have used a special mode to send the data back. Instead of using the a traditionnal TX+RX approach, they are using an ack reply. It means that you not only need to figure out how to receive the data and decode it (XN297 is scrambling the data) but you also must be able to turn from TX to RX instantly since the answer over air is really back to back (at least if it's acting like the NRF which the XN297 is based on).

pascallanger, 2020-08-17:
FYI I've already written in the past the code to receive a NRF ack reply (which I haven't kept sadly) but it never worked due to the switching time from TX to RX of the NRF in normal mode (ack packet is really sent right at the end of the original packet). Maybe the CC2500 will do better since it is using a bitrate of 250K but that's a lot of work with low chance of success (not saying it's not doable). Is it worth it? And without having the hardware to play with makes it even more a challenge...

richardclli, 2020-08-18:
Tonight I used my Jumper T16 to connect to my OMP M2 and found that is is stable even using freq tuning 0.  I think Jumper T18 has hardware problem that cause a lot of interference that may cause problems.  If using NRF24L01 instead of CC2250, is telemetry possible?

pascallanger, 2020-08-25:
Ok let's call this protocol done for now. If I ever put my hands on this heli I'll look at the telemetry.

closed 2020-08-25 by pascallanger