pascallanger/DIY-Multiprotocol-TX-Module #405: Radiomaster TS16X & FrSky G-RX6 receiver

JCOFlyer, 2020-07-29:
I am fairly new at this Open TX stuff but I can follow along and have used some youtube videos to help. I cant get the Radiomaster to bind with Frsky G-RX6. So i tried updating the receiver to the latest only to find that may have "broken" it. Frsky says to go under "downloads to get the old firmaware. So i did but it still doesnt work. Im running Open TX 2.3.9. Multiprotocol firmware was updated to 1.3.1.36 AETR no inv. stable and flashed to internal. Receiver was updated using external port (from botom up; signal, power, ground) ACCST D16 2.1.2 dated 07-14-2020 (NOT ACCESS) but later reverted (I think) to whatever 181025 is under their "history firmware download".

kilrah, 2020-07-29:
From the bottom the module slot pinout is signal, ground, power. Needs a cable with swapped leads. If you put a straight cable you reversed polarity to the receiver.

JCOFlyer, 2020-07-30:
I spoke from memory. But yes you are correct. I had to swap the pens around and it is signal ground and power. Also the lights start flashing the reciever. It seems to be writing firmware to it. It finishes up in about a minute and a half or so. Either way it doesn’t seem to work with either firmware.

zendes, 2020-07-30:
Can you post the exact file name you downloaded from FrSky (.zip) and the
file you're flashing (.frk)? That should help identify which version &
flavor of ACCST you have on the receiver.

JCOFlyer, 2020-07-30:
The first one is G-RX6_ACCST_2.1.2_FCC.frk. The one I tried reverting it to is G-RX6_FCC_181025.frk. I finally found a friend who has a QX7. We were able to bind to it. I will add that although we did bind it I did not have any movements in the servos.  I'll give the RadioMaster another chance when I get home if I can.

zendes, 2020-07-30:
ok, so these are both FCC flavors, but different versions of ACCST. For G-RX6_FCC_181025.frk you need to bind with FrskyX, sub CH_16 or CH_8 and for G-RX6_ACCST_2.1.2_FCC.frk you need FrskyX2, same sub protocols

JCOFlyer, 2020-08-01:
All right so I tried the 1.8X but I noticed that there was no FrSkyX only FrSky and FrskyX2. I tried them out either way. I updated the firmware to 2.1.2 for the FrSkyX2 D16 telemetry on 1-8 still was unable to bind. I was really hopeful that this was it. Please help. Not sure what else to do.

pascallanger, 2020-08-01:
Binding a RX is really easy. I'm wondering what you are doing wrong...
Here are the steps:
1. Power on the RX while pushing the bind button, wait 1-2 sec then release the button -> the RX's LED should act differently while in bind mode
2. Power on the TX
3. Create a new model
4. Select Multi FrSkyX2 D16
5. Press the bind button on the same page
6. Wait for the RX LED pattern to change
7. Power off the RX
8. Power on the RX
9. Done

JCOFlyer, 2020-08-01:
YES!!!!!!! Thank you. So the telemetry light going off really threw me off. Feel stupid now. Spektrum to Frsky transition has been weird. Although I wouldn’t have guessed the protocols anyways. Very much appreciate all you guys and your help. I’ll be donating to the project for sure!!

pascallanger, 2020-08-02:
With multi the bind process is the same whatever protocol you use.

closed 2020-08-02 by pascallanger