pascallanger/DIY-Multiprotocol-TX-Module #406: Range issues with telemetry and critical rf

FXDUTRUCROSSET, 2020-07-30:
U have one internal module and two stm32 external ( one irange 4+ one ils jp4in1) and one irangelite
I have many signals if telelemy Lost / recovered and Aldo rf critical alerts in hitec optima and frsky d8 modes.
I used these receivers flawlessly with original hitec modules ans aurora for hitec side and sky9x or diy radios based on mega2560 in the past without any problem in the past ( except d8 if the right antenne braid us in the fuselage unstead of outside)
And it still works if i tru back to original transmitter instead of multi.
I tried other New receivers but it's still a problem on my planes. Does anybody had same issues
I tried to change jumper antenne or with other modules, rechecked fine tuning but nothing clears the issue.
On the bench as in the Sky thé alerts never stops.
i may have some glitches on corona dsssv2 but since there is no telemetry return i don't hear any critical alarm or rssi alerts

for information i have 
internal module multistmopentxnoinv v1.3.1.9 jumper
 t16pro  opentx t16 2.3.8-otx

i use mainly frsky d8, hitec opitma corona dssv2

zendes, 2020-07-30:
have you tried https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md ?

NCBob, 2020-08-02:
I am using a radio master TX16S and have loved it. 

I had bound a S8R receiver and it was working fine,no Rssi issues at all. 
Last night I was working on using some gv to adjust the amount of crow I’ve got applied, then it started complaining about low RSSI signal (usually around 10) and constantly losing frames and disconnecting. 

I tried my old transmitter and bound it just fine and didn’t have any problems with RSSI. 

I am using the frsky d16 option when I bind and using 1-8 with telemetry. 

I imagine it’s something that I’ve done but I’ve been pulling my hair out trying to figure out what’s happened. I also tried the tuning steps and it doesn’t seem to have any effect on the signal strength at all.

I’m running opentx 2.3.9

HazzaHFPV, 2020-08-02:
Maybe you were simply too close to the receiver during setup? I need to keep about 1 - 1.5m away from the receiver for a solid link.
Can't imagine that it would suddenly stop working :)

NCBob, 2020-08-02:
I’ll have to try that when I get home but when I used my old transmitter and bound it and it works fine  , I’ll still give it a try.

NCBob, 2020-08-03:
Ok, just got home and tried moving away from the receiver and same results.

I tried using several receivers and the same thing happens.

I tried my original transmitter (Horus X12) and it works fine, 98-100 RSSI.

I then tried another receiver (I was originally using a S8R) I tried an S6R and had the same issue.

Then I tried a V8 receiver and now everything appears to be fine. 

I had just gotten both these receivers last week, so I'm not sure if it's a bad batch of receivers or if it's the fact that it's the S8R/S6R modules that have issues with how they are sending back the RSSI.

If anyone has something I can try to test it I'm willing to give it a shot, but at least for now I can fly my plane again :)

NCBob, 2020-08-03:
After posting that I realized that the receivers are not defective because they worked with my Horus, so it must be something in how the SXR receivers are returning the RSSI information, although the radio link is flaky indicating that it's an issue with working with the SXR receivers.  The green led keeps flickering and the transmitter signals are not getting through to the control surfaces.

kilrah, 2020-08-03:
This sounds a lot like bad frequency tuning, did you do it?

NCBob, 2020-08-03:
> This sounds a lot like bad frequency tuning, did you do it?

Yes I did

NCBob, 2020-08-03:
Ok, I just tried two planes that have sxr receivers that I flew last week and the rssi on those are just fine. 

The only thing that is different is those had already been setup in their planes and I just bound them. These four receivers I’m having problems with I ran the sxr.lua script to calibrate them and configure them. When I get home tonight I’ll try reflashing one and see if that helps.

pascallanger, 2020-08-25:
This issue has been opened for a long time without any new comments for 22 days... Can it be closed?

FXDUTRUCROSSET, 2020-08-26:
i haven't yet identified the reason of the issue but if i'm alone to face this kind of issue with 3 different external modules and 1 internal module maybe the way i'm involving the system is the only reason so maybe better close the issue
will come back if i find the reason

closed 2020-08-26 by FXDUTRUCROSSET

edgechapterone, 2022-10-07:
I have exactly the same problem as NCBob. Was there any solution on this case?
Radiomaster Zorro internal Multi module and SR8. Low RSSI ca10.
Jumper T16 with XJT module and SR8. Works perfect.
Installed internal Multi module in my Jumper T16 and SR8. Low RSSI ca10
Other RX works perfect.
I have tried to bind with different frequence from -128 to 128 and beween. no result.
module and Zorro have both latest Multi FW.

pascallanger, 2022-10-08:
The S8R had a bug where it is misbehaving with some RX numbers. Try to change it.

edgechapterone, 2022-10-08:
Sim sala bim, now it works. Changed from 8 to 14.
Does it help to update SR8 ACCST firmware?

Frequence middle point is very offset (-70) but this may be another issue.

pascallanger, 2022-10-08:
Yes it means that your S8R is running an old version with this bug...

edgechapterone, 2022-10-08:
I updated firmware and now channel 8 works again.
Over andf out!