pascallanger/DIY-Multiprotocol-TX-Module #407: Realacc R11 Mini drone protocol (MX5323 chip)

unknownPT, 2020-07-31:
Hello,

following forum conversation, would like to reverse new protocol for MX5323 chip based transmitter, some data from XN297dump log below -

Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84Detected wrong address length, using 4 intead

Packet detected: bitrate=1M C=84 Enhanced pid=2 S=Y A= 99 06 00 00 P(13)= DC 80 80 32 80 20 20 20 20 58 04 00 00
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
RX on channel: 84, Time:     0us P: DC 80 80 FF 80 20 20 20 20 58 04 00 00
RX on channel: 84, Time:  2262us P: DC 80 80 FF 80 20 20 20 20 58 04 00 00
RX on channel: 84, Time:  2274us P: DC 80 80 FF 80 20 20 20 20 58 04 00 00
RX on channel: 84, Time:  2267us P: DC 80 80 FF 80 20 20 20 20 58 04 00 00
RX on channel: 84, Time:  2267us P: DC 80 80 FF 80 20 20 20 20 58 04 00 00
RX on channel: 84, Time:  2266us P: DC 80 80 FF 80 20 20 20 20 58 04 00 00
Trying RF channel:

1 RF channels identified: 84
--------------------------------
Identifying RF channels order.
Time between CH:84 and CH:0

Channel order:
84:     0us

--------------------------------
Identifying Sticks and features.

Packet detected: bitrate=1M C=84 Enhanced pid=3 S=Y A= 99 06 00 00 P(13)= DC 80 80 CA 80 20 20 20 20 84 04 00 00 -------------------------------- Identifying all RF channels in use. Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84 
RX on channel: 84, Time: 0us P: DC F6 F5 EF 0A 20 20 20 20 84 04 00 00 
RX on channel: 84, Time:  2268us P: DC F6 F5 EF 0A 20 20 20 20 84 04 00 00 
RX on channel: 84, Time:  2272us P: DC F6 F5 EF 0A 20 20 20 20 84 04 00 00 
RX on channel: 84, Time:  2269us P: DC F6 F5 EF 0A 20 20 20 20 84 04 00 00 
RX on channel: 84, Time:  2264us P: DC F6 F5 EF 0A 20 20 20 20 84 04 00 00 
RX on channel: 84, Time:  2266us P: DC F6 F5 EF 0A 20 20 20 20 84 04 00 00

pascallanger, 2020-07-31:
Can you start to do some dumps while only moving one thing at a time after the module has written: Identifying Sticks and features.
Write down what you have moved: Throttle down -> up -> down and then the corresponding text output. Do this for all sticks and features (rate, flip, light, RTH...)
You also need to capture the bind.

unknownPT, 2020-07-31:
After "Identifying Sticks and features" nothing captured, I have tried to move all sticks and press buttons - no results in terminal.

pascallanger, 2020-07-31:
That's strange and not expected... I'll have to look into it.
Can you try to capture a bind packet?

pascallanger, 2020-07-31:
Can you comment out this line and retry: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/37138f03ae70d61499f55d934d4271996555fdf1/Multiprotocol/XN297Dump_nrf24l01.ino#L607

unknownPT, 2020-08-03:
Done, but now I have no packets.

`15:55:23.988 -> XN297 dump, address length=5, bitrate=1M
15:55:23.988 -> Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84Detected wrong address length, using 5 intead
15:56:00.190 -> 
15:56:00.190 -> 
15:56:00.190 -> Packet detected: bitrate=1M C=84 Enhanced pid=1 ack S=Y A= 02 99 06 00 00 P(1)= F6
15:56:00.190 -> --------------------------------
15:56:00.190 -> Identifying all RF channels in use.
15:56:00.190 -> Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
15:56:38.239 -> 
15:56:38.239 -> 0 RF channels identified:
15:56:38.239 -> --------------------------------
15:56:38.239 -> Identifying RF channels order.
15:56:38.239 -> Time between CH:0 and CH:0
15:56:38.709 -> 
15:56:38.709 -> 
15:56:38.709 -> Channel order:
15:56:38.709 -> 0:     0us
15:56:38.709 -> 
15:56:38.709 -> --------------------------------
15:56:38.709 -> Identifying Sticks and features.
`

pascallanger, 2020-08-03:
This can't be... The line I've asked you to comment has an effect only after the Identifying Sticks and features. You have another issue.
May be the 2 TXs are too close to each others or the original TX is turning off or...

unknownPT, 2020-08-03:
I changed number of RF channels to 89 and now have output - 

16:07:57.288 -> XN297 dump, address length=5, bitrate=1M
16:07:57.288 -> Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84Detected wrong address length, using 4 intead
16:08:34.587 -> 
16:08:34.587 -> 
16:08:34.587 -> Packet detected: bitrate=1M C=84 Enhanced pid=0 S=Y A= 99 06 00 00 P(13)= DC 80 80 C6 80 20 20 20 20 F0 04 00 00
16:08:34.587 -> --------------------------------
16:08:34.587 -> Identifying all RF channels in use.
16:08:34.587 -> Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
16:09:13.181 -> RX on channel: 84, Time:     0us P: DC F6 7E 3E 0A 20 20 20 20 F0 04 00 00
16:09:13.181 -> RX on channel: 84, Time:  2259us P: DC F6 7E 3E 0A 20 20 20 20 F0 04 00 00
16:09:13.181 -> RX on channel: 84, Time:  2265us P: DC F6 7E 3E 0A 20 20 20 20 F0 04 00 00
16:09:13.181 -> RX on channel: 84, Time:  2273us P: DC F6 7E 3E 0A 20 20 20 20 F0 04 00 00
16:09:13.215 -> RX on channel: 84, Time:  2270us P: DC F6 7E 3E 0A 20 20 20 20 F0 04 00 00
16:09:13.249 -> RX on channel: 84, Time: 64843us P: DC F6 18 00 80 20 20 20 20 F0 04 00 00
16:09:13.249 -> Trying RF channel: ,85
16:09:13.249 -> RX on channel: 85, Time:     0us P: DC F6 18 00 80 20 20 20 20 F0 04 00 00
16:09:13.283 -> RX on channel: 85, Time:  2267us P: DC F6 18 00 80 20 20 20 20 F0 04 00 00
16:09:13.283 -> RX on channel: 85, Time:  2278us P: DC F6 18 00 80 20 20 20 20 F0 04 00 00
16:09:13.283 -> RX on channel: 85, Time:  2252us P: DC F6 18 00 80 20 20 20 20 F0 04 00 00
16:09:13.283 -> RX on channel: 85, Time:  2268us P: DC F6 18 00 80 20 20 20 20 F0 04 00 00
16:09:13.351 -> RX on channel: 85, Time: 61727us P: DC 80 08 00 8C 20 20 20 20 F0 04 00 00
16:09:13.351 -> Trying RF channel: ,86,87,88
16:09:14.292 -> RX on channel: 88, Time:     0us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.292 -> RX on channel: 88, Time:  2263us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.292 -> RX on channel: 88, Time:  2275us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.292 -> RX on channel: 88, Time:  2266us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.292 -> RX on channel: 88, Time:  2285us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> RX on channel: 88, Time:  2250us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> Trying RF channel: ,89
16:09:14.326 -> RX on channel: 89, Time:     0us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> RX on channel: 89, Time:  2019us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> RX on channel: 89, Time:  2274us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> RX on channel: 89, Time:  2261us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> RX on channel: 89, Time:  2267us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> RX on channel: 89, Time:  2276us P: DC 80 80 6F 80 20 20 20 20 F0 04 00 00
16:09:14.326 -> Trying RF channel: 
16:09:14.326 -> 
16:09:14.326 -> 4 RF channels identified: 84 85 88 89
16:09:14.326 -> --------------------------------
16:09:14.326 -> Identifying RF channels order.
16:09:14.326 -> Time between CH:84 and CH:85
16:09:14.361 -> Time:  2259us
16:09:14.361 -> Time:  2271us
16:09:14.361 -> Time:  2021us
16:09:14.361 -> Time between CH:84 and CH:88
16:09:14.461 -> Time: 14769us
16:09:14.528 -> Time: 16723us
16:09:14.596 -> Time: 13624us
16:09:14.663 -> Time between CH:84 and CH:89
16:09:14.663 -> Time: 13608us
16:09:14.730 -> Time: 15595us
16:09:14.797 -> Time: 13983us
16:09:14.865 -> 
16:09:14.865 -> 
16:09:14.865 -> Channel order:
16:09:14.865 -> 84:     0us
16:09:14.865 -> 85:  2021us
16:09:14.865 -> 89: 13608us
16:09:14.865 -> 88: 13624us
16:09:14.865 -> 
16:09:14.865 -> --------------------------------
16:09:14.865 -> Identifying Sticks and features.

TRH from 0 to 100%

16:12:48.412 -> P: DC 80 80 0A 80 20 20 20 20 F0 04 00 00
16:12:48.479 -> P: DC 80 80 19 80 20 20 20 20 F0 04 00 00
16:12:48.512 -> P: DC 80 80 1D 80 20 20 20 20 F0 04 00 00
[... 62 lines cut ...]
16:18:04.292 -> P: DC 80 F2 00 80 20 20 20 20 F0 04 00 00
16:18:04.360 -> P: DC 80 A5 00 80 20 20 20 20 F0 04 00 00
16:18:04.360 -> P: DC 80 95 00 80 20 20 20 20 F0 04 00 00
16:18:04.394 -> P: DC 80 A5 00 80 20 20 20 20 F0 04 00 00
16:18:04.394 -> P: DC 80 82 00 80 20 20 20 20 F0 04 00 00
16:18:04.428 -> P: DC 80 80 00 80 20 20 20 20 F0 04 00 00
16:18:04.733 -> P: DC 80 26 00 80 20 20 20 20 F0 04 00 00
16:18:04.800 -> P: DC 80 08 00 80 20 20 20 20 F0 04 00 00
16:18:05.001 -> P: DC 80 37 00 80 20 20 20 20 F0 04 00 00

AIL

16:20:15.914 -> P: DC 80 80 00 80 20 20 20 20 F0 04 00 00
16:20:18.742 -> P: DC 80 80 01 80 20 20 20 20 F0 04 00 00
16:20:18.742 -> P: DC 80 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.118 -> P: DC 08 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.251 -> P: DC 3D 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.285 -> P: DC 80 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.420 -> P: DC A9 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.420 -> P: DC C6 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.487 -> P: DC F5 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.487 -> P: DC F6 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.487 -> P: DC F5 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.487 -> P: DC F6 80 00 80 20 20 20 20 F0 04 00 00
16:20:20.854 -> P: DC E1 80 00 80 20 20 20 20 F0 04 00 00

unknownPT, 2020-08-03:
I set max channel value to 124 and some extra data - 

17:13:04.336 -> Packet detected: bitrate=1M C=84 Enhanced pid=0 S=Y A= 99 06 00 00 P(13)= DC 80 80 45 80 20 20 20 20 77 04 00 00
17:13:04.336 -> --------------------------------
17:13:04.336 -> Identifying all RF channels in use.
17:13:04.336 -> Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
17:13:43.183 -> RX on channel: 84, Time:     0us P: DC 80 80 9F 80 20 20 20 20 77 04 00 00
17:13:43.183 -> RX on channel: 84, Time:  2263us P: DC 80 80 9F 80 20 20 20 20 77 04 00 00
17:13:43.183 -> RX on channel: 84, Time:  2268us P: DC 80 80 9F 80 20 20 20 20 77 04 00 00
17:13:43.183 -> RX on channel: 84, Time:  2263us P: DC 80 80 9F 80 20 20 20 20 77 04 00 00
17:13:43.250 -> RX on channel: 84, Time: 59582us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.250 -> RX on channel: 84, Time:  2281us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.250 -> Trying RF channel: ,85
17:13:43.250 -> RX on channel: 85, Time:     0us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.250 -> RX on channel: 85, Time:  2003us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.250 -> RX on channel: 85, Time:  2275us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.250 -> RX on channel: 85, Time:  2269us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.318 -> RX on channel: 85, Time: 63134us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.318 -> RX on channel: 85, Time:  2261us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:43.318 -> Trying RF channel: ,86,87,88
17:13:44.260 -> RX on channel: 88, Time:     0us P: DC 80 80 37 80 20 20 20 20 77 04 00 00
17:13:44.260 -> RX on channel: 88, Time:  2266us P: DC 80 80 37 80 20 20 20 20 77 04 00 00
17:13:44.260 -> RX on channel: 88, Time:  2272us P: DC 80 80 37 80 20 20 20 20 77 04 00 00
17:13:44.293 -> RX on channel: 88, Time:  2277us P: DC 80 80 37 80 20 20 20 20 77 04 00 00
17:13:44.293 -> RX on channel: 88, Time:  2257us P: DC 80 80 37 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 88, Time: 66831us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> Trying RF channel: ,89
17:13:44.361 -> RX on channel: 89, Time:     0us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 89, Time:  2262us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 89, Time:  2263us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 89, Time:  2272us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 89, Time:  2276us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 89, Time:  2013us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> Trying RF channel: ,90
17:13:44.361 -> RX on channel: 90, Time:     0us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 90, Time:  2272us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 90, Time:  2273us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 90, Time:  2270us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 90, Time:  2262us P: DC 80 80 C7 80 20 20 20 20 77 04 00 00
17:13:44.361 -> RX on channel: 90, Time:  2265us P: DC 80 80 F3 80 20 20 20 20 77 04 00 00
17:13:44.361 -> Trying RF channel: ,91,92,93,94,95,96,97
17:13:47.148 -> RX on channel: 97, Time:     0us P: DC 80 80 FA 80 20 20 20 20 77 04 00 00
17:13:47.148 -> RX on channel: 97, Time:  2013us P: DC 80 80 FA 80 20 20 20 20 77 04 00 00
17:13:47.148 -> RX on channel: 97, Time:  2271us P: DC 80 80 FA 80 20 20 20 20 77 04 00 00
17:13:47.148 -> RX on channel: 97, Time:  2275us P: DC 80 80 FA 80 20 20 20 20 77 04 00 00
17:13:47.182 -> RX on channel: 97, Time:  2256us P: DC 80 80 FA 80 20 20 20 20 77 04 00 00
17:13:47.182 -> RX on channel: 97, Time:  2275us P: DC 80 80 FA 80 20 20 20 20 77 04 00 00
17:13:47.182 -> Trying RF channel: ,98
17:13:47.182 -> RX on channel: 98, Time:     0us P: DC 80 80 FA 80 20 20 20 20 77 04 00 00
17:13:47.216 -> RX on channel: 98, Time: 60246us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:47.250 -> RX on channel: 98, Time:  2262us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:47.250 -> RX on channel: 98, Time:  2264us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:47.250 -> RX on channel: 98, Time:  2274us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:47.250 -> RX on channel: 98, Time:  2274us P: DC 80 80 FF 80 20 20 20 20 77 04 00 00
17:13:47.250 -> Trying RF channel: ,99,100,101,102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124
17:13:59.270 -> 
17:13:59.270 -> 7 RF channels identified: 84 85 88 89 90 97 98
17:13:59.270 -> --------------------------------
17:13:59.270 -> Identifying RF channels order.
17:13:59.270 -> Time between CH:84 and CH:85
17:13:59.270 -> Time:  2253us
17:13:59.304 -> Time:  2271us
17:13:59.304 -> Time:  2257us
17:13:59.371 -> Time between CH:84 and CH:88
17:13:59.371 -> Time: 13620us
17:13:59.439 -> Time: 15566us
17:13:59.506 -> Time: 14612us
17:13:59.573 -> Time between CH:84 and CH:89
17:13:59.607 -> Time: 11336us
17:13:59.674 -> Time: 13609us
17:13:59.741 -> Time: 15629us
17:13:59.808 -> Time between CH:84 and CH:90
17:13:59.808 -> Time: 26910us
17:13:59.909 -> Time: 26583us
17:13:59.976 -> Time: 29245us
[... cut ...] H:84 and CH:97
17:14:00.078 -> Time: 56228us
17:14:00.145 -> Time: 58036us
17:14:00.212 -> Time: 59388us
17:14:00.245 -> Time between CH:84 and CH:98
17:14:00.279 -> Time: 56212us
17:14:00.347 -> Time: 57368us
17:14:00.447 -> Time: 59910us
17:14:00.447 -> 
17:14:00.447 -> 
17:14:00.447 -> Channel order:
17:14:00.447 -> 84:     0us
17:14:00.447 -> 85:  2253us
17:14:00.447 -> 89: 11336us
17:14:00.447 -> 88: 13620us
17:14:00.447 -> 90: 26583us
17:14:00.447 -> 98: 56212us
17:14:00.447 -> 97: 56228us

unknownPT, 2020-08-03:
Packets from [transmitter](https://i.imgur.com/WvHgCEM.jpg) buttons: 

Headless mode - 
P: DC 80 80 00 80 20 20 20 20 A2 24 00 00
P: DC 80 80 00 80 20 20 20 20 A2 04 00 00

One-touch calibration -
P: DC 80 80 00 80 20 20 20 20 A2 04 00 00
P: DC 80 80 00 80 20 20 20 20 A2 04 01 00

One-click return - 
P: DC 80 80 00 80 20 20 20 20 A2 04 00 00
P: DC 80 80 00 80 20 20 20 20 A2 04 20 00

Lighting control -
P: DC 80 80 00 80 20 20 20 20 A2 04 00 00
P: DC 80 80 00 80 20 20 20 20 A2 04 00 80

Gear shift (sensitivity control) - 
mode1 (default) 
P: DC 80 80 00 80 20 20 20 20 A2 04 00 00
mode2
P: DC 80 80 00 80 20 20 20 20 A2 05 00 00
mode3
P: DC 80 80 00 80 20 20 20 20 A2 06 00 00

Buttons without label (3pcs) -
P: DC 80 80 00 80 20 20 20 20 A2 04 00 00
P: DC 80 80 00 80 20 20 20 20 A2 04 80 00

Left and right fine-tuning - 
P: DC 80 80 00 80 20 20 20 20 C5 04 00 00
P: DC 80 80 00 80 5F 20 20 20 C5 04 00 00
P: DC 80 80 00 80 E0 20 20 20 C5 04 00 00
P: DC 80 80 00 80 61 20 20 20 C5 04 00 00
P: DC 80 80 00 80 20 20 20 20 C5 04 00 00

Fine tuning before and after -
P: DC 80 80 00 80 20 20 20 20 C5 04 00 00
P: DC 80 80 00 80 20 A1 20 20 C5 04 00 00
P: DC 80 80 00 80 20 E0 20 20 C5 04 00 00
P: DC 80 80 00 80 20 9F 20 20 C5 04 00 00
P: DC 80 80 00 80 20 20 20 20 C5 04 00 00

How to catch bind packet?

pascallanger, 2020-08-03:
On a lot of the dumps the byte 10 is different (C5 on the last line above but before A2, 77, F0, 84, 58). It looks to be different every time you power cycle the TX can you confirm? It could be related to the frequencies being used...
Can you try to put some distance between the 2 TXs and redo the frequencies? usually it finds well the frequencies but here you seem to have ghost. I hop adding distance even a wall or two in between will attenuate them.
For the bind I don't know it depends of your model. Usually on these small models binds end when you apply full throttle then cut it. But it should be written in the manual. I haven't found it online to see how they do it...

unknownPT, 2020-08-03:
Yes, byte 10 changed every time after power cycle and bind. Bind on this model works exactly as you explained - throttle 0-100-0%, but I see no bind packet at this moment, just changed byte 10.

Its not clear about frequencies and distance, I have only one TX. Need to add some distance or/and wall between TX and NRF and see results, right?

pascallanger, 2020-08-03:
> Bind on this model works exactly as you explained - throttle 0-100-0%, but I see no bind packet at this moment, just changed byte 10.

That's strange... So if you set throttle to 0%, power on the TX then launch a dump, you can't see any packets at all?

> Its not clear about frequencies and distance, I have only one TX. Need to add some distance or/and wall between TX and NRF and see results, right?

Yes

unknownPT, 2020-08-04:
> 

> That's strange... So if you set throttle to 0%, power on the TX then launch a dump, you can't see any packets at all?

Yes, got first packets only after the bind.

unknownPT, 2020-08-10:
Found something interesting, TX itself not fully functional if no bind (buttons has no sounds, etc). But anyway TX send some packets out, see below. 

XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79Detected wrong address length, using 4 intead

Packet detected: bitrate=1M C=79 Enhanced pid=1 ack S=Y A= 4D 41 49 4E P(10)= B1 99 06 00 00 55 59 5A 5A 62
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79
RX on channel: 79, Time:     0us P: B1 99 06 00 00 55 59 5A 5A 62
Trying RF channel: ,80
RX on channel: 80, Time:     0us P: B1 99 06 00 00 55 59 5A 5A 62
RX on channel: 80, Time:  2447us P: B1 99 06 00 00 55 59 5A 5A 62
RX on channel: 80, Time:  3962us P: B1 99 06 00 00 55 59 5A 5A 62
RX on channel: 80, Time:  2985us P: B1 99 06 00 00 55 59 5A 5A 62
RX on channel: 80, Time:  2977us P: B1 99 06 00 00 55 59 5A 5A 62
RX on channel: 80, Time:  2971us P: B1 99 06 00 00 55 59 5A 5A 62
Trying RF channel: ,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124

2 RF channels identified: 79 80
--------------------------------
Identifying RF channels order.
Time between CH:79 and CH:80

Channel order:
79:     0us

--------------------------------
Identifying Sticks and features.
P: B1 99 06 00 00 55 59 5A 5A 62

pascallanger, 2020-08-10:
Yes it's what was missing:
Bind address: 4D 41 49 4E = 'MAIN'
Bind channel: 80 (most likely from above dump)
Paylod: B1 99 06 00 00 55 59 5A 5A 62
 - B1 indicates bind packet (DC = normal packet)
 - 99 06 00 00 = ID = address of normal packets
 - 55 59 5A 5A 62 = 85, 89, 90, 90, 98 = RF channels to be used (kind of match previous dumps)

unknownPT, 2020-08-10:
Sounds good! But I dont do bind (THR 0-100-0) during this scan, just moving sticks a little bit.

pascallanger, 2020-08-10:
Normal packets:
- Bitrate=1Mbps
- Type = Enhanced
- Scramble = Yes
- Address = 99 06 00 00
- P(13)= DC 80 80 32 80 20 20 20 20 58 04 00 00
  - DC = normal packet
  - 80 80 32 80 : AETR 00..80..FF
  - 20 20 20 20 : Trims
  - 58 : changing every time the TX restart
  - 04 : |0x20=headless, |0x01=rate2, |0x02=rate3
  - 00 : |0x01=calib, |0x20=return, |0x80=unknown
  - 00 : |0x80=light, |0x01=flip

pascallanger, 2020-08-10:
@unknownPT Please test Multi version 1.3.1.53 protocol Realacc/R11 and report.
If it works, then try with #define FORCE_REALACC_ORIGINAL_ID commented in REALACC_nrf24l01.ino.

unknownPT, 2020-08-11:
Great, will test it soon, thanks!

pascallanger, 2020-08-25:
Any news? You said you will test it soon 14 days ago...

unknownPT, 2020-08-31:
I have hardware issue with my setup now. Have ordered some new parts needed. Sorry for delay.

unknownPT, 2020-09-19:
Hi,

Have created temporary setup with Arduino ProMini and NRF for tests, flashed with 1.3.1.59.
But can't bind (and nothing transmitted to the air, I have checked by XN297 dump running on STM+NRF same time on another PC), tried both options #define FORCE_REALACC_ORIGINAL_ID enabled and commented.
I have no rotary switch and fill all 14 points in Bank 1 with   /*	1	*/	{PROTO_REALACC,	NONE		,	0	,	P_HIGH	,	AUTOBIND	,	0	,	0x00000000 }, in Config,h.

Have no idea how to troubleshoot it...

pascallanger, 2021-01-19:
Closing this thread, too bad that I've spent the time and nobody has ever tested this protocol...