pascallanger/DIY-Multiprotocol-TX-Module #408: DSM Forward Programming

BeanieBots, 2020-07-31:
When Spektrum brought out 'Forward Programming' it was a convienient way of setting up a Rx from your Tx.
Unfortunatley, with their latest Rx (the AR637T), it is now the ONLY way to configure the Rx.
This means it is not possible to fully utilise all the features of such a Rx using a 4-in-1 and OPEN Tx.
Would it be possible that the protocol used for 'Forward Programming' could be implemented?

pascallanger, 2020-07-31:
Wow that's a 100€ RX... Do you know of a RX using this 'Forward Programming' that could be cheaper?

BeanieBots, 2020-07-31:
Only the AS3000 Flight Stabilization Module which I think is about the same price.
I'll try to find something. Maybe a satelite Rx.

BeanieBots, 2020-07-31:
I've spent a couple of hours browsing with dissapointing results. Plenty that are even more expensive!
I did find the SPM4651T which when plugged into a FC6250HX flybarless flight controller claims the flight controller can be configured using 'Forward Programming'. Not sure if the just the Rx without the controller would be much use. (nothing to configure!).
I'm guessing that it is done over SRXL2. So, perhaps any Rx that supports SRXL2?
Sorry I couldn't help any more but an obvious reason for taking the OPEN Tx / 4-in-1 route is to avoid these high prices.
Having said that, there is a lot inside the AR637T, including 6-axis gyro, AS3X stabilisation, barometer and full range telemetry.
Such a shame they locked it out from anything not Spektrum.

pascallanger, 2020-07-31:
Thanks for looking. I'll see what I can do and let you know.

BeanieBots, 2020-07-31:
Thank you for considering this issue. I'll wait to see if you can get anywhere with it.

pascallanger, 2020-08-23:
I've finally got hold of a TX and RX.
I'm not far into it but I know how the communication is done between the TX and RX which is the first step.
On the TX side they use a virtual channel 14 to transmit the commands.
The RX responds using telemetry (of course) and the ID 0x09.
For example:
TX sends this command:
0x70 | 0x11 | 0x71 | 0x06 | 0x72 | 0x00 | 0x73 | 0x14 | 0x74 | 0x00 | 0x75 | 0x00
 -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
No idea what it means yet...

RX reply:
0x09 | 0x01 | 0x00 | 0x15 | 0x02 | 0x22 | 0x01 | 0x00 | 0x14 | 0x00 | 0x00 | 0x00 | 0x00 | 0x00 | 0x00 | 0x00
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
RX version: 0x02 0x22 0x01 -> 2.34.01
0x15 or 0x14 might be the RX type...

pascallanger, 2020-08-25:
I've made some progress. I think at this stage and despite all the unknowns I could get the menu navigation working (page change). I'm starting to see how the values are being managed but fram from having something usefull.

BeanieBots, 2020-08-27:
That's great news and excellent work.
If there's anything I can do to help (with my limited knowledge and resources) then please let me know.

pascallanger, 2020-09-01:
Video of the current progress: https://photos.google.com/share/AF1QipPLJS6sBmVoEQcbom6krMbLFjrCV_8HZYfilmk-SuwarWG8Aov0H8YF26ArhYb0BA/photo/AF1QipNGD-tldSHYwsv4YJ78RcUqERllNsD_jSPUHAKC?key=cWlTTXE1aG9DWjlzdHR4cHFHMWRYWV9BRG1URUR3

BeanieBots, 2020-09-01:
That's really great progress.

BeanieBots, 2020-09-01:
OOps, closed by mistake.

pascallanger, 2020-09-09:
It's not fully working yet but I've pushed my changes on GitHub so you can test it:
 - Need the latest OpenTX nightly
 - Need to compile by yourself Multi v1.3.1.64
 - Put the LUA script https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Lua_scripts/DSM%20FwdPrg.lua in the radio SD card under \SCRIPTS\TOOLS
On the AR637T that I have, everything under the menu "Other Settings" is working at the exception of putting the RX in bind mode. Changing the failsafe values for example work.
If you navigate and you see Unknown_XXX, please report the XXX and what's the exact text should be when you do the same on a genuine radio.

BeanieBots, 2020-09-09:
Great, thanks.
I'll have a go tonight and let you know the outcome.

pascallanger, 2020-09-18:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts#dsm-forward-programming
[image]

BeanieBots, 2020-09-18:
Really sorry I have not done any testing. I won't bore you with my issues except to say it was due to health and a change of medication that made things worse. I hope to be in a possition to download/compile & test later this weekend. 
Thanks for all your efforts. It's looking really great.

wbmitchell138, 2020-10-07:
I just flashed 2.3.10 OTx stable to my Radiomaster TX16S and have 1.3.1.69 Multi Module firmware.  I can open the DSM forward programming Lua and connect to a Spektrum HX6250FC flybarless contoller with an SPM4651T Rx, but I get an "_RX unknown_18 v5.3.0_" message.  I know that this is a work in progress but thought I would share what I am seeing on my end.  I'm really excited to have the forward programming feature functioning.

[image]

pascallanger, 2020-10-07:
@wbmitchell138 Do you have access to a spektrum radio? If yes, everywhere you see Unknown_XXX, please write down the XXX and the **exact** text which appears on the spektrum radio when you do the same. You could also post a picture of both screens side by side.
From there I can add the missing texts in the script and you will see them after you update it.
The menu should work, meaning that you have a lot of "Unknown" but if it's a menu entry, you should be able to get in and probably see other unknowns. Therefore just repeat the previous and write them down or a picture.

pascallanger, 2020-10-07:
If you look at this sequence on youtube: https://youtu.be/VJ0m2S42QPI?t=135
You'll see that the menu is similar (line positions including spacing):
[image]
So most likely:
Unknwon_D3 = Swashplate
Unknown_DD = Tail rotor
Unknown_DA = SAFE
Unknown_DE = Setup
Unknown_86 = System Setup
Unknown_18 = FC6250HX
Updating the code with this would allow you to see the first page. The other pages have to be built the same way. This is why having an original tx at hand is easier.

If you want to get going yourself at adding the text, open the luaa script at this line:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/baf9a0f978f184e36f34a83ca516447836dbb597/Lua_scripts/DSM%20FwdPrg.lua#L459
Then add the lines
Text[0xXXXX]="YYYYY" where XXXX is what you have behind Unknown_XXXX and YYYY the text to be displayed.

wbmitchell138, 2020-10-09:
I'll see if i can get some pictures of the various menus for the 6250 FBL.   Also will try to put together a flow chart of the menus, submenus and setting and values for each.

FOTO-PETE, 2020-11-15:
Hi Pascal,
I have an AR637T Receiver (V2.37.7) and an iX12 (V1.13.08), and I have added some texts that showed as "Unknown" before.

Here comes the modified section "Text to be displayed" of the lua file with my addidions/changes:

--Text to be displayed -> need to use a file instead?
  --RX names--
  Text[0x0014]="SPM4651T"
  Text[0x0015]="AR637T"
  --Lists--
  Text[0x0036]="Throttle"
  Text[0x0037]="Aileron"
  Text[0x0038]="Elevator"
  Text[0x0039]="Rudder"
  Text[0x003A]="Gear"
  -- for i=1,7 do -- 3B..41
  --   Text[0x003A+i]="Aux"..i
  -- end
  -- for i=1,8 do -- 41..49
  --   Text[0x0041+i]="XPlus-"..i
  -- end
  Text[0x0040]="Roll"
  Text[0x0041]="Pitch"
  Text[0x0042]="Yaw"
  Text[0x0046]="Priority"
  Text[0x004A]="Failsafe"
  Text[0x004B]="Main Menu"
  Text[0x004E]="Position"
  Text[0x0050]="Outputs"
  Text[0x0051]="Output Channel 1"
  Text[0x0052]="Output Channel 2"
  Text[0x0053]="Output Channel 3"
  Text[0x0054]="Output Channel 4"
  Text[0x0055]="Output Channel 5"
  Text[0x0056]="Output Channel 6"
  --Text[0x005E]="Inhibit"
  Text[0x005F]="Hold Last"
  Text[0x0060]="Preset"
  --Text[0x0061]="Custom"
  --Messages--
  Text[0x0078]="FM Channel"
  Text[0x0080]="Orientation"
  Text[0x0085]="Frame Rate"
  Text[0x0086]="System Setup"
  Text[0x0087]="F-Mode Setup"
  Text[0x0088]="Enabled F-Modes"
  Text[0x008A]="Gain Sensitivity"
  Text[0x0090]="Apply"
  Text[0x0093]="Complete"
  Text[0x0094]="Done"
  Text[0x0097]="Factory Reset"
  Text[0x009A]="Capture Failsafe Positions"
  Text[0x009C]="Custom Failsafe"
  Text[0x00A5]="First Time Setup"
  Text[0x00AD]="Gain Channel Select"
  Text[0x00B6]="FM1"
  Text[0x00B7]="FM2"
  Text[0x00B8]="FM3"
  Text[0x00B9]="FM4"
  Text[0x00BA]="FM5"
  Text[0x00BB]="FM6"
  Text[0x00BC]="FM7"
  Text[0x00BD]="FM8"
  Text[0x00BE]="FM9"
  Text[0x00BF]="FM10"
  Text[0x00F9]="Gyro settings"
  Text[0x00FE]="Stick Priority"
  Text[0x0100]="Make sure the model has been"
  Text[0x0101]="configured, including wing type,"
  Text[0x0102]="reversing, travel, trimmed, etc."
  Text[0x0103]="before continuing setup."
  Text[0x0104]="0104"
  Text[0x0105]="0105"
  Text[0x0106]="Any wing type, channel assignment,"
  Text[0x0107]="subtrim, or servo reversing changes"
  Text[0x0108]="require running through initial"
  Text[0x0109]="setup again."
  Text[0x010A]="010A"
  Text[0x010B]="010B"
  Text[0x0190]="Relearn Servo Settings"
  Text[0x019C]="Enter Receiver Bind Mode"
  Text[0x01DC]="AS3X"
  Text[0x01DD]="AS3X Settings"
  Text[0x01DE]="AS3X Gains"
  Text[0x01E0]="Rate Gains"
  Text[0x020A]="Restore from Backup"
  Text[0x0209]="Save to Backup"
  Text[0x020D]="First Time SAFE Setup"
  Text[0x021A]="Set the model level,"
  Text[0x021B]="and press  Continue."
  Text[0x021C]="021C"
  Text[0x021D]="021D"
  
  Text[0x021F]="Set the model on its nose,"
  Text[0x0220]="and press Continue. If the"
  Text[0x0221]="orientation on the next"
  Text[0x0222]="screen is wrong go back"
  Text[0x0223]="and try again."
  Text[0x0224]="Continue"
  Text[0x0229]="Set Orientation Manually"
  
  Text[0x0227]="Other settings"
  Text[0x022B]="WARNING!"
  Text[0x022C]="This will reset the"
  Text[0x022D]="configuration to factory"
  Text[0x022E]="defaults. This does not"
  Text[0x022F]="affect the backup config."
  Text[0x0230]="0230"
  Text[0x0231]="This will overwrite the"
  Text[0x0232]="backup memory with your"
  Text[0x0233]="current configuartion."
  Text[0x0234]="0234"
  Text[0x0235]="0235"
  Text[0x0236]="This will overwrite the"
  Text[0x0237]="current config with"
  Text[0x0238]="that which is in"
  Text[0x0239]="the backup memory."
  Text[0x023A]="023A"
  Text[0x023D]="Copy Flight Mode Settings"
  Text[0x0240]="Utilities"
  Text[0x8001]="Flight Mode 1"
  Text[0x8002]="Flight Mode 2"
  Text[0x8003]="Flight Mode 3" 

--
Kind regards
Peter

pascallanger, 2020-11-15:
@FOTO-PETE Thanks for the unknowns.
Does it mean the script works well for you?

FOTO-PETE, 2020-11-15:
The unkown text ist solved now, but unfortunately some menues doesn't show correctly (for example fields with parameters are not at the correct position).

wbmathis, 2021-01-27:
TX16S on OpenTX 2.3.11 with Multi-Module on 1.3.2.12. AR8360T receiver bound as dsm x 2f 22ms. DSM Forward Programing script seems to work with regards to other settings, but gyro settings shows only blank page with nonfunctional back and next buttons. RX Unknown_1A 2.27.4 at bottom of screen. Gyro needed for setting up AS3X and SAFE. OpenTX 2.3.10 stated support for basic forward programming. Is it talking about your LUA script or is that functionality somewhere else in latest OpenTX? Thanks for your efforts!

prognosis2001, 2021-03-02:
I am also experiencing the same with the AR630. Unknown RX V.... Same blank screen on gyro setup and also unknown_85... Would like to see what I can do to contribute or confirm with others the same issues..

jbolene, 2021-04-27:
any updates on this?  I just got an ar637t and no way to program it without buying a NX6, 8 or 10

Schaezard, 2021-04-28:
I have the same issue now where I have two spektrum receivers (the AR637T and the AR630) where I can use neither AS3X nor safe select. I really wanna see this moving forward and I'd love to try my best to help. That being said I don't own a spectrum TX to demonstrate the menu states

jbolene, 2021-04-28:
can't wait on this to happen, NX8 is back in stock today and have ordered one
let me know how I can help fix this to make it work, I can always sell the NX8 if this gets to working

gbine, 2021-05-14:
Here is an update to the LUA script.  I have added some place holders for newer recievers.  Working on getting formatting to look better and First Time Setup to function (at least on my AR637TA)
If anyone finds some unknow texts please take a screenshot and send it to me.

--Text to be displayed -> need to use a file instead?
  --RX names--
  Text[0x0014]="SPM4651T"
  Text[0x0015]="AR637T"
  Text[0x0016]="AR637TA"
  Text[0x0017]="0x0017"
  Text[0x0018]="0x0018"
  Text[0x0019]="0x0019"
  Text[0x001A]="AR8360A"
  Text[0x001B]="0x001B"
  Text[0x001C]="0x001C"
  Text[0x001D]="0x001D"
  Text[0x001E]="0x001E"
  Text[0x001F]="0x001F"
  --Lists--
  Text[0x0036]="Throttle"
  Text[0x0037]="Aileron"
  Text[0x0038]="Elevator"
  Text[0x0039]="Rudder"
  Text[0x003A]="Gear"

  --******
  --This part is strange since the AR637T needs
  for i=1,7 do -- 3B..41
    Text[0x003A+i]="Aux"..i
  end
  for i=1,8 do -- 41..49
    Text[0x0041+i]="XPlus-"..i
  end
  --But FOTO-PETE reports that it should be:
  Text[0x0040]="Roll"
  Text[0x0041]="Pitch"
  Text[0x0042]="Yaw"
  Text[0x0046]="Priority"
  --******
  
  Text[0x004A]="Failsafe"
  Text[0x004B]="Main Menu"
  Text[0x004E]="Position"
  Text[0x0050]="Outputs"
  Text[0x0051]="Output Channel 1"
  Text[0x0052]="Output Channel 2"
  Text[0x0053]="Output Channel 3"
  Text[0x0054]="Output Channel 4"
  Text[0x0055]="Output Channel 5"
  Text[0x0056]="Output Channel 6"
  --Text[0x005E]="Inhibit"
  Text[0x005F]="Hold Last"
  Text[0x0060]="Preset"
  --Text[0x0061]="Custom"
  --Messages--
  Text[0x0078]="FM Channel"
  Text[0x0080]="Orientation"
  Text[0x0082]="Heading"
  Text[0x0085]="Frame Rate"
  Text[0x0086]="System Setup"
  Text[0x0087]="F-Mode Setup"
  Text[0x0088]="Enabled F-Modes"
  Text[0x008A]="Gain Sensitivity"
  Text[0x008B]="Panic Mode"
  Text[0x008E]="Delay"
  Text[0x0090]="Apply"
  Text[0x0093]="Complete"
  Text[0x0094]="Done"
  Text[0x0097]="Factory Reset"
  Text[0x009A]="Capture Failsafe Positions"
  Text[0x009C]="Custom Failsafe"
  Text[0x00A5]="First Time Setup"
  Text[0x00AA]="Capture Gyro Gains"
  Text[0x00AD]="Gain Channel Select"
  Text[0x00B6]="FM1"
  Text[0x00B7]="FM2"
  Text[0x00B8]="FM3"
  Text[0x00B9]="FM4"
  Text[0x00BA]="FM5"
  Text[0x00BB]="FM6"
  Text[0x00BC]="FM7"
  Text[0x00BD]="FM8"
  Text[0x00BE]="FM9"
  Text[0x00BF]="FM10"
  Text[0x00CA]="SAFE/Panic Mode Setup"
  Text[0x00CD]="Level model and capture attitude"
  Text[0x00D2]="Panic/SAFE Channel"
  Text[0x00DA]="SAFE Select Enabled"
  Text[0x00F2]="Fixed"
  Text[0x00F3]="Adjustable"
  Text[0x00F9]="Gyro settings"
  Text[0x00FE]="Stick Priority"
  Text[0x0100]="Make sure the model has been"
  Text[0x0101]="configured, including wing type,"
  Text[0x0102]="reversing, travel, trimmed, etc."
  Text[0x0103]="before continuing setup."
  Text[0x0104]="0104"
  Text[0x0105]="0105"
  Text[0x0106]="Any wing type, channel assignment,"
  Text[0x0107]="subtrim, or servo reversing changes"
  Text[0x0108]="require running through initial"
  Text[0x0109]="setup again."
  Text[0x010A]="010A"
  Text[0x010B]="010B"
  Text[0x0190]="Relearn Servo Settings"
  Text[0x019C]="Enter Receiver Bind Mode"
  Text[0x01D7]="SAFE Select Channel"
  Text[0x01DC]="AS3X"
  Text[0x01DD]="AS3X Settings"
  Text[0x01DE]="AS3X Gains"
  Text[0x01E0]="Rate Gains"
  Text[0x01E2]="SAFE Settings"
  Text[0x01E3]="SAFE Gains"
  Text[0x01E6]="Attitude Trim"
  Text[0x01E9]="Roll Right"
  Text[0x01EA]="Roll Left"
  Text[0x01EB]="Pitch Down"
  Text[0x01EC]="Pitch Up"
  Text[0x01EE]="Throttle to Pitch"
  Text[0x01EF]="Low Thr to Pitch"
  Text[0x01F0]="High Thr to Pitch"
  Text[0x01F3]="Threshold"
  Text[0x01F4]="Angle"
  Text[0x01F6]="Failsafe Angles"
  Text[0x01F8]="SAFE Mode"
  
  Text[0x01F9]="SAFE Select"
  Text[0x01FC]="Panic F Mode"  
  Text[0x01FD]="SAFE Failsafe FMode"
  Text[0x020A]="Restore from Backup"
  Text[0x0209]="Save to Backup"
  Text[0x020D]="First Time SAFE Setup"
  Text[0x021A]="Set the model level,"
  Text[0x021B]="and press Continue."
  Text[0x021C]="021C"
  Text[0x021D]="021D"
  
  Text[0x021F]="Set the model on its nose,"
  Text[0x0220]="and press Continue. If the"
  Text[0x0221]="orientation on the next"
  Text[0x0222]="screen is wrong go back"
  Text[0x0223]="and try again."
  Text[0x0224]="Continue"
  Text[0x0226]="Angle Limits"
  Text[0x0227]="Other settings"
  Text[0x0229]="Set Orientation Manually"
  Text[0x022B]="WARNING!"
  Text[0x022C]="This will reset the"
  Text[0x022D]="configuration to factory"
  Text[0x022E]="defaults. This does not"
  Text[0x022F]="affect the backup config."
  Text[0x0230]="0230"
  Text[0x0231]="This will overwrite the"
  Text[0x0232]="backup memory with your"
  Text[0x0233]="current configuartion."
  Text[0x0234]=""
  Text[0x0235]=""
  Text[0x0236]="This will overwrite the"
  Text[0x0237]="current config with"
  Text[0x0238]="that which is in"
  Text[0x0239]="the backup memory."
  Text[0x023A]="023A"
  Text[0x023D]="Copy Flight Mode Settings"
  Text[0x023E]="Source Flight Mode"
  Text[0x023F]="Target Flight Mode"
  Text[0x0240]="Utilities"
  Text[0x024C]="Gains will be captured on"
  Text[0x024D]="Captured gains will be"
  Text[0x0254]="Positive = Up, Negative = Down"
  Text[0x0260]="WARNING: 'TARGET'"
  Text[0x0261]="flight mode will be overwritten"
  Text[0x0262]="by the 'SOURCE'"
  Text[0x0263]="Fixed/Adjustable Gain"
  Text[0x0266]="Heading Gain"
  Text[0x0267]="Positive = Nose Up/Roll Right"
  Text[0x0268]="Negative = Nose Down/Roll Left"
  Text[0x0269]="SAFE - Throttle to Pitch"
  Text[0x026A]="Use CAUTION for YAW gain!"
  Text[0x8001]="Flight Mode 1"
  Text[0x8002]="Flight Mode 2"
  Text[0x8003]="Flight Mode 3"
end

johnnym007, 2021-06-10:
Thanks everyone for all your work on this.  Tested on an AR6610T and works fine setting failsafes and triggering bind mode.

Text[0x0017]="AR6610T"
Text[0x00F8]="0x00F8"  - I would guess this should be something like "System Settings" but do not have a Spektrum TX to verify

gbine, 2021-06-10:
> Thanks everyone for all your work on this. Tested on an AR6610T and works fine setting failsafes and triggering bind mode.
> 
> Text[0x0017]="AR6610T"
> Text[0x00F8]="0x00F8" - I would guess this should be something like "System Settings" but do not have a Spektrum TX to verify

@johnnym007 ,  I have a more updated file now with move text fields and also deals with some of the formatting issues.
I will add those two text fields and repost entire file after lunch (GMT-7) 
Testing and feedback is appreciated.

johnnym007, 2021-06-11:
Thanks @gbine 
Was there any progress on first time set-up for AR637T, is this working now?  Or working well enough to setup an RX without a Spektrum TX?

gbine, 2021-06-11:
No I have not sorted out the First Time Setup. I ended up having to order the USB logic analyzer and it just arrived on Tuesday.   I will be looking at that and how the visual menu items work (orientation) hopefully this weekend, if the flying conditions aren’t good ;-)

I would like to look at fixing the formatting more too.  There is some strangeness on some input fields, like the flight control switch.

I also see that my posting of the code is all messed up.   If you would like I can email you the file to work with.  I only have AR537TA Rx to work with, so any help with other receivers is appreciated.

I’m not sure is Pascal reads this, but if you do, I successfully installed the JP5in1 into my Jumper T16.   The internal connection cable is not straight through, but I was able to re-wire it to work.

Gordon

johnnym007, 2021-06-11:
Thanks @gbine I fixed the formatting in vscode, but have not had chance to try the script yet. 
I am planning on switching to Spektrum RXs for my growing plane fleet, but want to keep using my TX16s (if possible).  I guess the main blocker at the moment is setting up the stabilised RXs. 
I am happy to run testing as I get further receivers and will order an AR637T if I can help with testing. 
I have some Python scripting experience, but this is my first time looking at LUA scripts. 

John

johnnym007, 2021-06-16:
@pascallanger are you able to give me some pointers on the best way of snooping on the forward programming communications between a Spektrum TX and RX? (DX6 G3 & AR637T)
I have the USB/forward programming debug working on a MPM connected to an X9D+ so can debug on the OpenTX side of things.

pascallanger, 2021-06-17:
You need to open the DX6 G3 and connect/solder a logic analyzer on the RF module. From there you can see the messages between the TX and RX.
On the multi side you can see what's being sent/received for forward programming by using USB debugging as you said.
I do have an unpublished version of the lua script with some improvements but never really finished it so I haven't posted it...

gbine, 2021-06-17:
Hi Pascal, would you share your LUA script?
I’ve made some headway and don’t want to duplicate anything.
I was hoping to make some progress this past weekend as my logic analyzer arrived, but the Spektrum radio I had arranged does not support forward programming.  I am going to have to purchase one as the other people I know with suitable TX do not want me opening them up .

Thanks again for all your work in this.

pascallanger, 2021-06-17:
Here is the latest LUA script version with some debugs enabled to help finding new field types:
[DSM FwdPrg.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/6671784/DSM.FwdPrg.zip)

I have also modified the Multi DSM RX code to act as a RX with built-in forward programming: 
[DSM_Rx_cyrf6936.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/6671801/DSM_Rx_cyrf6936.zip)
It's working but I can't guarantee in which state I left the source code so you might have to play around. But basically with this (and without opening the original TX) as an example you can dump all the TX known texts (use the option field and assign it to the text id so you can change it live) and explore how the menus are working since you can send whatever you want to the TX to be displayed there.

For the first time setup, I have the info to go past the first page somewhere in an excel file where I've disassembled the different exchanges but I've never coded it... I should be able to find it if someone is interrested.

Pascal

pascallanger, 2021-06-17:
I don't think I'll have the time to look at it but I can answer questions if you have any.
I originally spent tons of time on reversing it (more than 2 weeks doing only this) but there was no much interrest since nobody was testing or providing feedback so I put it on the side even if as you've seen from above I've made the tools to progress further and faster...

Pascal

gbine, 2021-06-17:
Thanks Pascal,  this will help alot!
For the first time setup, I can only assume that the RX needs to know a number of things like wing configurtion and servo direction.  I would be interested to see your info on that if you happen to find it.

johnnym007, 2021-06-17:
Thanks Pascal - really appreciated. I'm sure there will be some questions once I have had chance to play with Multi DSM RX code.

gbine, 2021-06-17:
I should have stated this in previous email, but I would like to see your information on the First Time Setup.

kkriegel, 2021-06-18:
Folks, 
I have a brand new NX6 with the latest firmware loaded.  I also have a Salae Logic16. I am also willing to crack the case on the NX6 and see about instrumenting it. I am capable of doing a clean job, because I have all of the proper tools. Let me know if you want my help or if you've got this under control. I would need to learn how to get an IRX4 Plus into a debuggable state. I have a Horus X12S to place the IRX4 Plus into.

kkriegel, 2021-06-18:
I forgot to mention that I also have an AR637T receiver, non-BNF.

gbine, 2021-06-18:
@kkriegel , help is always appreciated.  Right now it appears that myself and @johnnym007 are attacking this independantly so maybe we should think about how to more easily share information.

I am hopeful that with Pascal's RX emulation code (shared a few posts back) for the MM, we won't need to perform surgery on anyone's TX 🤞

The LUA script included in the MM package is a good starting point, and I have added to it some, but needs work, both in logic and functionality.  Pascal mentioned yesterday that he had an unpublished (I read better) script, so I"m hoping he shares that.

I would like to unravel how to do the 'First Time Setup' as well as work on the Orientation section.
After that, I think it would be functional enough to take a virgin unit and configure it to a working state.

johnnym007, 2021-06-18:
@gbine I have a sacrificial Spektrum TX now, so plan to start analysing the forward programming comms over the weekend. As soon as I have anything worthy of sharing I will do.

johnnym007, 2021-06-18:
@pascallanger Are you able to assist with the exact connections and comms protocol to the Spektrum TX module?  From the information I have managed to uncover I am assuming it is SPI but I cannot seem to get anything sensible from the logic analyser.

pascallanger, 2021-06-18:
Share a picture of your module? And the file from the logic analyzer.

johnnym007, 2021-06-18:
This is the module - I have just stripped everything away - I was not expecting you to respond so quickly ;-) 

[image]

Connections where:
Pin 1 - GND
Pin5 - MOSI
Pin6 - ~CS
Pin7 - SCLK
Pin8 - MISO

I didn't run a capture, but it those I the correct connections I'll set it back up and try again.

pascallanger, 2021-06-18:
Humm on your picture pin 12 is obviously gnd, 11 is VCC...
Not sure where you have found the connections you state but I would connect all 8 ports starting from 1 run a capture with gnd connected and hopefully we will find the right signals.

johnnym007, 2021-06-18:
@pascallanger Sorry in my rush to reply I messed-up the numbering:

[image]

I will connect everything back up tomorrow, when I have some more time, and send the trace through if i am still having problems.

pascallanger, 2021-06-18:
Connect all 8 wires and not only the one you are pointing out. I don't know where you found your information about the spi but spektrum have a lot of different RF modules so having nearly all of the wires connected 8 out of 10 (+gnd and VCC) gives us the possibility to find our signals.

johnnym007, 2021-06-18:
Ok - thanks Pascal.

johnnym007, 2021-06-19:
@pascallanger  - Attached are some Logic 2 (Saleae) captures with 9 channel connected (one additional pin on the module is GND).  In my ignorance I had though this would have been easier to interpret, but I could really use some help trying to identify the protocol and correct signals.
[Traces.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/6681023/Traces.zip)

pascallanger, 2021-06-20:
Channels 3,4, 5 and 8 do not see any signals. Verify if they are correctly connected and if they are then move 2 of them to the other pins which you haven't covered yet.

johnnym007, 2021-06-20:
@pascallanger Sorry I just deleted that post......I was hoping you had not looked at it yet.  I have been playing around further and found that my ground was not properly connected.  Sorry for wasting your time.  This explains why things looked so strange on the traces.
Looks like with that sorted I am getting SPI comms on Pins 3,4,5 & 8 - trace attached.  I still need to work-out which of Pins 4 & 5 are MISO & MOSI, but I finally feel like I am making progress.
[Traces.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/6681947/Traces.zip)

pascallanger, 2021-06-20:
As far as I can see MOSI and MISO order are ok on the dumps.
On MOSI you can see the channels being sent starting from the 3rd byte. On MISO the telemetry dat should start on the 4th byte.

johnnym007, 2021-06-20:
Thanks Pascal - really appreciated. Sorry again for wasting your time with the erroneous traces.

johnnym007, 2021-06-22:
@gbine @kkriegel 

Right I'm up and running with the logic analyser now and have written an Excel VBA function to help format it in a half sensible way, ignoring anything that is not related to the forward programming communications.

Link below (if it works) is to the raw logic analyser data and some sorted data where I have started to break-up the code to start to understand things.  Also included is a video of the screen recorded at the same time as the capture, so the time stamp on the logic output should roughly correlate with the videos.

https://drive.google.com/drive/folders/1jBNWKU9Ybh2kexh-DWsj0j4Hdnogc2lp?usp=sharing

After some time looking at this I have realised (and tested to confirm) that the RX leads most of the communcations, with the TX effectively acknowledging receipt or providing the requested info.  Once I start to understand this a bit further I plan to put together a command set document as a reference to understanding the command/response structure.

Anyway have a look and if you would like anything testing to prove a theory let me know and I will grab a capture and video and upload it (this might take a day or two).

johnnym007, 2021-07-01:
> Here is the latest LUA script version with some debugs enabled to help finding new field types:
> [DSM [...]

Hi Pascal - I am making some real progress and now have the first part of the "First Time Setup" working.  This sends the Channel order and reversing status from the TX.  There is still more to do as the RX also requests endpoints, mixing etc which I still need to decipher. 
One thing I believe it that the are more than one text lists held by the TX as I am definitely seeing an overlap of text strings.  I would like to run the DSM_RX code that you have linked above, but I cannot get it to bind to the TX with telemetry (it always binds without).  I am using multi version 1.3.2.86, but just do not seem to be able to get it working.
Heres the Debug from the bind:

16:01:10.528 -> Protocol selected: 70, sub proto 0, rxnum 0, option 0
16:01:10.528 -> Proto=DSM_RX, nbr_sub=0, Sub=, Opt=0, FS=0, CHMap=1, rfSw=3
16:01:17.341 -> RX:0, CH:5
16:01:17.341 -> RX:0, CH:5
16:01:17.341 -> RX:0, CH:5
16:01:17.341 -> RX:1, CH:5
16:01:17.374 -> RX:0, CH:5
16:01:17.374 -> RX:0, CH:5
16:01:17.374 -> RX:0, CH:5
16:01:17.442 -> RX:10, CH:7
16:01:17.442 -> RX:16, CH:7
16:01:17.442 -> ID= 7B E8 8B 49, num_ch=8, type=B2

I guess I am missing something somewhere. could you provide some assistance?
Thanks,
John

pascallanger, 2021-07-01:
> I am making some real progress and now have the first part of the "First Time Setup" working. This sends the Channel order and reversing status from the TX. There is still more to do as the RX also requests endpoints, mixing etc which I still need to decipher.

This is where I ended up too, sorry to have not shared the file...

> One thing I believe is that there are more than one text lists held by the TX as I am definitely seeing an overlap of text strings. 

Yeah I've seen it too. It is even more obvious when you start playing with the RX side to display all the text, you end up with different stuff...

Let me check the DSM_RX with forward telemetry code, like I said this was a working file so I don't know in which state I let it but I know it is working so it must be something small.

[86 more replies]

closed 2023-10-30 by pascallanger