pascallanger/DIY-Multiprotocol-TX-Module #410: Support WFLY New Protocol ET07 Series Radio

richardclli, 2020-08-02:
As I know that this protocol is not supported yet.  I just got a 2nd hand radio and receiver, let's see what can be worked out.

First of all the radio used the RF chip A7106, I think is should related to the A7105 in the multiprotocol module, right?

[image]

Will try to work out some dumps and post up here later.

pascallanger, 2020-08-02:
I gave up on this protocol some time ago. Let's see if we will have more inspiration this time...

richardclli, 2020-08-02:
Any dumps that I can help analyze? Or I need to work out the dump by myself?

pascallanger, 2020-08-02:
Having a second dump would help to see the ID and frequencies. May be that would give a clue...

richardclli, 2020-08-03:
Will try to do some dumps this week when I have some spare time.

richardclli, 2020-08-26:
After the OMP protocol, I already know what you need about the dumps.  Will work out some nice dumps this week.  Let see if we can make it works ^.^

richardclli, 2020-09-03:
Finally got everything ready for the dump.  Since I am not sure whether it uses a 3-wires or 4-wires SPI interface, I have all 5 signals tapped in the dump:
- 0: GIO2
- 1: GIO1
- 2: SDIO
- 3: SCK
- 4: SCS

The dump is done by recording the signal  for 20 secs after switching on the transmitter.

After reviewing the dump, I suspect it uses a 3 wires SPI, am I right?

I am not sure if you may have the decoding script for A7106 decoding, if not then I will try to write one for the DSView.  I just rename the file to give it an extension .zip to pass upload.

[ET07-1.dsl.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5167547/ET07-1.dsl.zip)

richardclli, 2020-09-07:
Some More dumps:

1. Bind to Receiver: txid: 00002313, rxid: 00000BD5
[ET07-bind-txid00002313-rxid00000BD5.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5184752/ET07-bind-txid00002313-rxid00000BD5.zip)

2. Bind to Receiver: txid: 00002313, rxid: 00002E26
[ET07-bind-txid00002313-rxid00002E26.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5184754/ET07-bind-txid00002313-rxid00002E26.zip)

3. Switch on and auto bind to receiver: rxid: 00000BD5
[ET07-on-autobind-rxid00000BD5.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5184757/ET07-on-autobind-rxid00000BD5.zip)

3. Stick Move: THR Down, RUD Left -> THR Up -> THR Down -> RUD Right -> RUD Left -> THR Up -> THR Down -> RUD Right -> RUD Left
[ET07-thr-rud.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5184758/ET07-thr-rud.zip)

4. Stick Move: ELE Down, AIL Left -> ELE Up -> ELE Down -> AIL RIGHT -> AIL Left -> ELE Up -> ELE Down -> AIL Right -> AIL Left
[ET07-ele-ail.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5184761/ET07-ele-ail.zip)

Please check and see if you can find any insight in these files.  I will try to read and understand the protocol as well.  Let's see what can we find out.

pascallanger, 2020-09-07:
Sorry, I'm currently reversing the DSM forward programming which is taking more time than I thought. I'll get to this one later...

richardclli, 2020-09-08:
No problem, just take your time and have fun ^.^

richardclli, 2020-09-24:
I just did some study on the 32-bytes TX Payload P[0:31], the sticks seems to be encoded with 12-bits:
Ele = P[7](7-4bits) P[6]
Ail = P[7](3-0 bits) P[5]
Rud = P[10](7-4bits) P[9]
Thr = P[10](3-0 bits) P[8]

I will continue to work on more details of the payload or other inputs.

richardclli, 2020-09-24:
During bindings P[0] = 0x0F, P[1,2] Has fixed value 0x13, 0x23, P[5] = 0x01, P[6] is counting

richardclli, 2020-10-09:
I have done more studies on other channels and got the following information in general:

Ch1 (Ail) = P[7](3-0 bits) P[5]
Ch2 (Ele) = P[7](7-4 bits) P[6]
Ch3 (Thr) = P[10](3-0 bits) P[8]
Ch4 (Rud) = P[10](7-4 bits) P[9]
Ch5 = P[13](3-0 bits) P[11]
Ch6 = P[13](7-4 bits) P[12]
Ch7 = P[16](3-0 bits) P[14]
Ch8 = P[16](7-4 bits) P[15]

P[19] always = 0x88

richardclli, 2020-10-09:
@pascallanger Anything comments?  Maybe if you can point out where the pseudo random is used, I can see if I can hack the pseudo random formula.

richardclli, 2020-10-16:
Today I example the captured data again and see if I can get anything about the binding process, the TX will transmit these payloads to the receiver, which is quite different from normal channel payload:

0F13230000013500000000000000000000000000000000000000000000000000
0F13230000010E00000000000000000000000000000000000000000000000000
0F13230000011000000000000000000000000000000000000000000000000000
0F13230000011200000000000000000000000000000000000000000000000000
0F13230000011400000000000000000000000000000000000000000000000000

P[0] = 0F seems to define this kind of binding packet, normal packet has a 00 here
P[1-4] = the transmitter ID?  The ID of my transmitter is 00002313, LSB first
P[5] always = 01, not sure what it means
P[6] seems to be some kind of counter

Still not sure how the transmitter get the Receiver ID

richardclli, 2020-10-20:
@pascallanger Would you mind take a quick look on my findings and give me some advice what can I help to do next?

richardclli, 2020-11-02:
@pascallanger Do you have time to work on this?  What can I help?

pascallanger, 2020-11-02:
Ok let me start looking into it.
The important part of a protocol is the checksum (addition, xor, CRC...) and the frequency hopping. This is where you first need to look at since without this information there is no way to transmit valid packets.

pascallanger, 2020-11-02:
Here is a graph of the 1st 222 freq values:
[image]
As you can see there are no repeat in the sequence.
This is what I was referencing at the begining of this thread: "If I remember they are using a pseudo generator for the RF".

pascallanger, 2020-11-02:
My guess is that P[1] and P[2] are the pseudo random generator on 16 bits and somehow gives an index to the cur/next RF channel to be used.

pascallanger, 2020-11-02:
Ok I kind of know now how to go from P[2]P[1] from one line to the next. Just need a little more time to fully have it. It's a left shift by 1 of the prev value with a xor 0x8000 depending if the prev value high bit was set or not.
But there is one trick which needs to be understood when the left shift by 1 gives a 1 on the high bit then an operation is applied (+ or - or OR or ...) to test generate a new value which then follow the above.
Getting somewhere ;-)

pascallanger, 2020-11-02:
Next: need to figure out how the RF channel is selected based on P[2]P[1]

pascallanger, 2020-11-02:
FYI, the above function is initialized with 0x4000 in P[2]P[1] on the dumps.

pascallanger, 2020-11-02:
There are 64 RF channels being used starting at 0x10 and spaced by 0x02
So an ordered table is given by:
for(uint8_t i=0;i<64;i++)
  chan[i]=(i<<1)+0x10;
I think what we need to look for is "i" inside P[2]P[1].
To be continued...

pascallanger, 2020-11-03:
> Today I example the captured data again and see if I can get anything about the binding process, the TX will transmit these payloads to the receiver, which is quite different from normal channel [...]

P[6] is not a simple counter. This is the next RF channel to be used using this: ch=(P[6]<<1)+0x10 (see my previous post). 
Therefore: 35 -> ch=7A, 0E -> ch=2C, ...

pascallanger, 2020-11-03:
As said yesterday, the only missing key now to be able to reverse this protocol is to figure out how the RF channels are selected based on the P[2]P[1] values.

pascallanger, 2020-11-03:
I'm a bit in stuck on how the RF channel is calculated. I've looked at my old dump, I did only one and no bind or startup in it, which might give an insight:
[image]
As you can see I've been able to find the same pseudo sequence in both your and my dump.
What we can conclude is that most likely not only P[2]P[1] are used but also P[4]P[3] in the channel calculation.
The calculation gives the same low nibble but the high nibble differs sometime by 10 or more.

pascallanger, 2020-11-03:
Based on the above, we might say:
 - the low nibble being identical might indicate that it's based on the 3 of 022**3** and 031**3**.
 - the high nibble having a +/-10 difference could indicate that it's based on the 2/1 of 02**2**3 or 03**1**3, but it could also be with 0**2**23 and 0**3**13.
Not sure how to progress...

richardclli, 2020-11-04:
Let me take a look and see if I can figure out anything.  BTW, can you share the Excel that you have made about the dump and mappings?  I will see if I can do more dumps and match more examples for further decoding.

pascallanger, 2020-11-04:
[ET07-RF.xlsx](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5486416/ET07-RF.xlsx)
The RF channel & index column could be out of phase by 1 line with the Pseudo since I don't know if the Pseudo is giving the current channel or the next channel.

pascallanger, 2020-11-08:
00000223 is my TXID. So P[4]P[3] is somehow the TXID.
00002313 is your TXID which shows that P[4]P[3] = 0x0313 = TXID & 0x1FFF -> 1FFF could be as low as 0x03FF. We are missing other examples to narrow it down...

pascallanger, 2020-11-14:
> Let me take a look and see if I can figure out anything. BTW, can you share the Excel that you have made about the dump and mappings? I will see if I can do more dumps and match more examples for further decoding.

Have you looked into it?

richardclli, 2020-11-15:
Seems the random generator is not a Linear Congruential type.  Will continue to work on more hacks to see if I can find out more.

benlye, 2020-11-15:
I am *so* close to getting this.  I have 5 of the six bits in the channel index worked out, I just need the sixth bit and it will all fall into place.

The five bits I have so far are derived from the pseudo random number.  I believe the sixth is too, I just haven't cracked the pattern yet.  I have tested what I've got so far against all the data in the ET07-RF file and it looks good (comparing the 5 bits I generate with the rightmost 5 bits in the index value)

There's probably a more elegant way to do the binary operations but this is working for me (PowerShell code).

```powershell
$ChannelIndex = 0

# First bit
If ($Pseudo -band 0x80) {
    $ChannelIndex += ($Pseudo -band 0x1)
} Else {
    $ChannelIndex += ($Pseudo -band 0x1) -bxor 0x1
}

# Second bit
If ($Pseudo -band 0x100) {
    $ChannelIndex += ($Pseudo -band 0x2)
} Else {
    $ChannelIndex += ($Pseudo -band 0x2) -bxor 0x2
}

# Third bit
If ($Pseudo -band 0x200) {
    $ChannelIndex += ($Pseudo -band 0x4) -bxor 0x4
} Else {
    $ChannelIndex += ($Pseudo -band 0x4)
}

# Fourth bit
If ($Pseudo -band 0x800) {
    $ChannelIndex += (($Pseudo -band 0x10) -bxor 0x10) -shr 1
} Else {
    $ChannelIndex += (($Pseudo -band 0x10) -shr 1)
}

# Fifth bit
If ($Pseudo -band 0x2000) {
    $ChannelIndex += (($Pseudo -band 0x40) -shr 2)
} Else {
    $ChannelIndex += (($Pseudo -band 0x40) -bxor 0x40) -shr 2
}
```

benlye, 2020-11-16:
Hmm, just realized I only tested one of the two data sets.  The first four bits are the same in both but bit five is always inverted for the data from ID 0223.

Still need to crack bit 6.  I'm close, but it's late and I'm not getting it tonight.

pascallanger, 2020-11-16:
> Seems the random generator is not a Linear Congruential type. Will continue to work on more hacks to see if I can find out more.

The random generator is already reversed... Now the game is to find how the channel is selected based on it.

pascallanger, 2020-11-16:
> I am _so_ close to getting this. I have 5 of the six bits in the channel index worked out, I just need the sixth bit and it will all fall into place.
>
> The five bits I have so far are derived from [...]

Awesome!!! Getting somewhere

benlye, 2020-11-16:
I'm still stuck on the 6th bit of the index.  I suspect it is linked to the last bit in the pseudo, reversed based on other bits in both the high and low bytes of the pseudo, and then affected by the ID, but I can't figure it out yet.

This script works for bits 1-5 for both IDs.  I used the fifth bit of the radio ID for no good reason other than I was calculating the fifth bit of index.  It could just as easily be the sixth or ninth bit of the ID influencing it.

```powershell
$ChannelIndex = 0

# First bit
If ($Pseudo -band 0x80) {
    $ChannelIndex += ($Pseudo -band 0x1)
} Else {
    $ChannelIndex += ($Pseudo -band 0x1) -bxor 0x1
}

# Second bit
If ($Pseudo -band 0x100) {
    $ChannelIndex += ($Pseudo -band 0x2)
} Else {
    $ChannelIndex += ($Pseudo -band 0x2) -bxor 0x2
}

# Third bit
If ($Pseudo -band 0x200) {
    $ChannelIndex += ($Pseudo -band 0x4) -bxor 0x4
} Else {
    $ChannelIndex += ($Pseudo -band 0x4)
}

# Fourth bit
If ($Pseudo -band 0x800) {
    $ChannelIndex += (($Pseudo -band 0x10) -bxor 0x10) -shr 1
} Else {
    $ChannelIndex += (($Pseudo -band 0x10) -shr 1)
}

# Fifth bit
If (($Pseudo -band 0x2000 -and $RadioId -band 0x10) -or (-not ($Pseudo -band 0x2000) -and -not ($RadioId -band 0x10))) {
    $ChannelIndex += (($Pseudo -band 0x40) -shr 2)
} Else {
    $ChannelIndex += (($Pseudo -band 0x40) -bxor 0x40) -shr 2
}
```

richardclli, 2020-11-17:
I have done an extensive search using your findings as a bases, using every pseudo bits XOR with another pseudo bits and then XOR with the id bits and have the following findings, which seems more general, but still no luck to figure out the final one:

P = pseudo
I = ID
X = Index

X[0] = P[0] ^ P[7] ^ I[0]
X[1] = P[1] ^ P[8] ^ I[1]
X[2] = P[2] ^ P[9] ^ I[2]
X[3] = P[4] ^ P[11] ^ I[3]
X[4] = P[6] ^ P[13] ^ I[4]
X[5] = ??

richardclli, 2020-11-17:
Just a guess, there should be a intermediate result that will finally XOR the ID to produce the final channel index.

benlye, 2020-11-17:
Could X[5] be a combination of bits already in the index, possibly combined with other P or I bits?

richardclli, 2020-11-18:
What I have tried is to pool up all different bits including P, I, X, and search for 2nd order (A ^ B) and 3rd order (A ^ B ^ C) using the pool, but still only what I have mentioned is discovered, still cannot got anything about the 6th bit.  Maybe it is not a combination using simple bit XOR.   Not sure what can be tried next.

richardclli, 2020-11-18:
Today, I have done an experiment in which I think can lead to a break through, however, more time is needed.........

What I did is to use the P (Pseudo) bits and the I (Id) bits and scan through the whole sample and try to eliminate bit by bit to see which bits can be eliminated but still gives a unique X (Index) output.  Here is my findings:

## X00 ##
['P00', 'P07']
## X01 ##
['P01', 'P08']
## X02 ##
['P02', 'P09']
## X03 ##
['P04', 'P11']
## X04 ##
['P06', 'P13', 'I08']
## X05 ##
['P00', 'P01', 'P02', 'P03', 'P04', 'P05', 'P06', 'P07', 'P08', 'P09', 'P10', 'P11', 'P12', 'P13', 'P14', 'I08']

Which means X05 depends almost all Pseudo bits, so how the bits are processed to give the final bit?  Still need to work on more possibilities.  However, the symptom about Id bits, since we have only 2 samples, it is hard to tell.

richardclli, 2020-11-18:
I tried XOR and Addition with the found dependency bits, and no result, seems the final bit is not that simple.........

pascallanger, 2020-11-18:
I think I have the bit 6. It needs to be fully ironed out and looked for exceptions but it is a rule to prevent 2 subsequent RF channels to be too close to each others which is common practice.
The rule looks to be the following:
- calculate the new index as you've indentified above using bit1..5
- if abs(prev_index-calc_index)<=9 then set bit 6  (|0x20) // prevents prev and cur channels to be too close

Pascal

pascallanger, 2020-11-18:
Shout I had something kind of working along the lines above but I've changed my code in excel and now I can't find the previous results anymore. I've tried undoing but can't get back... Still trying to figure out what I applied before to get a match...

benlye, 2020-11-18:
I think there is a strong correlation between index bit six and the two high bits in the pseudo.  I may have looked at this for too long and am seeing things, but...

If you take any RF index pair R and R+0x20 there are interesting patterns in the frequency at which they occur.  You might expect that they would appear at fairly equal frequencies throughout the range of pseudo values but they don't - there are definite boundaries where things change.

E.g. compare index 0x0A and 0x2A (same first 5 index bits, only bit 6 changes) for radio ID 0x0313:
| Pseudo | RF Index | Bit 6 |
| - | - | - |
| < 0x4000 | Always 0x0A | 0 |
| > 0x4000 && < 0x8000 | 0x0A or 0x2A | 0/1 |
| > 0x8000 && < 0xC000 | Always 0x2A| 1 |
| > 0xC000 | 0x0A or 0x2A | 0/1 |

RF indexes 0x05 and 0x25 are even more interesting:
| Pseudo | RF Index | Bit 6 |
| - | - | - |
| < 0x4000 | Always 0x25 | 1 |
| > 0x4000 && < 0x8000 | Always 0x05 | 0 |
| > 0x8000 && < 0xC000 | Always 0x05 | 0 |
| > 0xC000 | Always 0x25 | 1 |

Using the other radio ID changes the values, but the basic pattern is still evident.

Might be something, might not, I haven't looked all the way through it yet and there's obviously more to figure out (if this is even a step forward).

richardclli, 2020-11-19:
@pascallanger I got your message by e-mail ^.^ which means the final bit depends on previous Channel, let see if I can crack the logic using this direction.

richardclli, 2020-11-19:
Using bitwise correlation analysis using last Channel ID as well, 'X05' is related to 

['P15', 'X00', 'X01', 'X03', 'X04', 'LX00', 'LX01', 'LX02', 'LX03', 'LX04']

And........

I got the final bit ^.^, but still have one sample not following this rule:

The logic is:
  if (abs(X[4:0] - LX[4:0]) <= 9
    X[5] = P[15]
  else
    X[5] = ~ P[15]

Not sure why there is one exception, maybe need more dumps to figure out, we do not have enough samples, especially ID.

The exceptional line is:
  P = 52954
  I = 787
  X = 4

richardclli, 2020-11-19:
Oh, I know why, we already got the formula correct, the exceptional one is wrong because it is between 2 different IDs and the last Index is not valid.  Now I think we can start programming and generate different IDs to verify different ID correlation guess with only 2 samples.

pascallanger, 2020-11-19:
> Using bitwise correlation analysis using last Channel ID as well, 'X05' is related to
>
> ['P15', 'X00', 'X01', 'X03', 'X04', 'LX00', 'LX01', 'LX02', 'LX03', 'LX04']
>
> And........
>
> I got the final bit [...]

Ah I knew it ;-) Now I remember that I included Pseudo[15] but when I've cleaned my code and done my writeup I've completly forgotten about it which is what through me off yesterday evening... Somehow I was not able to put it back together, just too tired I think... I've been trying to get M-Link to work the last few days without any success yet...

benlye, 2020-11-19:
Well, that was fun ;-)

richardclli, 2020-11-19:
Well, finally got the work done, now we need to write more code to generate more Radio ID for testing and see if my assumptions about the X[4:0] and ID[4:0] is correct.  Only 2 samples cannot prove it.  Anyway, we can now develop working code, and just need to generate different radio ID samples to see if the assumptions are correct or not.

richardclli, 2020-11-19:
> 
> 
> Well, that was fun ;-)

It is always fun in cracking encodings.  Let me know if there are some other encodings awaiting to crack. ^.^

richardclli, 2020-11-23:
So, what else may I help to work out?  Coding?  Ah, I think I will do some dumps about the telemetry.

richardclli, 2020-12-03:
@pascallanger Are you going to work out some codes?  If you are busy, you can commit a template and I can try to fill out the details.

pascallanger, 2020-12-05:
I've put together the template (without telemetry). Now I need to code the protocol...

richardclli, 2020-12-10:
I will try to work this out this weekend.

pascallanger, 2020-12-10:
The latest version I've just uploaded gets the RX connectivity working time to time, it's totally jerky and unusable but it's getting somewhat there...
The RX telemetry also comes time to time. The first value is the RX voltage, the second none zero value seems to be the RSSI.
I haven't tried to bind the RX but used instead the TX ID the RX was bound to. So I don't know if this part works yet.
Pascal

richardclli, 2020-12-11:
Thanks you, let me see if binding works.

pascallanger, 2020-12-11:
I've fixed the normal mode. Link is perfect with no lag or jerkiness.
I've added telemetry: A1=RX_Batt (Ratio 12.7), A2=Ext_Batt (Ratio 12.7), RX RSSI, TX RSSI, TX LQI (100%=all telem packets received).
Bind is not working (not too surprised), I'll see if I can fix it today...

[30 more replies]

closed 2021-01-29 by pascallanger