pascallanger/DIY-Multiprotocol-TX-Module #412: Volantex Super Cub V756-2

Tygrys-1, 2020-08-05:
This model is using Volantex radio with 5 channel receiver EAR403. Is it possible to bind multimodule to this receiver? The V761 protocol does not seem to work on 4-CH mode, maybe this EAR403 is a different kind of a sub protocol, or maybe a different protocol?

pascallanger, 2020-08-05:
No idea... I can't even get a picture of what the original radio looks like.
Do you have the radio or only the receiver?

pascallanger, 2020-08-05:
I've found this link: https://fr.aliexpress.com/item/4000847736638.html
Are you sure it doesn't bind to the V761 protocol since the TXs look exactly the same from outside?

Tygrys-1, 2020-08-05:
Hi Pascal.
This is the [receiver](https://www.monkeyhobby.com/volantex-rc-ear403-5ch-receiver-with-gyro-for-rc-plane-76502-76801-76702-74701-74706-75706-75704/)
And you can find the transmitter on the same page as EAT 403, the description is wrong, as it is a 5 channel transmitter. It has extra switch on the left rear side, and it is driving the 5-th channel you can use for whatever function you want. The switch on the right is the gyro Easy-Normal-Expert function switch.

Tygrys-1, 2020-08-05:
I have tried V761 several times with different TX id's, with no luck on binding. So I think it might be a little different transmitter/receiver combo, mainly due to the spare 5-th channel.

Tygrys-1, 2020-08-05:
> 
> Do you have the radio or only the receiver?
I have them both, the receiver is glued in the plane, but it is the one you linked from Ali. The transmitter looks exactly he same from the outside, but EAT 403 has extra 3 position button on the left. There is no EAT 403 or other name on it.

Tygrys-1, 2020-08-05:
[YouTube](https://www.youtube.com/watch?v=F7cvBxGriBY), look at 1:57
It seams that this is a new version of 765-2, the older one was shipped with the Volantex transmitter with channel reverse buttons on the front. The new version have the gyro.

pascallanger, 2020-08-05:
Do you have an internal or external module?
Do you have a logic analyzer?
Can you do a picture of the radio internal and more specifically the chip(s) near where the antenna is connected.

Tygrys-1, 2020-08-05:
External URUAV TMX5 Lite on Taranis X9 Lite. 
I have Saleae 8 channel clone.
I can do the picture. Will try tommorow.

pascallanger, 2020-08-05:
Sounds really good!
I'm thinking that you can hook your logic analyzer to the SPI bus and do some dumps. If you know what you are doing then go ahead and post the dumps.

Tygrys-1, 2020-08-05:
Ok. I think I can, but never tried this so I'll need some guidance.

Tygrys-1, 2020-08-06:
[image]
[image]
[image]

pascallanger, 2020-08-06:
Great, on the last picture we see the signal names. You want to connect these to your logic analyzer:
From top to bottom on the last picture:
1. SPI_SS -> analyzer port 1
2. unconnected
3. GND -> anaylzer GND
4. RST_N? ->analyzer port 2
5. MISO -> analyzer port 3
6. SPI_CLK -> analyzer port 4
7. MOSI -> analyzer port 5

Tygrys-1, 2020-08-06:
Will do. Should I set Analyzer to SPI or just take the raw dump?

pascallanger, 2020-08-06:
If you know how to do it then sure add a SPI decoder and point it to the different signals: SPI_SS (CS), MOSI, MISO, SPI_CLK (CLK).
Set the sample rate to 10M to start with.

Tygrys-1, 2020-08-06:
I have no 10MS rate, so I've choosen 12MS. Here is the 10s dump of the start sequence of the transmitter.
[12 MHz, 120 M Samples [1].zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5034403/12.MHz.120.M.Samples.1.zip)

pascallanger, 2020-08-06:
Can you get a picture of the other side of the red RF board?
I have the impression that the RF component is a LT8910 or similar. V761 is using a XN297.
FYI depending on how they use the LT8910 we might not be able to emulate it...

Tygrys-1, 2020-08-06:
[image]

pascallanger, 2020-08-06:
Ok I don't know what the ref means but I'm pretty sure it's a LT8910.
The bit rate is 62.5Kbps which is annoying since the NRF based emulation that I've written cannot be used directly...
I'm thinking the CC2500 will be able to emulate it but that will have to be tested.
I can try to look into it but without access to the original TX/RX, it's going to be difficult in this case...
If anyone has a spare, it would be awesome to send it my way, I can return it when I'm done.

Tygrys-1, 2020-08-06:
So I can unsolder the analyzer?
Was my dump readable? Can you suggest the right clock settings for the SPI analyzer? I think of triggering at falling edge, because otherwise it has little sense. I'm new to this and much to learn.

BTW, the externall antena on this transmitter is a fake. The whole antenna wire was hanging inside the case, and no way to fit it inside the plastic antenna stick. Another chineese crap...

pascallanger, 2020-08-06:
The dump is readable, well done and SPI decoder configured correctly.
Don't unsolder anything yet. In the hope that I can emulate it, can you do more dumps :
 - one dump where you move only one stick in one direction or a feature button. Make sure to name the files accordingly like Aileron_mid_left_mid_right_mid.
 - one dump where you launch a bind if there is a specific bind procedure on the TX

Tygrys-1, 2020-08-06:
There might be a problem - after assembling the TX after photos the radio now refuses to boot with the analyzer connected. When I detach the CLK wire, it boots ok. I do not know what has happened. No shortcuts, no other things changed - just unscrewed the board and then screwed it to the same place. Have no idea what to do.

pascallanger, 2020-08-06:
The logic analyzer must be powered on otherwise the SPI bus will be affected.

Tygrys-1, 2020-08-06:
You are right, that way it is working ok. I have tested channels, but without arming, and this has side effect of maybe reversing channels, so to return to the normal I need to bring the model and test if everything is ok. Will continue tomorrow.

pascallanger, 2020-08-06:
Channels direction are not really important for now. What's important is to identify the bytes of the payload and bind.
On the V761 remote even the trims have features, make sure to capture those too if this is the same on this remote.

pascallanger, 2020-08-06:
The CC2500 can do GFSK @62.484kbps using M=59 and E=11.
From what I see, all LT8910 advanced options are off (FCC, whitening, encoding...). So it should be doable. But not having the TX (& RX) is going to be quite a challenge since a single mistake in the datasheet interpretation will prevent it to work with little to no way to know what's really going on...

Tygrys-1, 2020-08-07:
[Startup with receiver on.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041895/Startup.with.receiver.on.zip)
[Arming- Thr Up-down.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041896/Arming-.Thr.Up-down.zip)
[Ail mid-left-mid-right-mid-x2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041899/Ail.mid-left-mid-right-mid-x2.zip)
[Ele-Mid-Dn-Mid-Up-Mid-x2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041900/Ele-Mid-Dn-Mid-Up-Mid-x2.zip)
[rud-mid-left-mid-right-mid-x2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041904/rud-mid-left-mid-right-mid-x2.zip)

Tygrys-1, 2020-08-07:
[THR 0-100.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041915/THR.0-100.zip)
[5th-mid-back-mid-fwd-mid-x2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041921/5th-mid-back-mid-fwd-mid-x2.zip)
[Ail-Mid-Left-Mid-Right-Mid-x2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041924/Ail-Mid-Left-Mid-Right-Mid-x2.zip)
[Ail-Mid-Left-Mid-Right-Mid-x2 armed model off.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041926/Ail-Mid-Left-Mid-Right-Mid-x2.armed.model.off.zip)

Tygrys-1, 2020-08-07:
This is the 2 cycle stick move, but without the model turned on, and without the arming cycle (THR UP/DN)
[Ail-Mid-Left-Mid-Right-Mid-x2a.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041994/Ail-Mid-Left-Mid-Right-Mid-x2a.zip)
This is the 2 cycle stick move, but without the model turned on, and without the arming cycle (THR UP/DN)
[rud-mid-left-mid-right-mid-x2a.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5041995/rud-mid-left-mid-right-mid-x2a.zip)
This is the 2 cycle of the switch, but there were no beeps, as the TX was not armed I think.
[5th-mid-back-mid-fwd-mid-x2a.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5042011/5th-mid-back-mid-fwd-mid-x2a.zip)

The files are much shorter at this stage.

Tygrys-1, 2020-08-07:
If I FOOBAR the plane, the radio is yours ;-)

pascallanger, 2020-08-07:
Don't make it on purpose that would be too bad ;-)

Tygrys-1, 2020-08-07:
Sure, but it will be less pain if thera is a purpose ;-)

MirlynG, 2020-12-23:
Hi Pascal. New to all this, and as there has been no movement on this topic for some time I don't know if you will still be interested. However, like a number of others, I assumed that my new Volantex Mustang 750mm would to bind to my Taranis QX7 with an IRX4 MM using the V761 protocol. I updated the firmware on both Tx and MM to no avail. So having done the research, I now find that the protocol is different, i.e. the V765, and in the process came across this thread. 
As it turns out, there appears to be a problem with my receiver anyway. On both the two stabilizing channels the rudder will not re-center on releasing the stick. It will if you flick the stick back a bit and, rather oddly, it will also re-center if you flick the aileron stick too, but not the elevator?? If you use expert mode it works fine! I swapped channels with the elevator and it then sticks but rudder is fine. Problem has to be the receiver. 
Only found two suppliers (Chinese) of the required EAR403 receiver, and I have now been waiting three weeks for one which seems to be stuck in China. There is very limited space in the plane and the only alternate Rx with auto level I could find to fit is the LEMON RX STABILIZER PLUS, so I have ordered one of these from a UK supplier so should be here soon. It runs DSMX so will work with my Taranis/MM.
Anyway, the whole point of this rambling post is that I'm sure there must be a number of people who would like to bind their Tx's to this plane and I saw you suggested that if you got hold of a TX and Rx you might be able to sort out a file to update the IRX4 firmware to include this V765 protocol. Is this still the case? If so, I would be happy to ship mine to you as they will be redundant, although I would like to get them back when you're done with them. Hopefully this might help a few people. Don't forget about the receiver problem described above though. 
Let me know your thoughts.

pascallanger, 2020-12-23:
Sure send me a PM on RCGroups (user hpnuts).

pascallanger, 2021-01-19:
@MirlynG Will you ship your V765 TX in the end?

MirlynG, 2021-01-24:
Hi Pascal. Sorry I have not replied sooner. For a number of reasons I have
decided that I will send you the Tx and Rx, but it will not be necessary
for you to return it to me when done. Could you please confirm your
address. I will ship it this week and let you know when it is on its way.
Regards
Mike

pascallanger, 2021-01-24:
Can you PM me on RCGroups user hpnuts?

canalrodrigo, 2021-01-28:
Hi, I also being struggling for a while with this issue, as many others I think it would be great to have access to the protocol to bind the Volantex Mustang 750mm (768-1), if there is anything I can do to help please let me know

pascallanger, 2021-01-28:
@canalrodrigo I'm waiting for someone to ship me the TX + RX so I can have a look...

MirlynG, 2021-01-28:
Tx and Rx are on the way today. Hope you will get them within 5 days. Look forward to hearing how you get on. Let me know how I goes and I will decide about whether or not I need them back.

pascallanger, 2021-01-28:
> Tx and Rx are on the way today. Hope you will get them within 5 days. Look forward to hearing how you get on. Let me know how I goes and I will decide about whether or not I need them back.

I'm surprised, you have my address?

pascallanger, 2021-02-08:
> Tx and Rx are on the way today. Hope you will get them within 5 days. Look forward to hearing how you get on. Let me know how I goes and I will decide about whether or not I need them back.

I still haven't received anything... Do you have a tracking number by any chance?

MirlynG, 2021-02-08:
Sorry Pascal. Totally forgot to let you know. I had a problem with a customs declaration which I had to complete, now that we have left the EU! Anyway , I sorted it but it may still be a few days before you get the package, but should be there soon.
Also, cannot remember if I told you that the receiver had a problem on the rudder channel. The rudder would deflect ok but it stuck there. It would return if I moved the stick the opposite way a little bit. This happened in both directions. It would also return to center if I moved the other controls....weird! Anyway, I swapped the elevator and rudder leads on the Rx and the elevator behaved in the same way so not a servo or linkage issue. In fact I have since fitted a Lemon Rx and everything works fine. Give it a couple of days and if still not with you, I chase it up.

pascallanger, 2021-02-13:
I've received the parcel this morning and I have a bad news. Not about the parcel bu about the protocol.
It uses GFSK @1Mb but they are sending 277bytes. None of the multi RF components can do this... End of story...

MirlynG, 2021-02-13:
That's really disappointing but thankyou anyway Pascal.  Could you let me know the cost to return it to me please and I will decide whether to get it back. I paid 11euro to get it to you.

pascallanger, 2021-02-13:
Here it is:
[image]

MirlynG, 2021-02-13:
Ok. Let me have your PayPal email and I will send the cost as a friend so you get any charges. Then you can return it to me. Thanks.

MirlynG, 2021-02-13:
Sorry. "Do not" get any charges

pascallanger, 2021-02-13:
Sent on RCGroups PM.