pascallanger/DIY-Multiprotocol-TX-Module #413: Possible to modify the firmware to act as a dsmx receiver? (Just enough to get the GUID)

teozfrank, 2020-08-09:
Hi, I am currently doing some research into how spektrum dsmx protocol works over the air. I have myself a DX9 black edition along with multiple spektrum receivers (genuine). 

I want to learn more about it specifically being able to find the GUID of my own radio. Is there a way to modify this firmware to act as a receiver? For example, make it go into bind mode and bind with my tx so I can have it send the GUID and print it out via serial? 

Im guessing since the module is a transceiver it could also act as an rx instead of a tx?

pascallanger, 2020-08-10:
DSM RX is already implemented : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#DSM_RX---70

If you put the module in debug mode it will display your gid.
Pascal

closed 2020-08-10 by pascallanger

teozfrank, 2020-08-11:
How do you put it into debug mode? Is this documented somewhere? Thank you for your response!

pascallanger, 2020-08-11:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Topics.md