pascallanger/DIY-Multiprotocol-TX-Module #414: Fallos en protocolo corona V2

pancronic, 2020-08-14:
Well it seems that I have found the problem in the T16 of the channel cuts in corona v2 protocol both with internal multiprotocol module and external corona JR module
 I have a jumper R8 receiver
 With internal multiprotocol.  frsky D16 and running without interruptions in channels until the loss of the rssi return and continue perfect so the problem after.  Use more than 5 corona receivers and they all did the same, discarding the receivers, the external module and internal it seems to be a software failure in the corona protocols, so there are errors in those protocols

pascallanger, 2020-08-14:
The corona v2 protocol is working well. Whatever is not working is on your side.
Have you done the frequency tuning?
Have you upgraded your modules to the latest version?

benlye, 2020-08-14:
I've been having an interesting 'conversation' with this person on FaceBook.

He also has a problem with Corona when he is using an external Corona module in the JR bay of his T16, so whatever his issue is, it 100% is not anything to do with the MULTI-Module.  

He doesn't understand that though, and doesn't want to listen to reason.

pancronic, 2020-08-14:
If I did all fine tuning of frequency, in all of them, I have changed flex cables changed ginballs changed internal module changed sd reader also antenna, I have flashed the T16 more than 5 times and the multiprotocol module now has the last 49 and still discarded the corona receivers  since they have been tested with futaba 14mz and ct14f module and without problems, I opted to test a jumper R8 receiver and did not give any channel cut and everything was perfect,

pancronic, 2020-08-14:
To Mr. Belyle I think that the one you do not understand is you, but the world is full of heads, if there is a problem in the communication of the protocol both with the external and internal with the cpu

pancronic, 2020-08-14:
Moreover, it has been tested with another T16 pro and it gives the same problem with corona this afternoon

pascallanger, 2020-08-14:
If the problem happens with a genuine external corona module then the issue is either the T16 hardware or OpenTX.
Please open an issue there instead.

closed 2020-08-14 by pascallanger