pascallanger/DIY-Multiprotocol-TX-Module #415: XK K110 servo stuttering

PKhair, 2020-08-15:
Using IRX4+ module on release 1.3.1.49. Radio is Graupner MX22 with module running in PPM mode.  I'm getting a periodic and quite important cyclic servos stuttering every second or so, which makes the flight impossible. Binding is very consistent and fast, range looks good.

I tried everything suggested on the forum and docs, I performed frequency tuning, and tried tweaking SFHSS_DATA2_TIMING with no luck. Also tried getting back to 1.3.1.24 release or using pre-compiled binaries from https://downloads.multi-module.org/ and getting the exact same behavior. In order to rule out possible transmitter issue I tried HOTT & DSMX receivers with servos and everything just works perfectly. Looks like a specific issue with K110 receiver or SFHSS protocol. Any suggestion or help would be much appreciated.

pascallanger, 2020-08-16:
Not sure what your issue is. The k110 works perfectly fine. This is my favorite heli and flying it a lot along with the k123 and e160. Loads of people are flying helis with this  protocol with no issues.
Do you have the original k110 radio to test?
I killed once a k110 main board in a bad crash. It was still working but time to time the servos will just stop responding or go to failsafe for a split second. But it became worse with time to a point where the main motor kept spinning even with the radio off... I was able to reproduce the behavior by slightly bending the board. Vibrations were able to do this in flight.

PKhair, 2020-08-16:
Hello Pascal, it works and flies perfectly fine with the original K110 radio. I made a few checks for cracks bending an gently tapping on the PCB, but everything works reliably. I captured two short videos enclosed demonstrating the issue. One with the original X6 radio delivered with K110 (X6_radio.mp4), and one with IRX4 module (IRX4_module.mp4). Note they are zipped as github does not like video attachment.
[X6_radio.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5080643/X6_radio.zip)
[IRX4_module.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5080649/IRX4_module.zip)

pascallanger, 2020-08-16:
Hummm it's worse than what I thought you were describing earlier... It's there all the time. I've never seen this!
I'm wondering what's causing it. It's really specific to your setup.
What's the supply voltage of the module?
How is the module connected to the graupner?
What's the LEDs at the back of the module do?

PKhair, 2020-08-16:
Yeah, it's pretty bad indeed :-(

Module is inserted in the slot behind the radio in place of the legacy FM transceiver. Old Graupner/JR radios were manufactured by JR and are equiped with a standard JR slot. Supply is thru the radio, powered with 8 cell Nimh battery, voltage measured is 11.4V. The 2 leds when module operates are solid green & solid red. 

One thing which might matter,  the radio can output either PPM18 or PPM24 frames. I tried both, and behavior is the same with each configuration (looking at the PPM ISR routine it should not matter as input is clamped to 16 channels anyway).  I doubt the radio is faulty as it works fine with the same module using HOTT & DSMX receivers (smooth servo output, no stuttering).

pascallanger, 2020-08-17:
I don't have an idea of what's going on.
Things you can try:
- disable failsafe in _config.h by commenting this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/e8037c857ff4191f232ca438baa6acf5c4beb614/Multiprotocol/_Config.h#L277
- increase the signal loss delay from 70 to 1000: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/e8037c857ff4191f232ca438baa6acf5c4beb614/Multiprotocol/Multiprotocol.ino#L863

PKhair, 2020-08-18:
I did not have a chance to recompile the code with the various tweaks you suggested yet.  In the meantime, I had a chance to try the exact same module running in serial mode in a Horus X10 radio, and... everything worked perfectly with my K110 ! 

I also tried with the Horus to run the module in PPM mode, tyring 8 channels and 16 channels output. Again everything works as a charm, no stutteing, smooth servos responce. I guess there is something funky going on with my Graupner MX22 radio, might be worth probing PPM signal with a scope to see what's going on there. I believe we can close the issue.

Many thanks for your support and suggestions!

closed 2020-08-18 by pascallanger