pascallanger/DIY-Multiprotocol-TX-Module #417: Sketch too big (SMT32)

Strat00s, 2020-08-18:
Hey guys. I wanted to update my (probably) jumper clone module (JMT jp4in1) using the arduino method, as using the flash multi tool shows me an error (either error reading or error erasing the module). So I went with the arduino method. Got Arduino IDE 1.8.9, downloaded the repo -> multiprotocol.ino -> verify (just to make sure, that everything works). And here comes the problem. It says, that the sketch is too big.
`Sketch uses 122840 bytes (101%) of program storage space. Maximum is 120808 bytes.
Global variables use 5128 bytes (25%) of dynamic memory, leaving 15352 bytes for local variables. Maximum is 20480 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Multi 4-in-1 (STM32F103CB).`
So, what am I doing wrong? (also, any idea why the multi flash is giving me the errors?)
<details>
  <summary>Images</summary>

  [image]
  [image]
  [image]

</details>

kilrah, 2020-08-18:
Comment out some protocols you'll have no use for.

Strat00s, 2020-08-18:
> Comment out some protocols you'll have no use for.

Ok, no problem. But does that mean, that the SMT32's memory is no longer enough?

Strat00s, 2020-08-18:
Now it compiles, but it seems like the problem is with my module :(
The arduino says pretty much the same stuff as the multi flash. Except that it specifices, what the problem is (Module in BOOT0 mode not found on COM13).
Gotta search somewhere else then :/

pascallanger, 2020-08-18:
Open the module and do pictures of the internal to see what this new module looks like.
If it's like the other you must only plug the USB cable and not power it from the radio to be able to flash it. What are the LEDs doing when you do this?

Strat00s, 2020-08-18:
> Open the module and do pictures of the internal (close pictures of both sides) to see what this new module looks like.
> If it's like the others you must only plug the USB cable and not power it from the radio to be able to flash it. What are the LEDs doing when you do this?

Looks just like jumper 4in1, except it's a newer board revision (I guess). And I am powering it from the USB only. Only the green LED lights up that way.
<details>
  <summary>Spoiler!</summary>

[image]
[image]

</details>

pascallanger, 2020-08-18:
It should work with the LEDs as you describe...
One thing I've noticed is that the module is on port COM12 in the first picture, but the message is pointing to
"Module in BOOT0 mode not found on COM13"

Strat00s, 2020-08-18:
> It should work with the LEDs as you describe...
> One thing I've noticed is that the module is on port COM12 in the first picture, but the message is pointing to
> "Module in BOOT0 mode not found on COM13"

Yea that. Copied that when trying to use seperate serial to usb converter. It is always the same com port however

pascallanger, 2020-08-18:
I mean there is no mystery. Plugging the USB cable is setting boot0 which forces the STM32 to go in bootloader and the USB to serial adapter is interfacing with it.
The thing you can try is to connect on the PCB 3.3v and boot0 before plugging the USB cable.

Strat00s, 2020-08-18:
> I mean there is no mystery. Plugging the USB cable is setting boot0 which forces the STM32 to go in bootloader and the USB to serial adapter is interfacing with it.
> The thing you can try is to connect on the PCB 3.3v and boot0 before plugging the USB cable.

Tired that. Same outcome. :(

pascallanger, 2020-08-18:
You must be doing something wrong. Is it really the right com port? Do you have the right driver?

Strat00s, 2020-08-18:
> You must be doing something wrong. Is it really the right com port? Do you have the right driver?

I am pretty sure that it is the correct com port. When I connect the module via USB, device manager shows new device on COM12 with the CP210X adapter (which is the one on the board).
[image]
[image]

benlye, 2020-08-18:
Not sure I can think of much else to try.  Seems like there must be a hardware fault with the module.  Maybe try re-flowing the solder on the USB port pins?

If the green LED is on and the red LED is off when USB is plugged in it has power and seems to be in BOOT0 mode. If you've tried wiring the BOOT0 pin directly to 3.3V before plugging the USB cable in then it really must be in BOOT0 mode.

I'd look closely at the USB port D+ and D- pins, the CP2102, and the traces to the RX and TX pins on the STM32.

Strat00s, 2020-08-18:
> Not sure I can think of much else to try. Seems like there must be a hardware fault with the module. Maybe try re-flowing the solder on the USB port pins?
> 
> If the green LED is on and the red LED is off when USB is plugged in then it has power and seems to be in BOOT0 mode. If you've tried wiring the BOOT0 pin directly to 3.3V before plugging the USB cable in then it really must be in BOOT0 mode.
> 
> I'd look closely at the USB port D+ and D- pins, the CP2102, and the traces to the RX and TX pins on the STM32.

I tried searching for the STM32's pins, but couldn't find anything useful. Do you know, which ones are RX and TX?

benlye, 2020-08-18:
This image is accurate.  You want USART1 (PA9 and PA10) when in BOOT0 mode.
[image]

https://i.stack.imgur.com/uEky4.jpg

Strat00s, 2020-08-18:
Then it should work. Checked the usb with CP2102 and CP2102 via multimeter with continuity and everything checks out :/

benlye, 2020-08-18:
Have you tried different USB cables?  Different USB ports?  Do you have a powered hub you could plug the module in via?

Strat00s, 2020-08-18:
> Have you tried different USB cables? Different USB ports? Do you have a powered hub you could plug the module in via?

Tried different cables and USB ports. I doubt that it's gonna be power. How much does it eat, that USB3.0 isn't enough? But I can surely try it.

benlye, 2020-08-18:
Yeah, I'm just running through the list of things I can think to try.  I think you have a faulty module because 99% of the time all that's needed is plugging in the USB cable.

Can you flash it from your radio?  No idea if it has the necessary bootloader installed, but it's worth trying.

Strat00s, 2020-08-18:
> Yeah, I'm just running through the list of things I can think to try. I think you have a faulty module because 99% of the time all that's needed is plugging in the USB cable.
> 
> Can you flash it from your radio? No idea if it has the necessary bootloader installed, but it's worth trying.

External power didn't change anything. And I currently don't have a radio (waiting for a new one, which should arrive in a few days). So once I have it, I can try it, I might also try "tiny soldering" and wire external converter to test that, as there are no headers for rx and tx.

Strat00s, 2020-08-18:
Update (kinda not tho)
Tiny soldering did, once again, nothing. When I power it from the external adapter without the jumper, the red LED starts flashing rapidly and then blinks slowly. Powering it with the jumper on (boot0 - 3.3v), only green LED lights up (as it should), but arduino gives me still the exact same error.

Strat00s, 2020-08-18:
Wow, just wow. Tried it on a different PC (because why not at this point). And it works like a charm there. What the heck? What's wrong with my PC then? 😠 Everything else I have does work no problem (AVR, SAMD, ESP) but this just won't...
I guess I can close this issue now for sure (but still gonna wait for someone's response, just to make sure)

Tygrys-1, 2020-08-19:
Maybe the USB drivers for STM32?

pascallanger, 2020-08-19:
There is no USB driver for STM32. The only driver in question here is the CP2102 which must but in trouble on the laptop.

Strat00s, 2020-08-19:
> There is no USB driver for STM32. The only driver in question here is the CP2102 which must but in trouble on the laptop.

Device manager recognizes the CP2102 and I have an external programmer with the CP which does work (for arduino tho). It is quite strange.

pascallanger, 2020-08-19:
It doesn't cost anything to reinstall the latest driver as suggested already multiple times: https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

Strat00s, 2020-08-19:
> It doesn't cost anything to reinstall the latest driver as suggested already multiple times: https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

I did reinstall them multiple times. Still nothing.

closed 2020-08-23 by pascallanger