pascallanger/DIY-Multiprotocol-TX-Module #419: Hardware info on designing 4 in 1 RF Module

Rpi112, 2020-08-24:
Hi,

If I want to build the 4 in 1 RF module from scratch, is there any circuit diagram for it?

I can see that I can buy all four chips individually. Can the common inputs like SCLK, SI & SO be simply connected to each chip pins or do I need to divide them using some additional circuitry. Similarly can I simply merge the output RF signals of 4 chips or not? Do I need to have any additional components for this module (other than the 4 chips) like crystal oscillators?

I am unable to buy the 4 in 1 RF module in Banggood for some reason and want to build it by myself..

pascallanger, 2020-08-24:
Hi,
Your questions are answered on the different schematics of that page: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Module_Build_yourself_PCB.md

closed 2020-08-24 by pascallanger

Rpi112, 2020-08-24:
Hi,

In that file It says I need to buy the 4 in 1 module from Banggood... Even the STM32 board BOM says i need to buy the module from Banggood only..

I want to build this module by buying the 4 chips individually..

pascallanger, 2020-08-24:
Look at the Atmega schematic which is the same with the 4 RF modules.

Rpi112, 2020-08-24:
Got it.. Thanks..

Rpi112, 2020-08-24:
Does this mean the PE1 & PE2 outputs from STM32 board are equivalent to antenna switch control inputs (A1 & A2 on ATMEGA board) ?

pascallanger, 2020-08-24:
Yes, you can always also refer to the pins definition: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Pins.h

Rpi112, 2021-01-14:
Hi,

Just a follow on, in the image (https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/images/DIY_Mulitprotocol_Module_Schematic.jpeg), for A7105 module - GIO1 & GIO2 pins are connected to TXEN and RXEN.. But in the application note of A7105 schematic (photo attached), there are no TXEN and RXEN pins. Can I leave GIO1 & GIO2 pins without connecting to anything?
[image]

Same is the case with GDO0 & GDO2 pins in CC2500...

pascallanger, 2021-01-14:
We use GIO1 & GOI2 and GDO0 & GDO2 pins of the RF components to drive the amp TX/RX pins.

Rpi112, 2021-01-14:
HI,

I am sorry but could you expand on "amp TX/RX pins" ? I am new to this and don't know what amp is...

pascallanger, 2021-01-14:
amp=RF power amplifier. The RF chips don't have a lot of power on their output. You need a power amplifier to increase the range.
Something like https://www.skyworksinc.com/Products/Front-end-Modules/RFX2401C which has a TX/RX pins to enable transmitting and receiving at long range.
The module you have purchased might have a RF power amp. You need to know how to drive it.

Rpi112, 2021-01-14:
Got it.. Thanks for the support..

Rpi112, 2021-01-16:
Hi, I am designing 3 in 1 RF module and using the following schematic to drive the external PA & LNA (placed after RF switch PE42641).

[image]

From the code I could gather that CC2500 and A7105 registers for GPIO pins are getting changed by SetTxRxMode function. And since NRF24L01 doesn't have any such pins, I am using CE and VDD_PA (with logic lever shifter from 1.8 V to 3.3 V) instead.

Do you see any problems with the above schematic ? My only worry is, if one RF chip is working and others are in sleep/shutdown mode, can I depend on GPIO pins of these chips to be active low?
Also, hope the combination of pins I am using for driving TXEN and RXEN are correct and not swapped...

Rpi112, 2021-01-26:
Hi, 
Can you comment on the above design ?