pascallanger/DIY-Multiprotocol-TX-Module #420: Multiplex FHSS M-LINK Protocol

Josep65, 2020-08-25:
Hi There,
Is it possible to add the Multiplex FHSS M-LINK Protocol or do they use different hardware? I have various receivers and would like to use them with my new Jumper. I have looked around the forum and I have not been able to see if it has already been asked before. Thank you.

pascallanger, 2020-08-25:
I've already looked at this protocol but failed to reverse engineer it in the given time and I needed to send back all the hardware (TX+RX). The issue was mainly around the use of a pseudo random generator for the hopping frequencies...
rwijnhov is supposed to send me a TX+RX so I can try again but I haven't received anything. Without the hardware there is no chance to get it done.

Josep65, 2020-08-25:
Thank you very much for your quick reply. I can send you a receiver, but
not a transmitter, because it's broken. That is the reason why I have
multiple receivers.
If you need a receiver, do not hesitate to ask me and I will send it to
you.
Thanks again.
Josep

Missatge de pascallanger <notifications@github.com> del dia dt., 25 d’ag.
2020 a les 18:39:

weixelgeist, 2020-10-04:
Hello Pascal
I can send you a radio and a transmitter

pascallanger, 2020-11-02:
I've made some progress on reversing this protocol. I think I should be able to replicate the TX I currently have on my desk but only this specific TX. More ground work is needed to understand how to create more IDs if possible but definitively getting there.

pascallanger, 2020-11-02:
Basically they've added a 8 bits CRC at the end of the payload with a poly of 0x31 reflected in and out with no xorout.
The problem is now to determine the CRC init value which is a combination of the hopping table index and something else depending on ???.
More to come when I'll have time.

pascallanger, 2021-02-01:
M-Link is kind of working:
 - Bind works most of the time but not 100%
 - On 2 RXs, one works fine including telemetry, the second works for a couple of seconds and stops...

Josep65, 2021-02-01:
Good job Pascal,
I think there will be a lot of users hopeful with your progress.
If you need any other receiver I can send it to you.
Thank you very much for the work you are doing.

pascallanger, 2021-02-05:
The M-Link protocol is part of the test build available here: https://downloads.multi-module.org/latest-test/
Details:
- 16 channels
- Extended limits supported
- No failsafe for now but working on it
- Telemetry is "strange". I have 2 RXs RX-5 and RX-9-DR both sends completly different telemetry packets so I'm not sure how to decode it. I also don't have a TX capable of displaying telemetry neither sensors or even an understanding on how telemetry works for M-Link...
- Full ID support
- Bind works. One of the RX binds within a meter but the other needs to be 10cm or less away so put your RX really close (even nearly touching the antennas if bind doesn't work).

Please report.

pascallanger, 2021-02-05:
This protocol took Ben and me a long time to figure out. The implementation has been a real pain which has been going on for several months because some RXs where working and not others... I hope all this has been addressed now and that we will get positive reports.

If you know how the telemetry works, have sensors, an original radio that display it and an external multi module contact me!

pascallanger, 2021-02-06:
Failsafe has now been added.
**Warning**: Failsafe MUST be configured once with the desired channel values (position) while the RX is up (wait 10+sec for the RX to learn the config) and then failsafe MUST be set to RX/Receiver otherwise the servos will jitter!!!

Josep65, 2021-02-06:
Thanks Pascal. I will tray and report.

hgo-be, 2021-02-07:
Hi Pascal,  great job, I like to test the protocol on my T18 but unfortunately MLINK is excluded for the T18. No problem I downloaded the sources and changed the configuration. Unfortunately I must change validate.h, which make it more complicated to adapt new versions. I'm not familiar with the project how to implement a elegant solution ( #define T18_EXCL_MLINK ? )
Coming week I will test MLINK for a friend.

Josep65, 2021-02-07:
Hi Pascal. I have tried several receivers with success. Two RX-7, one RX-7-DR and one RX-5 Light. All of them have easily bind to 30 cm. I have also configured Failsafe as you have explained without jitter.
In all of them I received a telemetry signal (except RX-5 light, don’t has this function).
I try to connect a Vario/altimeter sensor without results, I’m not sure if its due to no signal or my poor hability getting results.
I’m really satisfied with your job. Thanks.

pascallanger, 2021-02-07:
@Josep65 if you use the latest test version v1.3.2.31, you'll be able to plug some sensors. I haven't added the vario but I've added current, tempetature, voltage and rpm.

pascallanger, 2021-02-07:
> Hi Pascal, great job, I like to test the protocol on my T18 but unfortunately MLINK is excluded for the T18. No problem I downloaded the sources and changed the configuration. Unfortunately I must [...]

I'm not sure how to deal with flash saturation. Since I've added M-Link, there is no more flash space for the normal and T18 build... So something has to give up, I'm just unsure what to do. If you have an idea...
I'll remove the validate.h back that I've just put there to enable me to do the release. I'll modify the automatic builds instead.

hgo-be, 2021-02-08:
> > Hi Pascal, great job, I like to test the protocol on my T18 but unfortunately MLINK is excluded for the T18. No problem I downloaded the sources and changed the configuration. Unfortunately I must change validate.h, which make it more complicated to adapt new versions. I'm not familiar with the project how to implement a elegant solution ( #define T18_EXCL_MLINK ? )
> > Coming week I will test MLINK for a friend.
> 
> I'm not sure how to deal with flash saturation. Since I've added M-Link, there is no more flash space for the normal and T18 build... So something has to give up, I'm just unsure what to do. If you have an idea...
> I'll remove the validate.h hack that I've just put there to enable me to do the release. I'll modify the automatic builds instead.
I like the idea to modify the automatic builds

Srelliks, 2021-02-08:
Have you seen the MPX Sensor bus configuration [here](https://www.multiplex-rc.de/service/downloads/multiplex/schnittstellenbeschreibungen.html) or is this only useful for developing sensors.

pascallanger, 2021-02-08:
> Have you seen the MPX Sensor bus configuration [here](https://www.multiplex-rc.de/service/downloads/multiplex/schnittstellenbeschreibungen.html) or is this only useful for developing sensors.

This is exactly the document I'm using to decode the telemetry: beschreibung-sensor-bus-v2.pdf

Josep65, 2021-02-08:
> @Josep65 if you use the latest test version v1.3.2.31, you'll be able to plug some sensors. I haven't added the vario but I've added current, tempetature, voltage and rpm. Let me know if it works as I have no way to test since I don't have sensors. If it works I'll add more if them.

Unfortunately I only have vario sensor. I'm trying to get some of the ones you implemented to test, but I'm not sure I can get them. I'll keep you informed.

Srelliks, 2021-02-08:
Pascal, I am in the UK so probably not practical to send you any of my sensors. You note that the telemetry is strange, one thing that occurred to me is the ability to set up a priority telemetry address on the Multiplex receivers so that the sensor value on that address is transmitted to the transmitter on a more regular basis and it maybe that this affects the way the telemetry is sent. You can check this on the Multiplex receivers through the Multiplex launcher program.

pascallanger, 2021-02-09:
Just to be straight, all I need at this very moment is someone to confirm that if one or more of current, tempetature, voltage or rpm sensors are plugged then the values are displayed somewhat correctly.
I'm only asking for a test to confirm that what I've done works so I can add everything else.

Josep65, 2021-02-09:
Sorry Pascal, I don't have these kinds of sensors. I'm trying to get some but so far I have not succeeded ... let's see if in two days I can get one.

pascallanger, 2021-02-09:
I'll add your vario tomorrow so you can test.

hgo-be, 2021-02-09:
Next week I can test a roxxy speed controller which returns voltage, current, charge, temp and rpm values

Josep65, 2021-02-09:
> I'll add your vario tomorrow so you can test.

Perfect! I Will do.

pascallanger, 2021-02-10:
[v1.3.2.35](https://downloads.multi-module.org/latest-test/) should have vario telemetry.

Josep65, 2021-02-10:
Great Pascal.  It works!! Now i can see vario sensor. Thanks.

pascallanger, 2021-02-10:
Future versions of Multi will not work unless you compile your own version with MLINK_HUB_TELEMETRY enabled. Working with @3djc to add the telem directly in OpenTX.

Josep65, 2021-02-10:
Hi Pascal, it would have been wonderful if all the options worked, but I understand that it may not always be possible. From my point of view I am satisfied since my initial request has been fulfilled and now I will be able to use my receivers. I can only congratulate you and thank you, once again, for the work you are doing.

hgo-be, 2021-02-11:
Results speed controller
Sniffer
id:0 1 78 0 Voltage:6.00               ok with ratio 12.7 ( pitty to stick on frsky ratio )
id:1 1A C8 0 LQY:100                   ok
id:2 21 F4 0 Voltage:12.20            ok with ratio 12.7
id:3 32 8 0 Current:0.40                ok
id:5 55 1C FB Rpm:6260                wrong display value -1020600rmp : if value < 0 then value = abs(value) * 10 else value = value * 100
id:6 6B C4 FF Charge:-30               sensor mAh is not discovered

pascallanger, 2021-02-11:
@hgo-be
The telemetry implementation is currently being implemented directly in OpenTX. All sensors will be decoded there (with the right decimal place and right calculations). What I've done so far was just a test to see if the sensors sent over RF were working the way I thought before touching OpenTX.
Thanks for the tests but you might want to hold for the upcoming OpenTX pull request.

3djc, 2021-02-11:
@hgo-be on what radio can you test ?

pascallanger, 2021-02-11:
He is on a T18 if I'm not mistaken.
@hgo-be you need to run the latest dev firmware if you want to test the current dev on OTX. The T18 multi test firmware now includes M-Link: https://downloads.multi-module.org/latest-test/

hgo-be, 2021-02-11:
@3djc indeed T18
@pascallanger thank you very much, I was not aware about multi raw data. I will install latest test firmware.

3djc, 2021-02-11:
You can try with this
[firmware.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5965031/firmware.zip)

It also requires multi 1.3.2.36

hgo-be, 2021-02-11:
Great job, 
Tmp1 OK
RPM   Wrong  rpm sensor value is negative , display value 3218100 , should be 5870 (see my first test )
Capa   Wrong  charge sensor value is negative for discharge, display 32675 should signed value -93
RxBt  OK
RSSI  OK
VFAS OK
Curr  0.4A

3djc, 2021-02-11:
The formula looks got in the code, but we have a conversion issue, could you try this ?
[firmware.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5966495/firmware.zip)

hgo-be, 2021-02-11:
Almost ready, 3 minors
RPM    OK
Capa    OK
I programmed a arduino for  the other sensor values
VSpd OK
Hdg   OK
Alt      OK
Fuel    OK
GRVI    OK      class 12 is Fluid, GRVI ?
Dist      BAD    class 13   factor 10 too high,  sensor value 1 should be displayed as 100 m  but shows 1000 m 
0004    class 4  Speed has no name, precision should be 0.0

3djc, 2021-02-11:
What can we use for fluid using 4 letters ?

[firmware.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5967027/firmware.zip)

hgo-be, 2021-02-11:
Super, all values are correctly visualized.
Fluid : Cons (umption)  ? No idea

3djc, 2021-02-11:
Fuel :)

[firmware.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5967294/firmware.zip)

hgo-be, 2021-02-11:
There is already a value Fuel (level) ?
firmware file is the same as the previous ?

Srelliks, 2021-02-11:
Use Tank for fuel level and Fuel for fuel flow?

3djc, 2021-02-12:
I went for Flow

[firmware.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5969962/firmware.zip)

Can anyone confirm that when using GPS, you don't get GPS coordinates on the radio ?

Srelliks, 2021-02-14:
On Multiplex radios it's not possible to display the GPS co-ordinates. However the SM Modelbau GPS logger does send them as alternate values as described in the manual "In M-Link after 2 minutes without moving, Latitude and Longitude are displayed on the GPS-Logger 2 on addresses Vario and Speed every 5 seconds. The display changes between the decimal point value with the unit "mAh" and the decimal value with the unit "ml". More meaningful units – such as degrees and minutes – are unfortunately not allowed in M-Link." I don't have a Multiplex GPS sensorbut I am pretty sure the Multiplex GPS sensors do not send the GPS co-ordinates.

jmdfs, 2021-02-28:
bonjour

Je viens de parcourir les informations sur le module 4in1 de chez Banggood en STM32 et le Mlink Multiplex, c'est génial. J'ai le module et un ancienne radio multiplex que je souhaite passer en Mlink via le module avec une entrée en PPM.

Il y a un point que je ne comprends pas. Sur la liste des protocoles disponibles, Mlink est en numéro 78 et si j'ai bien compris il n'y a que 14x5 soit 70 protocole possible, comment avoir un protocole supérieur à 70 (78 pour moi) ?

D'avance merci et désolé si la question semble évidente mais je débute.
Bonne journée

pascallanger, 2021-02-28:
@jmdfs Ce n'est pas du tout comme ça que ça fonctionne... Pour le mode PPM il faut compiler soit même le code car il faut configurer le module par l'intermédiaire du fichier _config.h. Dans ce fichier config, on indique quel protocole on associe à quel position du dial: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/f19bb05c7ab971d0677187abe93110b2a7855fb4/Multiprotocol/_Config.h#L412

jmdfs, 2021-02-28:
Bonsoir Pascal

Merci pour les informations, j'ai avancé un peu et je commence à comprendre le mécanisme => ide Arduino sur mon ordi + téléchargement du DIY-Multiprotocol-TX-Module-master + adaptation du Config.h (Là il faut bien comprendre et prendre son temps)  La suite un peu plus tard (Compilation et injection dans le module) pour ensuite essayer.

Deux autres questions:
1) Les fichiers Bin téléchargeables sur la page "downloads.multi-module" ne sont donc utilisables que pour le mode série ?
2) Le liens sur la rubrique "Download the Multiprotocol source and open the project"
vers le fichier source. zip est toujours à la bonne version, la dernière version stable ou autre ?

D'avance un grand merci pour les explications et surtout le travail fourni
Jean Michel

pascallanger, 2021-02-28:
1. Exact. Les versions PPM sont juste pour test.
2. Pour une version stable il faut télécharger le source depuis les releases.

closed 2021-03-02 by pascallanger