pascallanger/DIY-Multiprotocol-TX-Module #421: Help changing channel order for one Protocol/Sub Protocol only

HippoDan, 2020-08-25:
I've got a special Micro Flysky receiver that has the ESCs built in, and the channels can't be changed.  I'd like to make a new protocol for that model that has the channel order modified.  I can't see where in the code the channels are specified per protocol.

Any help would be appreciated.  Thanks.

pascallanger, 2020-08-26:
Which radio do you have?

HippoDan, 2020-08-26:
I'm using a GT3B surface radio for this project.  If I was using one of my airplane radios I would just rearrange the outputs

pascallanger, 2020-08-27:
All you have to do is to make achange change to this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/4b4393952e9ba0b99b7e2bffa5a4b29273b40ae1/Multiprotocol/FlySky_a7105.ino#L147
You can use an arbitrary sub_protocol 7.
Something like:
uint8_t CH_SPECIAL[]={0,1,2,3,4,5,6,7}; // reorganizing the numbers will change the output channel order
uint16_t temp;
if(sub_protocol!=7)
  temp=convert_channel_ppm(CH_AETR[i]);
else
  temp=convert_channel_ppm(CH_SPECIAL[i]);
Pascal

closed 2020-08-27 by pascallanger