pascallanger/DIY-Multiprotocol-TX-Module #422: FlySky AFHDS2A: problems using together with FS-i6s

wimalopaan, 2020-08-31:
Following setup gives reproducible telemetry problems:

1. radio: Radiomaster TX16s with internal module
2. radio: FlySky FS-i6s
both receivers: FS-ia6B

The RX are at the same position. Moving both TX to a distant location results in loss of telemetry at the very same point on both TX. This is reproducible. Using only one TX (anyone) no problem arises.

pascallanger, 2020-08-31:
Sounds to me that Multi is doing the same as Flysky from your description which is good.
It looks like that the TXs are swamping each others.
Not sure I can do anything... Have you opened a case with Flysky since it's also affecting their TX?

wimalopaan, 2020-08-31:
Do you have teh HW to reproduces that?

Will check with FlySky also.

pascallanger, 2020-08-31:
I don't have any Flysky radio.
What you can try to do is to put Multi in low power mode, I would expect the FS-i6s to recover telemetry.

wimalopaan, 2020-08-31:
But this results in very low possible range for the RM?

Can this be an issue with some sort of EU-LBT?

pascallanger, 2020-08-31:
The test I was asking would show that the issue is swamping.
There is no LBT on this protocol. In fact you are not losing the control of the RX which shows that theTX to RX works. It's RX to TX.

MartyHugh, 2020-08-31:
Sounds like your two RXs are too close together.  They are both transmitting telemetry and are talking over each other.  What happens when you separate them by 12 inches ?

wimalopaan, 2020-09-07:
The same effect arises with two RadioMaster with internal MPM and two FlySky FS-iA6b receiver. The receivers are seperated by more than 3 meters. The TX are seperated by 2-4 meters.

What tests should we perform to help with this issue?

pascallanger, 2020-09-07:
The tests in asked for may be?

pascallanger, 2020-09-07:
And again you are taking about 2 multi modules. What about 2 genuine flysky radios. Are they doing any better?

wimalopaan, 2020-09-20:
Looks like duplicate of #398

pascallanger, 2020-09-20:
This has nothing to do with it but since you think it is related I'm closing this issue
Note that you still haven't replied to any of my questions to pinpoint your issue since 21 days...

closed 2020-09-20 by pascallanger

wimalopaan, 2020-09-21:
I will come back with tests, but this will take time until the next weekend.

Is it sufficient to do the test with to MPM, one with reduced power?

wimalopaan, 2020-09-24:
Please help use with the needed test procedure (see post above). 

What tests and what infos do you need?