pascallanger/DIY-Multiprotocol-TX-Module #423: A support for FMS receivers possible?

evildronepilot, 2020-09-05:
There's a wide range of FMS models that come with the receivers... 
Does the support for those exist or can it be implemented please?
Happy to offer testing.

pascallanger, 2020-09-06:
Which RF component in the TX/RX?
Are you able to do SPI dumps? If not will you send me a TX and RX?

pascallanger, 2020-09-14:
No answer?

pascallanger, 2020-09-21:
The FMS Ranger 850mm aircraft is using the FrSky D8 protocol.
Could it be that this is what you are looking for?

evildronepilot, 2020-09-22:
Hi guys sorry for leaving you hanging for so long.
I ordered the FMS P51D 1.4m  and it's supposed to come with a receiver... unfortunately this is not longer the case so can't really help with testing it :( 
I still believe this might become useful future for someone who is using FMS receivers ?

pascallanger, 2020-09-22:
I'm closing this issue, since without a TX and RX to work on there is nothing to be done...

closed 2020-09-22 by pascallanger