pascallanger/DIY-Multiprotocol-TX-Module #424: Could SBUS/EX BUS be supported as a serial input to avoid PPM latency on a non-OpenTX radio?

nickthenorse, 2020-09-12:
Currently using the multiprotocol module with a PPM output from one of my Jeti receivers. It's fully operational like this, and you guys are doing a stellar job developing and supporting the firmware.

I can reconfigure my Jeti receiver to output SBUS or EX Bus on the same pin as it currently is outputting PPM.

Would it be possible to instead use this serial output to feed the multiprotocol module, as a way to reduce the latency? Ideally the module would continue operating in the "PPM-style" of using the bind button and using the dial to select the active protocol. I'm not concerned about telemetry or bidirectional communication, just want to use a serial connection to feed the module the control stick outputs.

pascallanger, 2020-09-12:
You are talking about latency and you are connecting the module behind a RX... Doesn't it ring a bell somewhere?
I would be ok to look at it if there was an sbus output directly from the radio but behind a RX...

nickthenorse, 2020-09-12:
Ok, I see. Thanks for a quick reply.

So you don't think that an sbus connection from Rx --> Module will still be faster than a PPM connection directly Tx --> Module?

I've contacted Jeti to see if they are willing to enable sbus/exbus directly from the internal data header on the Tx motherboard.

pascallanger, 2020-09-12:
Unless they add sbus directly in the radio you won't have any latency gain. Let's see if they will reply to request.

nickthenorse, 2020-09-24:
Jeti replied, they said they have no plans to support MPM or Crossfire.

Funnily enough that is not what I asked them about. I asked them about adding direct UDI/S.BUS/Ex Bus output from the transmitter. I said I was willing to pay part of their firmware development costs.

By their reply I guess they are afraid of this hurting first party Jeti Rx sales. Which is silly, if you ask me, but perhaps understandable from a pure business $$$ point of view.

Closing this issue, but thanks anyway Pascal!

closed 2020-09-24 by nickthenorse