pascallanger/DIY-Multiprotocol-TX-Module #425: Loli Protocol

ss8688, 2020-09-12:
Loli is a open source TX system. Both hardware and software. deviation has already supported that protocol. (please see https://github.com/DeviationTX/deviation/blob/master/src/protocol/loli_nrf24l01.c ),I hope DIY-Multiprotocol-TX-Module can support it too. So，I can make recivers myself. 

Hear is a image of loli project :https://github.com/wooddoor/Loli3

pascallanger, 2020-09-12:
Would it be possible to send me a RX so I can develop and debug the TX protocol?

ss8688, 2020-09-12:
Sorry，I don’t have one now

ss8688, 2020-09-12:
You can make one youself，It is very cheap.

ss8688, 2020-09-12:
You can join the official qq group to get more PCB files of Loli TX and recivers.

pascallanger, 2020-09-13:
Sorry I don't have the time to build one.
You don't even have one so you won't be able to test if what I code is working. And we won't know in case or issues if it's the TX or RX which is at fault.
Sorry but given the conditions I'm not even going to start looking at it...

pascallanger, 2020-09-14:
Is there anyone with a Loloi RX listening?

OpenUAS, 2020-10-03:
@pascallanger If I come across one of those LOLI receivers, I'll donate you one. For time being, code and schematic:

[LOLI_code_n_schmatic.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5322408/LOLI_code_n_schmatic.zip)

LoLi ? Lata in Lata out ;)

diego0815, 2020-10-11:
I actually would have some nrf modules lying around and would also be ready to try, but why bother wirth an absolutely unknown breed of 8051 in 2020, while lots of other MCUs are waiting in my parts box? Sounds like DOA in Europe...

q296291520, 2020-12-02:
Pascallanger ,I can send LOLI3 receivers, please send me the address privately, thank you!      296291520@qq.com

q296291520, 2020-12-02:
> 可以给我发送RX，以便我可以开发和调试TX协议吗？

Pascallanger ,I can send LOLI3 receivers, please send me the address privately, thank you! 296291520@qq.com

richardclli, 2020-12-17:
Seems Loli3 is an interesting RX that can store model setup in the RX, configuring the RX used similar concept of DSM forward programming.

diego0815, 2020-12-17:
Interesting would be a port of the Loli SW to some standard NRF52840 module, plenty of range for smaller copters, cars, ships. Readily available, cheap(ish) modules, various antenna types, lots of cpu power also to integrate a flight controller (baseboard with 6axis) or other fancy stuff.

pascallanger, 2020-12-17:
> Seems Loli3 is an interesting RX that can store model setup in the RX, configuring the RX used similar concept of DSM forward programming.
> 
> Failsafe concept is similar to WFLY ET07, in which each channel can have a specific failsafe setup.

You are not doing wrong comparaisons...

q296291520, 2021-01-06:
Hi, Pascallanger, have you received the express?&nbsp;&nbsp;&nbsp;If you need any help with LoliRX, please contact me.

pascallanger, 2021-01-06:
Still not received yet... I'll let you know. Pascal

q296291520, 2021-01-06:
http://www.ems.com.cn/ems/order/singleQuery_e  

Tracking Number：LZ541378714CN

2020-12-12  14:59:46 ShanghaiCity  Package Received   
2020-12-12  17:14:16 ShanghaiCity  Depature from Local Sorting Center   
2020-12-12  18:11:22 ShanghaiCity  Arrival at Regional Sorting Center   
2020-12-13  15:49:24 ShanghaiCity  Dispatched from Office of Exchange   
2020-12-14  03:40:51 ShanghaiCity  Departure from Regional Sorting Center   
2020-12-14  13:41:48 ShanghaiCity  Handed over to Forwarding Agent for Transit   
2020-12-14  13:47:00   Received by the carrier   
2020-12-14  13:47:00   Received by the carrier   
2020-12-16  11:16:00   Flight Departure   
2020-12-16  19:12:00   Flight Arrival   
2020-12-19  20:11:00   Transiting via the transit Airport   
2020-12-22  19:30:00   Flight Departure   
2020-12-23  10:05:00   Flight Arrival   
2020-12-23  10:39:00   Transiting via the transit Airport   
2020-12-24  04:23:00   Arrival at Processing Center   
2020-12-24  04:26:00   Arrival at Destination   
2020-12-25  03:49:00 FR  Departure from International Sorting Center   
2020-12-25  03:49:00   Customs Clearance Completed   
2021-01-06  09:04:00 FR  Scheduled for Delivery   
2021-01-06  09:04:00 FR  Delivered  

 

ok，Pascal，I've seen the track and it shows delivered. Please pay attention. Thank you.      ^_^

pascallanger, 2021-01-06:
On then it should be in the mailbox. I'll know this evening.

pascallanger, 2021-01-06:
I have the 5 RXs! They all have different stickers on the bags. And some have a different pcb.
But are they really different? If yes how?

q296291520, 2021-01-07:
HI pascall 

They are all loli3 protocols with the same firmware. These are receivers DIY by my friends and I. The stickers on the bags has nothing to do with the receiver. Short circuit 5 / 6 channel bind(one is 7 / 8 channel). You can test them with devo10 (deviation loli protocol) first, and the voltage supports 1s-5v. A white receiver test is recommended. ^_^

q296291520, 2021-01-08:
If you need any other help about loli3rx, please contact me. Thank you.

pascallanger, 2021-01-09:
I've added the LOLI protocol to in v1.3.1.97
Documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#loli---82
Assigning functions to outputs is not supported yet. I'll work on a LUA script but first OpenTX has to be updated...

q296291520, 2021-01-09:
Thank you very much! If you have any help with LoLI3 RX, please contact me.^_^

pascallanger, 2021-01-09:
One question, the RX returns 2 analog values. Where can I connect something to get a value sent?

q296291520, 2021-01-10:
You mean return voltag
[image]
e? Please see the picture.

pascallanger, 2021-01-10:
What voltage can be applied on it? 5V max out you need a divider? Or...?
What do you mean by mistakes? Is it safe to apply voltage on it?
I don't have a devo to test...

q296291520, 2021-01-10:
Max 3S ,&nbsp;The split voltage resistance may be reversed and the voltage display is wrong. It is recommended to use the blue test.

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "pascallanger"<notifications@github.com&gt;; 
发送时间: 2021年1月10日(星期天) 下午4:38
收件人: "pascallanger/DIY-Multiprotocol-TX-Module"<DIY-Multiprotocol-TX-Module@noreply.github.com&gt;; 
抄送: "L"<296291520@qq.com&gt;; "Comment"<comment@noreply.github.com&gt;; 
主题: Re: [pascallanger/DIY-Multiprotocol-TX-Module] Loli Protocol (#425)

什么电压可以施加在上面？5V最大输出你需要一个分频器？还是？
你说的错误是什么意思，在上面加电压安全吗？
我没有Devo可以测试...

—
你收到这个是因为你评论了。
直接回复此邮件，在GitHub上查看，或取消订阅.

q296291520, 2021-01-10:
Return power voltage port&nbsp;Max 3S , The split voltage resistance may be reversed and the voltage display is wrong. It is recommended to use the blue test.

------------------ 原始邮件 ------------------
发件人: "pascallanger"<notifications@github.com&gt;; 
发送时间: 2021年1月10日(星期天) 下午4:38
收件人: "pascallanger/DIY-Multiprotocol-TX-Module"<DIY-Multiprotocol-TX-Module@noreply.github.com&gt;; 
抄送: "L"<296291520@qq.com&gt;; "Comment"<comment@noreply.github.com&gt;; 
主题: Re: [pascallanger/DIY-Multiprotocol-TX-Module] Loli Protocol (#425)

什么电压可以施加在上面？5V最大输出你需要一个分频器？还是？
你说的错误是什么意思，在上面加电压安全吗？
我没有Devo可以测试...

—
你收到这个是因为你评论了。
直接回复此邮件，在GitHub上查看，或取消订阅.

stanekTM, 2021-01-10:
Hi,
according to the diagram, it looks like only one analog input should be used

[https://github.com/wooddoor/Loli3/blob/master/%E7%A1%AC%E4%BB%B6%E7%94%B5%E8%B7%AF/%E5%88%B6%E4%BD%9C%E6%95%99%E7%A8%8B/%E7%BD%97%E4%B8%BD%E4%B8%89%E4%BB%A3%E9%81%A5%E6%8E%A7%E5%99%A8%E5%88%B6%E4%BD%9C%E6%8C%87%E5%AF%BC%E5%8F%8A%E4%BD%BF%E7%94%A8%E6%96%B9%E6%B3%95.pdf](url)

q296291520, 2021-01-11:
> Hi,
> according to the diagram, it looks like only one analog input should be used.
> I'm confused from connecting a VCC 5V directly to a processor where there is no pre-regulator.
> The voltage of the divider will correspond to 3S lipo approx. 15V.
> 
> [rx_loli.pdf](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5792566/rx_loli.pdf)

这是a8的原理图，发个a10的

pascallanger, 2021-01-11:
There are 2 ways to do the different channels configuration:
1. write a LUA script where you select what you want for each output and send this to the RX
   - plus: easy to see what you are doing, sends the config when you want
   - minus: the script does not reflect what the current config of the RX (as it is not sent), so you have to reconfigure all the outputs each time you launch the script
2. use channels 9-16 to configure the function outputs of 1-8
   - plus: the RX config is stored in the radio so you don't have to redo it every time.
   - minus: it's not easy to see right away what the outputs are doing.

Option 2 looks good to me. I could write a LUA script which would display the current config and set the 9-16 outputs to the correct values for you so you don't have to think about which value to put where.
Let me know what you think.

q296291520, 2021-01-11:
I think the second option is better.

use channels 9-16 to configure the function outputs of 1-8
plus: the RX config is stored in the radio so you don't have to redo it every time.

minus: it's not easy to see right away what the outputs are doing.&nbsp; &nbsp; &nbsp; &nbsp; &nbsp;&nbsp;

q296291520, 2021-01-11:
> here are 2 ways to do the different channels configuration:
>
> 1. write a LUA script where you select what you want for each output and send this to the RX
>
>    * plus: easy to see what you are [...]

I think the second option is better.

pascallanger, 2021-01-11:
Thanks, I'll code that today.

stanekTM, 2021-01-11:
> 这 是 a8 的 原理 图 ， 发 个 a10 的
I see,
thank you, this is what i found on github, there are probably more suggestions.

pascallanger, 2021-01-11:
Check the implementation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#loli---82
This is available in the latest test build: https://downloads.multi-module.org/latest-test/
I'll try to write a lua script tomorrow but this is already fully functional.

q296291520, 2021-01-13:
> Check the implementation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#loli---82
> This is available in the latest test build: https://downloads.multi-module.org/latest-test/
> I'll try to write a lua script tomorrow but this is already fully functional.

Thank you for your beta, because I'm not very familiar with opentx system, so I just test 8 channels and use them normally. I've sent it to my friends for testing.^_^

pascallanger, 2021-01-18:
I've uploaded the LOLI RX configuration script: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts#multiloli
You can see a video of it in action here: [image]

Unless there is a bug, I think I'm done with this protocol. Enjoy!

closed 2021-01-18 by pascallanger