pascallanger/DIY-Multiprotocol-TX-Module #427: After power on the red led is flashing, but the module TMX5 Lite flashes program ok.

Tygrys-1, 2020-09-19:
Red led is flashing (do not look like SOS but I think it is it). The chip has 128 MB flash, as it programing ok with the firmware, but it looks less because of STM32 version. So the flash size does not test the real flash size.

Status in model settings  say now "no telemetry", and not the version of the multimodule as before

Open TX 2.3.9, Taranis X9 Lite.

pascallanger, 2020-09-19:
Open your module and check the STM32 if it contains C8. Then it's is a 64KB module which you are trying to use as 128KB...
5Or tooling was not checking the flash size unless you are now using flash multi 0.5.0.

kilrah, 2020-09-19:
Seems like you might have flashed the bootloader but no firmware, blinking at constant speed means it's in the bootloader.

Tygrys-1, 2020-09-19:
@Pascal It is C8, it is suposed to have 128KB flash. Otherwise it should not flash and run multi, as I think?
@Kilrah - not a constant speed. It is a series - - .. long .. and repeat. I think it is this SOS Pascal announced.

benlye, 2020-09-19:
The C8 officially supports 64KB of flash.  

[image]
While some have been known to have 128KB which is readable and writable, some will let you write 128KB but only read 64KB, but none officially have 128KB.

Many people with modules which have C8 parts have reported erratic or unreliable behavior.  That's why these modules are no longer supported.

Tygrys-1, 2020-09-19:
@benlye I know this STM32 "feature" :-)
As I might understand this decision from the develper POV, I am the user with C8 and 128KB working flash. And this change made my module locked on 3.1.59 just because flash checking algorithm does not chceck the real flash size and condition, but made the assumption based on the CPU type and "official" datasheet.

I think, that this "not supported modules" might have been "not supported" in more delicate way, other than just refusing to work when certain CPU type is detected.

benlye, 2020-09-19:
You still have options. You can download the source, change one option to disable the check, then compile and upload. Even in Flash Multi (if you use it) you can disable the check.

Tygrys-1, 2020-09-19:
@benlye I know I can. I think that DISABLE_FLASH_SIZE_CHECK will do. And it should be added to Config.h and documented.

As to "reason STM do this" is known and simple - it is cheeper to do one silicon mask instead of two, and then mark the chip as market needed. Of course there is a margin for errors and C8 chips might have defects in "non oficial" areas, but as the production matures this can be a rare case. We know NVidia did that to their GPU's, where by the BIOS upgrade lucky user might enable "locked" shaders, with the high rate of success.

Reason why the URUAV and others use such chips is obvious - they are cheeper. And maybe they test them for being able to use "out of spec features", as they base their production on them.  ;-)

pascallanger, 2020-09-19:
> @benlye I know this STM32 "feature" :-)
> As I might understand this decision from the develper POV, I am the user with C8 and 128KB working flash. And this change made my module locked on 3.1.59 just [...]

This is in no way a check based on a CPU type, check the code. The test is based on the reported flash size from the component itself. If you had a C8 reporting 128KB everything would work.
Instead of blaming us for a manufacturer which has not followed the open source schematic, contact them and ask them to support or replace your module. They are the one responsible for this mess not us.
We are in fact trying to protect your models of bad crashes due to an unexpected flash failure.

Tygrys-1, 2020-09-19:
@pascallanger I said earlier, beeing a dev myself I understand the dev's point of view.
I also understand the safety you implement. And I do not blame anyone. 

But OTOH I have 128KB of flash and the "Test the flash size" is wrong stating, that I have not. So maybe a rephrase of this statement will help.

I think, that a big warning should be added to release notes - that after the update to this version the module might stop to work. It only states, that it will blink some pattern, which does not look harmfull. :)

pascallanger, 2020-09-19:
> @pascallanger As I said earlier, being a dev myself I understand the dev's point of view.
> I also understand the safety you implement. And I do not blame anyone.
> 
> But OTOH I have 128KB of flash and "Test the flash size" is wrong - stating, that I have not. So maybe a rephrase of this statement will help.
> 
> I think, that a big warning should be added to the release notes - that after the update to this version the module might stop to work. It only states, that it will blink some pattern, which does not look harmful. :)

I think you do not have to support all the people having issues with their modules doing weird stuff...
I also think that you are kidding when you say that they are testing their products... If they would have tested anything they would have seen that they have even fitted the wrong inverter component preventing to actually invert the telemetry... Sadly, the CPU is not the only problem of these modules...
You are also doing a blank statement by saying that your CPU has 128KB fully working. Have you at least done some write/verification tests under different temperatures? Why do we find then perfectly working C8 reporting the right amount of flash 128KB and other C8 which are causing issues and reporting 64KB of flash?
Based on your suggestions:
- I've modified the release note and the troubleshooting page is also updated.
- in the next release and as a warning, I will allow the module to boot after 30sec/1min so that people who are sure they have the right flash will still can be able to use the module after the warning period.
Pascal

kilrah, 2020-09-19:
>  Of course there is a margin for errors and C8 chips might have defects in "non oficial" areas, but as the production matures this can be a rare case. We know NVidia did that to their GPU's, where by the BIOS upgrade lucky user might enable "locked" shaders

There have been multiple reported cases of misbehavior with wrong chips that did flash 128K correctly, that's the whole reason for the change. You don't risk hurting someone or damaging property with a graphics card, but with a model aircraft you do. People should NOT use modules with the wrong chip, and it should absolutely not be made easier to bypass the check as that would encourage people to do so.

Tygrys-1, 2020-09-19:
I think the changes you made to the description are right. Thank you.
I'm not sure about that delay. You did good with the safety. This should force manufacturers to keep to the rules and design.
Just document DISABLE_FLASH_SIZE_CHECK and set the responsibility for this to end users.

Tygrys-1, 2020-09-19:
> There have been multiple reported cases of misbehavior with wrong chips that did flash 128K correctly, that's the whole reason for the change. You don't risk hurting someone or damaging property [...]

You are right. But it is not obvious for normal user which chip is in his module. And it is not his fault, that the manufacturer is doing poor design and parts choice. In the end he lands with inoperable module after the upgrade.
I do not complain about the fact, that you are not supporting bad hardware. It is a good decision for the reason you pointed out. Only the way it was introduced is problematic for me. There should be the warning on update or even earlier announcement. For people not TRY to update to the nonworking version.

Thanks all MultiModule devs for their great work.

closed 2020-09-19 by Tygrys-1