pascallanger/DIY-Multiprotocol-TX-Module #428: Issues when flashing the MPM with/in a X9D+ SE

fiechr, 2020-09-20:
I use a IRX4+ MPM with a X9D+ SE and a internal MPM in a Jumper T16. Both with OpenTX 2.3.9.

Since I use Linux, I used the shell script provided by Flash Multi to upgrade the bootloader.

After that, I tried to flash the module again with the latest firmware in my X9D+, which I've done successfully before.

But there seem to be two issues: First, the X9D+ seems to have troubles displaying the original filename of the bin file. Although the *.frk files show up fine, even with long filenames (> 8 characters), the *.bin files do not, or at least not always (that's the weird part).
Only after renaming it to something like `mpmfw.bin` it shows up for sure. But now -- and this is new -- I can't select them anymore to flash the (internal/)external module anymore. These two options (flash internal/external) simply don't show up anymore in the context menu. They do for the *.frk files, though. But I know I've flashed the module multiple times on this device with the same OpenTX version (2.3.9).

I then put the module in my T16 and flashed it there, which worked flawlessly. It works now fine again back in the X9D+.

I know this maybe an OpenTX issue, but still wanted to mention it, because others may have the same problem and don't have another TX to flash the module.

pascallanger, 2020-09-21:
As you mention this is more something for the OpenTX team to check. I'll add a comment that for the X9D+ you need to shorten the file under 8 characters for now.

closed 2020-09-21 by pascallanger

fiechr, 2020-09-25:
FYI, the limit for the X9D seems to be 32 characters (depending on the the LCD), longer filenames are simply ignored:
https://github.com/opentx/opentx/issues/7966