pascallanger/DIY-Multiprotocol-TX-Module #429: ATmega328p powering and frequency

DEMONmachine, 2020-09-21:
I'am building TX module from scratch using arduino.
And I'am a bit confused about selected parts and wiring diagram. There is used Arduino Pro Mini 16MHz. But it is powered by 3.3v regulator. If I understanding it right 328P is not guaranted to run 16MHz at 3.3v. Isn't it a mistake? And which one should I use? Pro Mini 3.3v/8MHz or 5v/16MHz?
If I don't have Pro Mini 5v...
- Can I use regular Pro Mini 3.3v/8MHz?
- Can I use Arduino Nano (328P/5v/16MHz) instead? 
- Or I have to use Pro Mini 3.3v with replacing resonator from 8MHz to 16MHz?

Thank you in advance.

schwabe, 2020-09-21:
No, the chip is run out of spec. But it is recommended to build modules around the STM32 nowadays anyway. The AVR chip is a bit too small.

stanekTM, 2020-09-21:
Use 328P / 5V. The 3.3V is out of specification but works well. I built several Multimodules by the original author "Midelic". Due to the size of the flash, select only the protocols you need. Use the Arduino IDE to compile. There is a lot of information at the beginning of the Multiprotocol thread
[https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module](url)

pascallanger, 2020-10-01:
I think questions have been answered, closing the thread.

closed 2020-10-01 by pascallanger