pascallanger/DIY-Multiprotocol-TX-Module #430: Flysky AFHDS2A Vario

pascallanger, 2020-09-21:
Hi Pascall,
Sorry for the confusing message.
i and some friends are very grateful for the work you have done on the
multiProtocol Modules.
We fly gliders and have started using FlySky receivers and the altitude
Attachment FS-CAT01.
The latest open source firmware for the FS-i6 Tx is the FlyPlus

https://github.com/qba667/FlySkyI6/releases/tag/1.7.6

This upgrade allows the vario to work correctly and would be a great
addition to our RM TX16s transmitters
regards
Tony McPhee

On Sat, Sep 19, 2020 at 4:39 AM pascallanger <notifications@github.com>
wrote:

> No idea what flyplus is, neither rm16s...
>
> —
> You are receiving this because you commented.
> Reply to this email directly, view it on GitHub
> <https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/398#issuecomment-695025073>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/ABHLG3QKYY2X35D6UGAEMRDSGOSPNANCNFSM4PE35OCQ>
> .
>

-- 
Regards
Tony

_Originally posted by @UncleTone in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/398#issuecomment-695996466_

pascallanger, 2020-09-21:
Ok now I understand what you mean. You are looking for the altitude/barometer sensor to be used as a vario.
This is not something that would be done in Multi but OpenTX instead.
@3djc is this something OpenTX supports?

3djc, 2020-09-24:
No, we do not want to support this since telemetry is too slow to make a proper vario derived of alt.

If someone really wants this feature, it can easily be done using LUA in OpenTX

closed 2020-09-25 by pascallanger