pascallanger/DIY-Multiprotocol-TX-Module #431: Sensor Lost with Crossover / Oversky XR602T-F2 (Frsky D16) receiver

TedAus, 2020-09-23:
I bought a Crossover (a.k.a. Oversky) XR602T-F2 receiver which uses Frsky D16 protocol to try. It promises the benefits of easier binding (no need to hold down the bind button when powering up), no failsafe when close to the transmitter, and uninverted FPort.

All the features above work, but I'm getting frequent sensor lost (multiple times per battery pack) and when the values return, the battery voltage, for example, sometimes goes to negative 15,000!

I'm using multi-protocol version v1.3.1.59 with OpenTX 2.3.9 on a Jumper T16 Pro.

Could this be due to bad receiver firmware? I've tried contacting the manufacturer, but didn't get a response (unsurprising).

Thanks for your ongoing work on this module. Much appreciated.

pascallanger, 2020-09-23:
The D16 protocol has been out for a very long time with no reported issues. I would therefore point to a problem with the RX firmware...

TedAus, 2020-09-23:
That's what I thought. Hopefully they'll come out with a new firmware soon. 

Thanks Pascal!

closed 2020-09-23 by TedAus