pascallanger/DIY-Multiprotocol-TX-Module #434: Radiomaster TX16S with Flysky X6B - constant telemetry loss/recover on distance <4m

Tygrys-1, 2020-09-26:
When TX and RX are close,  there is constant telemetry loss/recover message. I have to move the TX about 4m from the receiver for this to stop occuring. I assume the telemetry signal to strong, but maybe this can be tuned somehow?

pascallanger, 2020-09-27:
This X6B RX is troublesome... Everyone is reporting strange things about it...
The messages are only telemetry lost and recover or you get something else?
The RX is swapping the TX input from what you are saying. But the weird thing is that it is not happening with any other RXs. So either this one has a stronger signal than all the ohers or it doesn't have a power adaptation based on the proximity of the radio like the others do.
There might be something feasible, like detecting high RSSI and turning down the agc for example but without this RX in my hands it will be hard to find the right settings. May be you can give it a try as you said you are a dev?

Tygrys-1, 2020-09-27:
The messages are just this. After the telemetry loss there is no RSSI indication on the TX screen, then after a second or a few seconds there is the telemetry recovered and 5 bars. And then it does this again. Sometimes Amber is constatnly talking.
I'm a dev in different brand and enviroment. :-) If you point me into the right place in repository, I may look at the code. But not sure I'll be able to find the solution. I have no knowledge of how radio chips works inside. I may however test settings you suggest. I will try test with another TX, the X9 Lite with TMX5 Lite (yea, that one with C8) to see if it is somehow hardware or TX related.

pascallanger, 2020-09-27:
Are you sure your internal module is off. Check the other issue already opened for this exact RX.

Tygrys-1, 2020-09-27:
It is off, for sure. I'll test other receivers and transmitters to find the pattern. It will take some time.

Tygrys-1, 2020-09-27:
Tested Radiomaster TX16S with iA6B receiver - everyting is ok, no telemetry loss even when antennas touch each other.

pascallanger, 2020-09-27:
Yes there problem is only with this specific RX no others.

Tygrys-1, 2020-09-27:
Tested another X6B, everything ok. I'm getting confused now. Will have to test problematic rx with the different transmiter.

pascallanger, 2020-09-27:
Different firmwares?

Tygrys-1, 2020-09-27:
Have no idea. I do not know how to check firmware version on X6B. There is the Update button, but I do not know how to update X6B.

pascallanger, 2020-10-01:
This issue is a duplicate with this one: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/398

Tygrys-1, 2020-10-01:
The original issue in #398 is not about the telemetry, but I think that the issue has the same source.

I tested with Taranis X9 Lite amd TMX5 Lite module on 3.1. 59 and the issue is the same - or even more visible. The "save distance" seem to be more 8m and 2 walls between :-)

I tested another X6B and it is ok with Taranis X9 Lite and TMX5 Lite module, got "Telemetry Lost" (solid) when antennas are about 1cm apart,  when 10 cm it is working ok. So I think it is dependant on RX6B version and I need to test if the problem could be solved by firmware update on the receiver.

I found the way to read the firmware from the receiver, so I'll try to compare versions, as I think the other two X6B's I have are not having such problems.

pascallanger, 2020-10-01:
> The original issue in #398 is not about the telemetry, but I think that the issue has the same source.

The first post yes but the latest is related. Anyway this is the same RX which is for me the common factor.
Really interrested if you find anything.

Tygrys-1, 2020-10-01:
Did some research. To my surprise, firmwares are the same on both working and non-working X6B. What is more interesting, the "non-working" X6B on the bench works pretty well. It is just making problems in the model. So a thing to investigate is the BEC quality - it is a 5V from the ESC - or other connections - voltage monitoring and outputs. The RX power show 4,99V, so the voltage is good. I'll check that later. The first thought is to add a big capacitor on 5V and watch the results. Any other suggestions are welcome.

Tygrys-1, 2020-10-03:
Further testing brought mixed results - I do not see the pattern. It sometimes works ok and sometimes looses the telemetry.

I checked the receiver in the model with Jumper T8SG V2+ on deviation software and with original Flysky i6X, with the rock solid telemetry. I can failsave the model by touching the RX antenna with the antenna of T8SG, but the telemetry stays readable with 1cm close to the receiver and on any further distance. As both multimodules on my transmiters (Radiomaster and TMX5 lite) have the telemetry problem with this X6B and other transmiters are working ok - I think there must be some incompatibility with the multimodule and X6B. Or maybe it is OpenTx issue?

What I have observed, with close proximity (about 10 cm TX to RX) on the bench, when there are no "telemetry loss situation" I can see the TRSS signal showing values 245-247, and suddenly drops for a moment to the range 100-116. It is repetable. During that TRSS lower values state - other telemetry readouts are stable. I'm not sure if it is a real weeker telemetry - or it might be just strange readout of the telemetry signal level. Or maybe they "talk" at the same time?

Any clues or suggestions for furher testing are welcome. Maybe I can make the video to show the situation, if it will help in some way.

pascallanger, 2020-10-03:
The big difference with the t8sg is that it won't report telemetry lost/recovered. The last numbers received is displayed and that's it.
I don't know for the flysky radio but if it's the same you won't notice anything like you can notice on OpenTX which is reporting loss of telemetry frames.

Tygrys-1, 2020-10-03:
I know, but I can observe telemetry data - voltage fluctuations 4,99-5,00V, and after lost of telemetry the displayed values on the main page turn background and text to oposite. So yes, it can be detected. On OpenTX telemetry stops for noticable periods of time. There is no such observation on deviation or FS-i6x.

I'm thinking of using some old version of multiprotocol software for test. Can you suggest which versions?

Tygrys-1, 2020-10-04:
On the last 1.69 things seams to be worse. Or more visible. Or it is TX16S I used for tests today. I have tested 5 X6B's and 2 of them are working with no problem, 3 have problems with the telemetry on TX16S. 4 of them are new, not used in models, one even unpacked today. One working and one not working have the exact same firmware, I have not tested others for firmwares.
Tested on the bench, 30 cm receiver to transmiter distance. Results:

Good receiver: Power up the receiver. Select the model in transmitter (previously bound). After a second receiver shows link, next second the transmiter shows telemetry signal ok. It is stable all the time.
Bad receiver: Power up the receiver. Select the model in transmitter (previously bound). After a second receiver shows link, no telemetry in transmiter. After some time - maybe few seconds or few minutes the telemetry catches for the while and then say "lost". It may repeat. The one I'm testing now had no telemetry for 10 minutes.

Second test: Switch the transmiter off then on. Receiver shows link, no telemetry on the transmiter.
Third test: Switch the reciver off the on. There is a momentary link on the receiver, but no telemetry on the transmiter.
Fourth test: Walk further from the receiver. No significant efect. Somtimes it catches telemetry for a second.

All receivers works ok with deviation T8SG V2+, showing the telemetry just after the link, no losses.

Conclusion - this is strange, but it is at least consistent.  :-)

pascallanger, 2020-10-04:
So it means that even if I go and purchase this RX to go to the bottom of this issue, I'm not even sure to get one with the issue...

The only thing I can see would be to try different values in this line:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/baf9a0f978f184e36f34a83ca516447836dbb597/Multiprotocol/AFHDS2A_a7105.ino#L275
Try to increment/decrement with steps of 50.

If it doesn't work then I would like to see what the module is receiving, would you be able to set it to debug?

Tygrys-1, 2020-10-04:
The package to France costs more then the receiver itself ... :-(

Ok, I'll look at the timing. I don't know if the deviation does this in similar way, if so I might look what the timings are there.
I have the second multiprotocol module and having the knowledge from the last tests I may try to get results there.
I tried it on 1.59 and there was no so consistent results.  I'll check it again.
Never did debug on multiprotocol before, but if it is just the firmware update I might give it a try.

pascallanger, 2020-10-04:
Most of the time Deviation timings can't be used directly. You better try to change the timings as indicated and see if it gets better or worse.

Tygrys-1, 2020-10-06:
The problem got worse. The previous receiver I put to tests stoped sending telemetry at all, even for deviation TX. It was working ok before tests, witch good telemetry on deviation. I do not know, if testing somehow may make some damage - closing antennas together and so. The receiver perfectly receives signal and binds, but I have not tested it on far range.
No I have two receivers working ok, two working sometimes (telemetry looses) on deviation and opentx and one working only one way. I'll stop wasting time on them and will not going to risk ones still working.

So the problem is in the receiver, or it's telemetry output. Now it is repetable on 3 transmitters, It was going worse and worse, from loosing telemetry on the close distance and gaining it after 4m, to totall lost telemetry fucntion. I do not know why and what went wrong.

closed 2020-10-06 by Tygrys-1

activa73, 2020-10-08:
Hi Pascal, I live in France and I can send you a reciever if you want.

pascallanger, 2020-10-08:
@activa73 Does this receiver shows the mentionned issues?

activa73, 2020-10-08:
Yes

pascallanger, 2020-10-08:
OK. Send me a PM.

activa73, 2020-10-08:
Where i can find your personnal address?

emolina82, 2021-02-11:
Did anybody find the solution?? I have the same problem, FS X6B and Radiomaster TX16s. I've expended a lot of time investigate this problem .. for now I've decided to live with it and silence telemetry alarm, I live with rssi in my goggles.

If somebody has news .. let us know

Tygrys-1, 2021-03-02:
Seems to be solved #398