pascallanger/DIY-Multiprotocol-TX-Module #435: Unstable input in bayang protocol

schugabe, 2020-09-28:
My iRangeX module started to have issues while using it with the Bayang protocol. After binding the quad flies normal but the TX beeps every few seconds and shows: _"No Input"_ or _"Prot. invalid"_ in the status flag. Both LEDs are on and do not blink.
Telemetry and quad are working I just get the beeps every few seconds (but it's not in a fixed intervall).

**Setup:** 

- TX: X7S with OpenTX 2.3.9
- iRangeX flashed with 1.3.1.69
- Eachine E011 quad

**Settings:**

- Mode MULTI
- Type Bayang
- Subtype Std
- Ch. Range CH1-16
- Low power disabled
- Telemetry 1, 
- Bind Ch enabled

I recently did the crossfire mod to enable 400k baud rate but as far as I can tell every other module is happy with the mod. I use the same iRangeX module with another whoop with the redpine protocol selected and do not get any warnings here. 

I'm not sure if this an opentx or mpm issue and if the hardware mod is the reason for this. I did not fly the quad for some time so I can't really tell what change triggered this issue. Thank you for any help.

pascallanger, 2020-09-28:
That shows connectivity issue between the module and the radio.
Check if the the bottom pin of the JR bay is correctly fitted in the module connector. Check the connector solder points and better redo them, I've seen few people with a broken pad and intermittent connection.
If nothing above then it will point to your mod. I know someone reported something may be similar but you better ask on the forum since I can't remember what was the issue after he modded his radio like you. The user was Capnbr(y/ie) from memory so a quick search should help you.
Pascal

closed 2020-09-28 by pascallanger

schugabe, 2020-09-30:
Thank you for the tipp I checked the connection (was not the issue) and added a pull up as [described here](https://www.rcgroups.com/forums/showpost.php?p=45121703&postcount=19145) and that fixed the issue.