pascallanger/DIY-Multiprotocol-TX-Module #436: Kyosho/Hype new protocol

pascallanger, 2020-09-30:
### Hype ST6DF protocol

**Normal packets**
ID is 0xAF 0x90 0x73 0x8C
15 freqs: 0x27 0x1B 0x63 0x75 0x3 0x39 0x57 0x69 0x87 0xF 0x7B 0x3F 0x33 0x51 0x6F
Payload is on 6 bytes: 0x80 0x7D 0x6C 0x7E 0x37 0x0
They calculate the payload once every 20ms (approximately), and then send it 22 times. The timing of the protocol is not stable not sure if it's important to keep the RX in sync but let's use 931µs...
P[0]=aileron, center 0x80, right 0x1D, left 0xE9
P[1]=elevator, center 0x7E, up 0x17, down 0xE0
P[2]=throttle, down 0x10, up 0xDE
P[3]=rudder, center 0x80, right 0x24, left 0xEE
P[4]=ch5
P[5]=aux=ch6, 0x00, 0xFA

**Bind packets**
Bind ID is 0xAF 0x00 0xFF 0x00
Bind channel is 1.
Payload: 0x90 0x8C 0x27 0x1B 0x63 0x75 0x3 0x39 0x57 0x69 0x87 0xF 0x7B 0x3F 0x33 0x51 0x6F
Normal ID is 0xAF P[0] 0x73 P[1], **how is the 0x73 calculated???**
P[2..16]: 15 channels to use in normal mode.
Bind packet is sent every 1421µs.

**Info**
A7105 quartz is 16MHz

pascallanger, 2020-09-30:
**0x73** is not constant since here is what is happening:
Payload: 0x35 0x9 0x47 0x7 0x28 0x3F 0x29 0xD 0x2F 0x48 0x4 0x41 0x12 0x23 0x1A 0x6 0x3A
Multi is using the ID: 0xAF 0x35 **0x73** 0x9
But the RX is using the ID: 0xAA 0x35 **0xF6** 0x9

Immediate question: is this byte related to the frequencies or the 2 other ID bytes?

cagatay87, 2020-09-30:
i have another 24c02 dump from another receiver , i know 24c02 keeps bind data.
https://s7.dosya.tc/server17/jxe3hk/benim_calismam.zip.html
maybe it helps,if i have to do sth else , please inform me.

pascallanger, 2020-09-30:
@cagatay87 can you modify the code like this:
```
		#ifdef KYOSHO_FORCE_ID_HYPE
			MProtocol_id=0xAF90738C;
			set_rx_tx_addr(MProtocol_id);
			memcpy(hopping_frequency,"\x27\x1B\x63\x75\x03\x39\x57\x69\x87\x0F\x7B\x3F\x33\x51\x6F",15);
		#endif
```
To:
```
		#ifdef KYOSHO_FORCE_ID_HYPE
			MProtocol_id=0xAF90738C;
			set_rx_tx_addr(MProtocol_id);
		#endif
		memcpy(hopping_frequency,"\x27\x1B\x63\x75\x03\x39\x57\x69\x87\x0F\x7B\x3F\x33\x51\x6F",15);
```
Do a dump of the RX while binding (like you've done just before).

pascallanger, 2020-09-30:
> i have another 24c02 dump from another receiver(which is working with same tx) , i know 24c02 keeps bind data.
> https://s7.dosya.tc/server17/jxe3hk/benim_calismam.zip.html
> maybe it helps,if i have to do sth else , please inform me.

The eeprom dumps contain the bind payload:
90 8c 27 1b 63 75 03 39 57 69 87 0f 7b 3f 33 51 6f
66 64 09 5d 0f 63 1b 69 27 75 33 7b 87 3f 51 99 6f
This is not helpful...

cagatay87, 2020-09-30:
here is file ,2 times i recorded . No bind
https://www.dosya.tc/server31/2ftff8/son.7z.html

cagatay87, 2020-09-30:
> > i have another 24c02 dump from another receiver(which is working with same tx) , i know 24c02 keeps bind data.
> > https://s7.dosya.tc/server17/jxe3hk/benim_calismam.zip.html
> > maybe it helps,if i have to do sth else , please inform me.
> 
> The eeprom dumps contain the bind payload:
> 90 8c 27 1b 63 75 03 39 57 69 87 0f 7b 3f 33 51 6f
> 66 64 09 5d 0f 63 1b 69 27 75 33 7b 87 3f 51 99 6f
> This is not helpful...
> 
> We need to work from the Multi TX code & capture if the original RX side to understand.

at 3th try i binded 
https://s7.dosya.tc/server17/61s0r4/3.th_try_binded.logicdata.7z.html

after bind happened , light again blinking when i reset rx.Same situation as old

pascallanger, 2020-09-30:
@cagatay87 I can see the RX getting the bind but the capture ended up before it went back to RX.
What you should do is
1. bind -> LED goes off
1. capture the RX power on

cagatay87, 2020-09-30:
> @cagatay87 I can see the RX getting the bind but the capture ended up before it went back to RX.
> What you should do is:
> 
> 1. bind -> LED goes off
> 2. capture the RX power on

do you mean , after bind , remove bind cable and start logging right?

pascallanger, 2020-09-30:
> > @cagatay87 I can see the RX getting the bind but the capture ended up before it went back to RX.
> > What you should do is:
> > 
> > 1. bind -> LED goes off
> > 2. capture the RX power on
> 
> do you mean , after bind , remove bind cable and start logging right?

yes

cagatay87, 2020-09-30:
here is full record , several times im trying to bind , than bind successful  and remove bind cable.
https://www.dosya.tc/server31/4hc7bw/bind_and_after.7z.html

and its another record , just after bind (several seconds later im testing throttle elevator etc)
https://s7.dosya.tc/server17/hwnt5q/after_bind__no_success_.7z.html

pascallanger, 2020-09-30:
So here is the result:
Bind: 0x35 0x9 0x27 0x1B 0x63 0x75 0x3 0x39 0x57 0x69 0x87 0xF 0x7B 0x3F 0x33 0x51 0x6F
ID: 0xAA 0x35 0xF6 0x9

pascallanger, 2020-09-30:
Valid IDs:
ID: **0xAF** _0x90_ **0x73** _0x8C_
ID: **0xAA** 0x35 **0xF6** _0x9_
ID: **0xAF** _0x0_ **0xFF** _0x0_

The 2 bytes used during bind  in italic are changing the values of the 2 other bytes in bold...

pascallanger, 2020-09-30:
The calculation is ((0xFFFF - 0xXXYY) & 0x0FFF)|0xA000 where XX is the first byte of the payload and YY the second byte

pascallanger, 2020-09-30:
Protocol tested and working.

closed 2020-09-30 by pascallanger