pascallanger/DIY-Multiprotocol-TX-Module #437: Multiprotocol Update for the STM32F103C8 processor.

Talha909, 2020-10-01:
I have MPM Vantec 4 in multi protocol . This module has STM32F103C8.   And it has 64k Memory. How can I install new updates on it.  Multi Flasher did not update the bootloader and new firmware.  What the procedure for update for future.

pascallanger, 2020-10-01:
Your module is not supported anymore since it does not have the minimal requirements to run this project. Please contact the manufacturer for future updates.

If this answer does not suit you then search on this forum or on github on a way to bypass the checks but it is at your own risk to use it in these conditions.

closed 2020-10-01 by pascallanger

pascallanger, 2020-10-01:
Duplicated with https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/393

tedbmoss, 2020-10-01:
WoW! the Chinese screwed up again! What are the correct modules for the "lite"

pascallanger, 2020-10-02:
iRangeX lite is fine

tedbmoss, 2020-10-04:
Thanks, I appreciate all your work.