pascallanger/DIY-Multiprotocol-TX-Module #438: DSM forwarding tool causes Emergency mode on Horus

Hobby4life, 2020-10-01:
I dont know if this tool is for Horus use also, but opening the tool causes the radio to go into emergency mode.

Situation, MPM not connected to the radio.

what i want to say is that the radio should never go into emergency mode at all times.

pascallanger, 2020-10-01:
Can you give more details? Like the config of the modules? Internal enabled or not? external enabled on multi? Configured in multi DSM?
Is it only happening when the module is not in?
The script is supposed to be run on 2.3.10 nightly builds... But that shouldn't matter.
More info is definitively needed

Hobby4life, 2020-10-01:
I am now running latest OTX Nightly 2.3.10N335

Internal module enabled (ACCESS or D16)

External Disabled.

kilrah, 2020-10-01:
Which Horus? There are 3 of these...

Hobby4life, 2020-10-01:
X10S Express,

X12S (Access upgrade)

pascallanger, 2020-10-01:
And the 2 radios in 2.3.10 are doing an emergency modeb when you launch the Lua?

Hobby4life, 2020-10-01:
Correct

pascallanger, 2020-10-01:
I can't reproduce on a tx16s...
What about if you disable the internal module ?

Hobby4life, 2020-10-01:
After disabeling internal module both TX still go into Emergy mode

pascallanger, 2020-10-01:
Is the HoTT Lua script doing the same?

Hobby4life, 2020-10-01:
HoTT script is executing ok, no problems

pascallanger, 2020-10-01:
So that must be something linked to the display

pascallanger, 2020-10-01:
Can you edit the end of the Lua file?
```
-- Main
local function DSM_Run(event)
  if event == nil then
    error("Cannot be run as a model script!")
    return 2
  elseif event == EVT_VIRTUAL_EXIT then
    DSM_Release()
    return 2
  else
    DSM_Menu(event)
    DSM_Send_Receive()
    DSM_Display()
    return 0
  end
end
```
Try commenting (add -- in front of the line) DSM_Menu(event), if it still goes to emergency comment DSM_Send_Receive(), if it still does it comment DSM_Display().

Hobby4life, 2020-10-01:
I have commented them 1 by 1, but even If I comment all 3 functions, it still goes into EM

pascallanger, 2020-10-01:
Ok we are narrowing.
Can you comment this whole section:
```
  --Set protocol to talk to
  multiBuffer( 0, string.byte('D') )
  --test if value has been written
  if multiBuffer( 0 ) ~=  string.byte('D') then
    error("Not enough memory!")
    return 2
  end
  --Init TX buffer
  multiBuffer( 3, 0x00 )
  --Init RX buffer
  multiBuffer( 10, 0x00 )
  --Init telemetry
  multiBuffer( 0, string.byte('D') )
  multiBuffer( 1, string.byte('S') )
  multiBuffer( 2, string.byte('M') )

```

Hobby4life, 2020-10-01:
--[[
--]] all the above,, still EM

pascallanger, 2020-10-01:
Then delete all the lines Text[]=...

Hobby4life, 2020-10-01:
Still EM

I will continue testing tomorrow if you have new ideas.. will take my TX to work :-)

pascallanger, 2020-10-01:
Can you attach the file you end up with? I would like to see what's left of it...

Hobby4life, 2020-10-01:
[DSM FwdPrg.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5314593/DSM.FwdPrg.zip)

pascallanger, 2020-10-01:
You haven't left the 3 lines commented:
    DSM_Menu(event)
    DSM_Send_Receive()
    DSM_Display()

pascallanger, 2020-10-01:
After instead of commenting try to delete...

kilrah, 2020-10-01:
Also post your otx please

Hobby4life, 2020-10-02:
I have deleted everything in Init and Run

[document1.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5316172/document1.zip)

`-- Init
local function DSM_Init()
 

end

-- Main
local function DSM_Run(event)
 

end

return { init=DSM_Init, run=DSM_Run }`

Hobby4life, 2020-10-02:
I  have also delete the lines below and kept everything else uncommented. Still EM

DSM_Menu(event)
DSM_Send_Receive()
DSM_Display()

pascallanger, 2020-10-02:
Starting with what you had previously: init empty and run empty, can you delete the functions one by one starting from the bottom and going up?
If you think that's the declarations just start deleting as well by sections so you can find out which one is causing issues.

pascallanger, 2020-10-02:
Could it be that the associated luac file got corrupted? may be worth a try to rename it (don't delete) to see if the behavior is the same.

kilrah, 2020-10-02:
Was a stack overflow in OpenTX, will be fixed in next nightly.

closed 2020-10-02 by pascallanger

Hobby4life, 2020-10-02:
Good job! :-)