pascallanger/DIY-Multiprotocol-TX-Module #439: DSMX LBT

teozfrank, 2020-10-03:
Hi is DSMX supported with LBT (EU) ? I couldnt see a reference to it.

pascallanger, 2020-10-03:
As far as I know DSMX is not doing LBT... If you think differently the nI would need a spi dump of the curf6936 from a DSM radio having this feature.

teozfrank, 2020-10-03:
In the EU it would have to under the regulations no? I have some eu rx's here maybe I can do that but what hardware would I need?

pascallanger, 2020-10-03:
LBT is not mandatory... It's less constraint on the RF system if you have LBT but that's it. I'm not sure if DSMX needs any modifications to match the regulation but we can check if something has been added.
You need a compatible $5 saleae logic analyzer and solder 4 wires to the cyrf SPI bus. You need to make sure if you use a RX that it is a long range telemetry since most DSM RXs are only short range and therefore won't implement anything since they have extremely low TX signals.

kilrah, 2020-10-03:
I'm not aware of there being different versions of DSMX, it was made precisely to be usable all around the world

teozfrank, 2020-10-03:
@kilrah Correct but regulations back in 2015 LBT based DAA is required in the EU for adaptive equipment. See here https://www.etsi.org/deliver/etsi_en/300300_300399/300328/01.09.01_60/en_300328v010901p.pdf Starts on page 19.

kilrah, 2020-10-03:
And DSMX was made to cater for that. That's why new Spek radios don't have DSM2 in EU anymore.

They were just a bit smarter than say FrSky and looked up in advance what they would have to comply to in a couple of years.

teozfrank, 2020-10-03:
> And DSMX was made to cater for that. That's why new Spek radios don't have DSM2 in EU anymore.

Thats true actually.

> They were just a bit smarter than say FrSky and looked up in advance what they would have to comply to in a couple of years when they designed it.

Thing is tho the US or international versions of radios do not need LBT based DAA, thats only an EU thing, I'm pretty sure. Thats why there is two version of radio firmware for Frsky for example one for international (outside of EU) and one for LBT which is for europe. That should mean that should be somewhat simular for Spektrum as they have to follow the same rules for EU. There is an EU specific TX and an FCC version but its not swappable on firmware like FrSky. Im just curious to see if Spektrum implemented LBT on there EU transmitters thats all.

schwabe, 2020-10-03:
You can design protocol to be compatible with LBT and without LBT. The non-LBT is then just always send instead listening for channel free. And the toy remote that came with the Inductrix was low power (a few mW only) still supports DSM2 (with lower power which is fine without LBT)

OpenUAS, 2020-10-03:
@teozfrank just FYI, LBT is allowed elsewhere, but no LBT is not allowed in EU (**if you use LBT** in your solution), but not applicable for DSMX since DSMX uses 10% MU (Medium Utilisation factor) not LBT AFAIK see sub 4.3.1.6. There more ways to Rome to adhere to ETSI compliance, see newer 2.2.2 for a more up to date description https://www.etsi.org/deliver/etsi_en/300300_300399/300328/02.02.02_60/en_300328v020202p.pdf

And about DSM in depth here https://wiki.paparazziuav.org/wiki/DSM

BTW maybe this was better not asked in a Issue tracker but on e.g. and RC forum

kilrah, 2020-10-03:
>Locking up channels isnt allowed in EU and thats why it had to be removed.

That's incorrect. Locking channels in CW is limited to 10mW, but DSM2 was DSSS so even if fixed it still falls under spread spectrum regs.
And as mentioned LBT is not mandatory provided other constraints (MU), it's just one of the available solutions.

teozfrank, 2020-10-03:
> That's incorrect. Locking channels in CW is limited to 10mW, but DSM2 was DSSS so even if fixed it still falls under spread spectrum regs.
> And as mentioned LBT is not mandatory provided other [...]

Ok so why did Spektrum remove DSM from EU firmwares then?

> 
> 
> @teozfrank just FYI, LBT is allowed elsewhere, but no LBT is not allowed in EU (**if you use LBT** in your solution), but not applicable for DSMX since DSMX uses 10% MU (Medium Utilisation factor) not LBT AFAIK see sub 4.3.1.6. There more ways to Rome to adhere to ETSI compliance, see newer 2.2.2 for a more up to date description https://www.etsi.org/deliver/etsi_en/300300_300399/300328/02.02.02_60/en_300328v020202p.pdf
> 
> And about DSM in depth here https://wiki.paparazziuav.org/wiki/DSM
> 
> BTW maybe this was better not asked in a Issue tracker but on e.g. and RC forum, maybe close this issue?

Interesting, yes maybe it might be. I think I may post on rcforums.

closed 2020-10-03 by teozfrank

kilrah, 2020-10-04:
> Ok so why did Spektrum remove DSM from EU firmwares then?

Becasue it didn't comply. I was just pointing out that you were citing the wrong reasons for it not to.

teozfrank, 2020-10-04:
> 
> 
> > Ok so why did Spektrum remove DSM from EU firmwares then?
> 
> Becasue it didn't comply. I was just pointing out that you were citing the wrong reasons for it not to.

Ahh ok, thank you.