pascallanger/DIY-Multiprotocol-TX-Module #441: Supported with OpenTX V2.3.9?

teozfrank, 2020-10-04:
Hi,

I am trying to get the multimodule setup with my QX7. I have flashed the module according to the instructions with multi-stm-opentx-taer-inv-v1.3.1.69.bin.

But on OpenTX companion 2.3.9 there is no multi option in build options, nor does it display "multi" in external module. Is there something I am doing wrong?

[image]

All the videos I find on YouTube are over 3 years old so appriciate any help here.

Frank.

kilrah, 2020-10-04:
It's always there, no option needed.

teozfrank, 2020-10-04:
> 
> 
> It's always there, no option needed.

Oh ok, what about multi then in external module. If I try and select DSM2 -> DSMX and try and bind with rx 00 it wont bind. I am using a dsmx bind and fly inductrix FPV +

pascallanger, 2020-10-04:
You must select multi then the protocol and not dsm2.

teozfrank, 2020-10-04:
> 
> 
> You must select multi then the protocol and not dsm2.

Options I get on External RF is OFF, PPM, XJT, DSM2, R9M and SBUS. I dont seem to have MULTI ?

pascallanger, 2020-10-04:
@3djc @kilrah any ideas?
@teozfrank This is OpenTX related. You might have more chance to find help if you open an issue on the OpenTX github.

bbc25185, 2020-10-04:
> > You must select multi then the protocol and not dsm2.
> 
> Options I get on External RF is OFF, PPM, XJT, DSM2, R9M and SBUS. I dont seem to have MULTI ?

Can you check the OpenTx version installed on your radio? (you also dont have the CRSF protocol)
(Long press menu, go to page 7 and read the version of OpenTx)

teozfrank, 2020-10-04:
> Can you check the OpenTx version installed on your radio? (you also dont have the CRSF protocol)
> (Long press menu, go to page 7 and read the version of OpenTx)

opentx - x7 2.3.9-frsky (e4336857)

bbc25185, 2020-10-04:
You get the firmware from companion or downloaded from Frsky Site?

teozfrank, 2020-10-04:
> 
> 
> You get the firmware from companion or downloaded from Frsky Site?

Frsky site.

bbc25185, 2020-10-04:
Get the firmware from companion and reflash with it! XD

pascallanger, 2020-10-04:
Wow good finding!!!
He showed a Companion screenshot earlier so I assumed he was using Companion to download and install it...

teozfrank, 2020-10-04:
> 
> 
> Wow good finding!!!
> He showed a Companion screenshot earlier so I assumed he was using Companion to download and install it...

I used the companion to flash it yes.

bbc25185, 2020-10-04:
DOWNLOAD and flash... DONT use the file downloaded from frsky site! XD

teozfrank, 2020-10-04:
> 
> 
> DOWNLOAD and flash... DONT use the file downloaded from frsky site! XD

Yes that was it!! Thank you!!!

version is now 2.3.9-otx 60e1edc1

:)

closed 2020-10-04 by teozfrank

bbc25185, 2021-10-10:
You connect the radio to PC WITHOUT turning on the radio????
if you connect it off, it connects to the PC in DFU mode which is what we are interested in (in DFU it is possible to flash bootloader and firmware, in the other ways only the firmware) ...
All the other ways only work if you don't touch the bootloader and, at least the first time, regardless if it works or not, it's best to get it complete.