pascallanger/DIY-Multiprotocol-TX-Module #442: DJI FPV Protocol

al21883, 2020-10-05:
Since the DJI digital FPV system is being a hit and several pilots are switching their video systems to this one on custom builds. It will be wonderful for a lot of people to have the option to use the built in receiver on the "Air Units" and "Caddx Vista" units.

Maybe, in a few months, it becomes the most common receiver mounted(not used) on quads an FPV machines.

bbc25185, 2020-10-05:
It will be difficult (even if it would be nice)
The DJI FPV remote control works in 5.8ghz and, from what I know, no module of the multi-protocol is able to transmit in 5.8Ghz

kilrah, 2020-10-05:
DJI uses completely different components and tech.

al21883, 2020-10-06:
Thanks for your replies.
I had read [somewhere](https://uavcoach.com/dji-fpv-racing-drone/#:~:text=The%20DJI%20FPV%20Remote%20Controller,~5725MHz%2C%205725~5850MHz.) that the transmitter could work in 2,4Ghz too.
I think it could be possible with DJI's collaboration. This time they have put on the market a product oriented to DIY RC people and have integrated open source OSD protocol to it's product.

teozfrank, 2020-10-07:
> Thanks for your replies.
> I had read [somewhere](https://uavcoach.com/dji-fpv-racing-drone/#:~:text=The%20DJI%20FPV%20Remote%20Controller,~5725MHz%2C%205725~5850MHz.) that the transmitter could work [...]

Thats OSD tho not control, Control needs a specific chipset (Or an SDR) in order to transmit the control signals as the rx is expecting it. I am pretty sure DJI will not release too how there RF system work, it would have to be reverse engineered.

gibbi, 2020-10-08:
The RF-Signal of DJI FPV is pretty much the same as OcuSync only that they do their frequency hopping within a specific channel. The Downlink is essential for the link to work so at the bare minimum we would need a 20+MHz SDR + fpga + internals from DJI... Yeah - not going to happen.

pascallanger, 2021-10-20:
I'm not sure if I should close it since people don't check closed issues and they open new ones. Adding the "Won't do" status was for me a way to show that I'm aware and won't work on it...