pascallanger/DIY-Multiprotocol-TX-Module #443: Flysky telemetry - incorrect voltage readings from FS-CVT01 sensor

Tygrys-1, 2020-10-05:
There are 2 types of voltage sensors for Flysky. One is AVT01 and the second one is CVT01. The first one is simmilar to the voltage sensor reporting receiver volltage. The second one has the different ID (AFAIR 03) and scaling. It can sense +/- voltage and the range is higher then AVT01, but less precision. Voltage Range: 0~100V DC, Accuracy: +/- 0.1V

Currently CVT01 sensor with connected sensing pins reports 655.27V instead of 0V.

pascallanger, 2020-10-05:
What's the ID of the sensor?

Tygrys-1, 2020-10-05:
My bad, sorry. Messed up two things. The sensor is ID 3 and it shows up as A3 on the sensor screen. But the result is a signed number, so it should read the value 65527 as -0,09V, not 655,27V. When connected to the voltage source it reads 8.4V on one direction, and -645V on the other. :-)

pascallanger, 2020-10-05:
Strange, the code is signed in OpenTX telemetry\flysky_ibus.cpp:
int32_t value;
value = (packet[3] << 8) | packet[2];
if (sensor->unit == UNIT_VOLTS) value = (int16_t) value;

@3djc, @Kilrah an idea of what could be wrong with the code above?

pascallanger, 2020-10-06:
@Tygrys-1 Which version of OpenTX do you have?
The current code is handling the values as signed so it should work...

Tygrys-1, 2020-10-06:
Was using 2.3.9. On 2.3.10 it is signed, and looks ok.

closed 2020-10-06 by Tygrys-1