pascallanger/DIY-Multiprotocol-TX-Module #444: Enable the STM32 module serial debug feature fails with compilation error

teozfrank, 2020-10-05:
As the title suggests following the full guide on setting this up from here:

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md

I get an compilation error:

c:/users/frank/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Frank\AppData\Local\Temp\arduino_build_236814/Multiprotocol.ino.elf section `.data' will not fit in region `rom'

c:/users/frank/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: region `rom' overflowed by 13392 bytes

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Multi 4-in-1 (STM32F103CB).

How can this be resolved?

Frank

kilrah, 2020-10-05:
Comment out protocols until it fits.

pascallanger, 2020-10-05:
You need to comment some protocole to enable the debug mode. That's what the message tells you.

closed 2020-10-05 by pascallanger

teozfrank, 2020-10-05:
Ohhh I see thank you. Im not that familure with the code base so how would I comment out the protocols I dont need? could you for example link me to the section in code where I need to do that please?

teozfrank, 2020-10-05:
Found it in config.h in protocols.h