pascallanger/DIY-Multiprotocol-TX-Module #445: DSMX over the air decoder

teozfrank, 2020-10-07:
Hi,

I have been doing some research on the DSMX implementations with the multimodule, QX7 and genunie spektrum rx's and a inductrix fpv +. So far so good, other than having to reverse the aileron and rudder channels on inductrix fpv + its perfectly useable so well done !!

So I had an idea, using the genunie Spektrum radio and RX is it possible to use the multimodule to listen to the radio communication and decode the stick positions live from the spektrum tx? If we created a model on the tx, bound to the RX feature of the multimodule and get the GUID for hopping pattern, then bound the TX to the actual device being controlled and fly it. We could then listen and display the stick positions? Or maybe there is a way to know display the stick positions from the RX feature of the multimodule? I couldnt really understand the code so I wasnt sure where to extract that information?

Kind Regards,
Frank.

kilrah, 2020-10-07:
You can bind as many RXes as you want to a radio, and the MPM is just another RX. So yes.

teozfrank, 2020-10-07:
> 
> 
> You can bind as many RXes as you want to a radio, and the MPM is just another RX. So yes.

How would I output the stick positions with the MPM in rx mode?

kilrah, 2020-10-07:
The TRx (trainer) mixer sources.

teozfrank, 2020-10-07:
> 
> 
> The TRx (trainer) mixer sources.

Around what lines / area of the code ? 
I want to output the received stick positions from a genunie Spektrum TX with the MPM in rx mode.

kilrah, 2020-10-07:
Not talking of source code, but the sources for the mixer page in opentx...

teozfrank, 2020-10-07:
> 
> 
> Not talking of source code, but the "source" setting for lines in the mixer page in opentx...

Sorry but I literally am not sure what you are talking about?
Do you mean showing the outputs in opentx itself is it? Does it do that already?

kilrah, 2020-10-07:
The point of multi RX is to use it for wireless trainer. So look up how use the trainer input channels on OpenTX.

teozfrank, 2020-10-07:
> 
> 
> The point of multi RX is to use it for wireless trainer. So look up how use the trainer input channels on OpenTX.

I did but I couldn't get it to work. I set a global function and set it to when a switch is on, its function is trainer. Set the QX7 as Master. Set the external rf as dsm rx. Bound to it with my Spektrum DX9. It doesnt seem to work. Maybe I need the internal RF to be on for it?

pascallanger, 2020-10-07:
Check the doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm_rx---70
You need to select Master/Multi to get the trainer working.

teozfrank, 2020-10-08:
> 
> 
> Check the doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm_rx---70
> You need to select Master/Multi to get the trainer working.

There is no "Master/Multi" I only have Slave/Jack and Master/Jack.

kilrah, 2020-10-08:
Didn't notice you had a QX7. It does not have the master/multi choice since it's pointless as the internal module cannot be operated simultaneously as a multi feeding telemetry. That's valid for all FrSky radios.

teozfrank, 2020-10-08:
> 
> 
> Didn't notice you had a QX7. It does not have the master/multi choice since it's pointless as the internal module cannot be operated simultaneously as a multi feeding telemetry. That's valid for all FrSky radios.

Ahh ok, that makes sense. I was able to print out in serial the live values from the dsm_rx when I bound my spektrum radio to that. Then with the same model memory on the spektrum radio I bound it to another rx and it still works which is quite cool. I can see the live values while I fly a small fpv drone on the screen. The next thing if I could overlay those stick values on opentx would be awesome. Would it be possible to modify opentx to do that? When Multi dxm rx is enabled it automatically overlays the stick movements of the other tx bound to it. Or would it be better to create a script for that?

kilrah, 2020-10-08:
As long as you have master/multi available you can just map the trainer inputs to opentx channels and look at the movements on the channel monitor, and log them too, no script needed. 
I've done that myself with a T16.

I know OpenTX has a comile option to force master/multi to show on the X10/X12 since it's possible to make it work with in some conditions and with hardware mods but not sure if that's also implemented for a QX7.

teozfrank, 2020-10-08:
> 
> 
> As long as you have master/multi available you can just map the trainer inputs to opentx channels and look at the movements on the channel monitor, and log them too, no script needed.
> I've done that myself with a T16.
> 
> OpenTX has a compile option to force master/multi to show on the X10/X12 since it's possible to make it work in some conditions and with hardware mods but not sure if that's also implemented for a QX7.

Thats quite interesting alright. Even if the trainer function would never actually function on the QX7 because of hardware limitations I dont really care, I would just like to do what you have tried. Maybe I could ask for it, or make a custom build on my own for playing around with.

teozfrank, 2020-10-08:
Found it, its a build option called ALLOW_TRAINER_MULTI

https://github.com/opentx/opentx/blob/2.3/radio/src/dataconstants.h#L211

https://github.com/opentx/opentx/pull/7226/commits/5d3bf274a340876759fc72569952e071ab192547

teozfrank, 2020-10-09:
> As long as you have master/multi available you can just map the trainer inputs to opentx channels and look at the movements on the channel monitor, and log them too, no script needed.
> I've done that [...]

It works with my QX7, just built opentx with that flag and its there. Had to reverse rudder and ailerons and now the sticks move the right way!! :D

pascallanger, 2020-10-10:
Closing issue since you seem to get what you wanted.

closed 2020-10-10 by pascallanger