pascallanger/DIY-Multiprotocol-TX-Module #447: Hott transports only 12 channels, extend to 16 channels

wimalopaan, 2020-10-08:
If an Hott-RX like Gr-12L is bound to an Hott-Tx as MC-32, the RX outputs all 16 Channels on SBus. If the same RX is bound to MPM, it outputs only 12 Channels as documented.

Is it possible to extend this to 16 channels?

pascallanger, 2020-10-08:
The 2 HoTT TXs I've seen were only able to send 12 channels. The payload sent over the air can only contain 12 channels. I don't know how they are transmitting more channels. Unless I have access to a TX which can do it there is nothing I can do.

pascallanger, 2020-11-06:
Unless someone provides a TX with this ability I can't do anything...

wimalopaan, 2020-11-14:
Would you like to get a MC-32 transmitter to verify/solve this issue? A buddy of mine is willing to send you a MC-32. Please let me know if you have time to investigate this further. Thx.

pascallanger, 2020-11-14:
Sure if he is ok to send me a TX + RX, I can have a look at adding more channels (at least the 16 first). Contact me by MP on RCG.

pascallanger, 2020-12-13:
I've added support for 16 channels in 1.3.1.82. Please test and report.

wimalopaan, 2020-12-15:
Tested with GR12L and GR16. Works!

Thanks you very much!!!

closed 2020-12-15 by pascallanger

G3Dmaker, 2021-05-03:
I use a Radiomaster TX16S. OpenTX Firmware: 2.3.11. 
MULTI-Module Firmware: multi-stm-serial-taer-v1.3.2.77
RX: Graupner GR-32 HoTT
I cannot control more than 12 channels. 
Ch 1-12 are OK.
Ch 13-16 not working.
Thanks for your help

wimalopaan, 2021-05-03:
Maybe this is related to an old firmware version on the GR32.

I will check that end of the week, since I'm on travel actually.

pascallanger, 2021-05-03:
@G3Dmaker Can you try the release 1.3.1.92 where the 16 channels have been introduced? (https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.1.92)

G3Dmaker, 2021-05-03:
Yes, release 1.3.1.92 is working. All channels are OK!
It would be great if you could fix it to the latest release. :-)
Many thanks for the quick response! And thank you for your work.

pascallanger, 2021-05-03:
@G3Dmaker can you test all subsequent releases and indicate in which one it stops working?

pascallanger, 2021-05-03:
@G3Dmaker Can you also test the latest firmware but with the no_sync subprotocol?

G3Dmaker, 2021-05-03:
Hello guys, I am speechless. Suddenly it works with the latest release too. I ask myself why. 1.3.2.77 was already installed before. Was still on the SD.

Back to an old version. And then back to the current one. Did that flip the switch for the channels 13-16?

It is important that it works. You helped me. Thanks!!