pascallanger/DIY-Multiprotocol-TX-Module #448: [Q] Telemetry stream format.

Tygrys-1, 2020-10-10:
I want to decode the telemetry stream from multimodule, using Telemetry Mirror OTX function on serial port of Radiomaster TX16S. Is there any documentation about data format and sensor id's?

pascallanger, 2020-10-11:
I don't know how OpenTX presents the data on serial. Anyway all the telemetry decoding source code is in the OpenTX telemetry folder.

Tygrys-1, 2020-10-11:
Here is the stream.
[2020-10-09 22-03-16.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5361145/2020-10-09.22-03-16.zip)

For FrSky radio telemetry, the format looks like:
02 - header
09 - package length
xxyy - sensor id?
6 bytes sensor data
1 byte checksum

or 
01 header
18 package length
and something I do not know, with the text FrSky X&D16 at the end.

closed 2020-10-11 by Tygrys-1