pascallanger/DIY-Multiprotocol-TX-Module #449: Can AFHDS 2a be emulated with any other chip tha A7105/6

diego0815, 2020-10-11:
As it's quite hard getting good quality long range A7105 modules for DIY module building is it in theory possible emulating this protocol on any other HW (also new one not currently in use with the 4in1 module)? Except of course the SDR approach.
I would love being able to do most protocols on an NRF52840 or so :-)

pascallanger, 2020-10-12:
Sure you can do anything but you need to reverse the RF hardware implementation of the A7105.

closed 2020-10-12 by pascallanger

diego0815, 2020-10-12:
Hmm sounds more complicated than I hoped for, when reading about emulation of xn297 with cc or nrf sounds alot easier ;-)

pascallanger, 2020-10-12:
It's easy because you have the solution in front of your eyes...
If it was that easy to emulate a component from another there won't be 4 RF component...

diego0815, 2020-10-12:
Valid point, very true, thanks!