pascallanger/DIY-Multiprotocol-TX-Module #452: Eachine E016H with different Protokoll

hpoorthuis, 2020-10-11:
Before write the Ticket i did research about the E016H
found at "goebish" Eachine E016H support #46 witch seems to be base for the EO16H Protokoll.
[https://github.com/goebish/nrf24_multipro/issues/46](url) (chip The chip: XNS1042CV)
today i open my Transmitter and found :HW217BP4SD so eachine did change the Protokoll.
any idea to find the right Protokol.
[image]

pascallanger, 2020-10-28:
No... Unless you have a SDR to do a capture over the air we won't know if we can reverse the protocol.

pascallanger, 2020-11-06:
I'm going to close this issue since there is no way to move forward with the given information...

closed 2020-11-06 by pascallanger