pascallanger/DIY-Multiprotocol-TX-Module #454: Problems w/ Spektrum BNF MRs & Helos

fdmusn, 2020-10-18:
Horizon Hobby Inductrix, Inductrix FPV, mCPX BL heli, and Nano CPx heli all bind and operate perfectly w/ Spektrum DX8 transmitter. After verifying models all have same output range, servo direction, etc. in OpenTx, then rebind to x9D+ w/ external iRangeX module. Have tried multiple rebind modes (auto, DSM2, DSMX) ... same results as below.

MRs flip out of control and heli swash servos all twitch and don't respond properly.

Rebind any of them back to DX8 transmitter, and they work perfectly again.

Firmware versions:
x9D+ : opentx-x9d+-lua-en-2.3.10-otx.bin
iRangeX : multi-stm-opentx-aetr-inv-v1.3.1.69.bin

kilrah, 2020-10-18:
Wrong channel order most likely.

fdmusn, 2020-10-20:
Actually, I got it to work. Channel order in radio was AETR when it was having problems. But figured out that it is something funky w/ the bind process. I put the new information here in the event it helps developers determine if there's a bug / fix needed.

From my spektrum radio, I know that it binds at DSMX 22ms. When I built the program in companion, I set it to DSM 2F from the pulldown options in companion. It seemed to bind ok, but the servos jittered as described above. However, when I continued troubleshooting the problem, I tried rebinding with radio set to DSM and "Auto." It bound again, and the auto changed automatically to DSM 2F. But suddenly it worked fine. I tried the same method with my other helo, and it operated fine when bound using this same method.

So if the protocol is set to DSM 2F and then bound, it caused problems. If set to DSM and "Auto", and rebound, even if it changes to the exact same setting, it will now work.

Not sure if this helps the programmers or not, but that's what I found.

closed 2020-10-26 by pascallanger