pascallanger/DIY-Multiprotocol-TX-Module #455: No Sync during update atmel mega328p from X9D+

stanislavdavid, 2020-10-20:
I have problem with no sync error during update, then I made fresh update of the module over flash multi with this result:

This is the log from writing firmware
Writing MULTI-Module via USBasp
[1/5] Erasing flash ... done
[2/5] Setting fuses ... done
[3/5] Writing bootloader ... done
[4/5] Writing firmware ... done
[5/5] Verifying flash ... done

MULTI-Module firmware written successfully.

Reading MULTI-Module via USBasp
[1/2] Reading flash ... done
[2/2] Reading EEPROM ... done

Multi Firmware Version: 1.3.1.69 (AVR)
Expected Channel Order: AETR
Multi Telemetry Type: OpenTX
Invert Telemetry Enabled: True
Flash from Radio Enabled: True
Bootloader Enabled: True
USB Serial Support: False
Serial Debug Enabled: False

EEPROM Global ID: 0x96B28DBB

MULTI-Module read successfully

Then I try to upload different bin file from radio and it show me No Sync error
I have latest versions of all. I tried even latest github source of multiprotocol with same result.
Multimodule is from banggood atmel mega328p YRFM02_V1.2
Do you have any idea, why?

benlye, 2020-10-20:
What radio are you using and what version of OpenTX is it running?

stanislavdavid, 2020-10-20:
I am using Frsky X9D+ with 2.3.10

benlye, 2020-10-20:
Interesting that it's an X9D as I saw at a report in RC Groups of a similar problem with (I think) the same radio.

I just tested on my TX16S with OpenTX 2.3.10 and I was able to flash my AVR module without an issue.

It seems like it might be a problem with that specific radio model, and unfortunately I don't have one to test with.

benlye, 2020-10-20:
Found the RCG post - that was for an X9D.
https://www.rcgroups.com/forums/showpost.php?p=45599793&postcount=20087

pascallanger, 2020-10-26:
From rcgroups:
`I have determined that a Taranis X9D flashing an external avr multiprotocol module worked at openTX 2.3.9 and fails at 2.3.10 and have opened openTX issue 8060.
https://github.com/opentx/opentx/issues/8060

Jim`

closed 2020-10-26 by pascallanger