pascallanger/DIY-Multiprotocol-TX-Module #456: How do you add a protocol?

picelgon404, 2020-10-25:
I recently received a drone from cobra rc toys, the sky night. It's model number is HS177 and I was disappointed to find that it wasn't on the protocol list for the TX16S. 

I recently just received this radio (TX16S) and would like to fly this drone but upon reading through forums and watching YouTube videos I cannot find how to add or "autobind" this drone to my remote. 

Any help would be appreciated!!!

pascallanger, 2020-10-26:
Unless you are willing to send the original TX + drone to possibly reverse the protocol, there is not much that can be done.

pascallanger, 2020-11-06:
I'm going to close this issue since there is no way to move forward with the given information...

closed 2020-11-06 by pascallanger