pascallanger/DIY-Multiprotocol-TX-Module #460: Would like to work on DF Models 9918 SkyTumbler

hexagon5un, 2020-10-28:
I've got a DF Models SkyTumbler and an iRangeX4 multi-module. Opening up the included transmitter, it's an XN297LBW-based module for the TX.  I tried running through the XN297 protocols on my iRx4, and couldn't get it to bind.  

So far, I've soldered to the XN297's SPI lines: looks like CSN, CLK, and MOSI are active, and the CSN is splitting the datastream into 16-bits.  There's obvious packet structure delimited by FC 00 and FD 00, with what looks like a header and data section.  More than that, I can't tell.

What's the best way for me to contribute?  Is there a datasheet where I can start to parse the fields in the hope that I could submit a patch directly?  Or is this the kind of thing that's easiest to just post up data dumps from this thing?

Are there any particular parts of the datastream that are worth collecting?  And why does the MISO line look like it's always low?  (Did I mess up wiring?)

pascallanger, 2020-10-28:
Share your dump so I can have a look.
The xn297 can work with only 3 wires mosi and miso sharing the same pin.
You'll need to do plenty of dumps to try to figure out the protocol: one file for bind, one file per axis (one file for aileron, one file for elevator...), one file per special function...

hexagon5un, 2020-10-28:
[Here is a preliminary couple dumps](https://littlehacks.org/df_drone_xn297_1.tgz) with Sigrok.  Is that an OK format? Other preferences?

pascallanger, 2020-10-28:
From a quick look at the first dump, this is the JJRC345 protocol: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#JJRC345---71
Bind address is the same, bind packet structure is the same and more.
Let me know if this protocol doesn't work for you. If it's not then it might be a small variation...
Pascal

hexagon5un, 2020-10-28:
I just tried it, and it doesn't work.

pascallanger, 2020-10-28:
You are running the latest firmware version, correct?
Do you get any difference on the drone LEDs pattern when you do a bind?
I need more dumps as requested before. For the different channels, don't do separated dumps for up/down or left/right but instead in the same file for a same channel.

hexagon5un, 2020-10-28:
OpenTX 2.3.10 and Multiprotocol-1.3.1.69
It's not even binding.  (This is one of those drones that wants you to do an up-down on the throttle to bind the TX.  I wonder if it's doing that wrong?)
I'll work on the dumps for you.  It has AETR + speed, flip, LED, and headless.

hexagon5un, 2020-10-28:
https://littlehacks.org/df_drone_xn297_2.tgz

hexagon5un, 2020-10-28:
Speed toggles between three states.  I'm not sure I got one of each in the dump. 
I'm also not sure whether the flip logic resides in TX or the drone.  If you need me to do that again, just let me know.

hexagon5un, 2020-10-28:
https://littlehacks.org/df_drone_binding.tgz

pascallanger, 2020-10-29:
This is really the same protocol as JJRC345, the only difference seems to be the bind channel.
On this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/5ed61f30e12a9bf91055ad0ebfad38151538f42a/Multiprotocol/JJRC345_nrf24l01.ino#L26
Change 5 by 40, compile and test.

hexagon5un, 2020-10-29:
Fantastic.  Will do.  

In the meantime, I've gotten my sigrok-cli-fu down, and viewing the SPI data live is fairly informative.  I definitely know which fields are whicn now.  Fun, fun.

hexagon5un, 2020-10-29:
Yup.  That did it-- it binds now.  I'll double-check all the functions I can figure out, and then it's a done deal!  Marvelous.

pascallanger, 2020-10-29:
One you've confirmed everything is ok, I'll create a sub protocol for your drone and do a release. How should I call it? SkyTumbl?

hexagon5un, 2020-10-29:
Sounds good to me!  

DF Models has a number of quads -- my guess is that they're all rebadgings of common models, or at least custom variations.  So it's probably a good idea to use the model name like that.  

SkyTmblr?  

I'm charging one up now for testing.

hexagon5un, 2020-10-29:
I played around with the drones for a while.  The basic features seem all there, but there are still some differences with the stock JJRC code.  

There is no RTH at all.  It's a cheap drone.  

packet[10], which is aileron trim, has an enable/disable LED button on it, which sets/clears the 0x40 that is on by default in the current code.  This could map to channel 7, taking the place of RTH on the Module's TX?

packet[9] has two of these flags, on 0x40 and 0x80, that are set momentarily by pushbuttons on the remote. I can't tell what they do, if anything.  But the transmitter sends them.

The speed mode setting goes out as the values 0/1/2 over packet[12] as in the code, but it also limits the throws of the aileron/elevator controls as they're sent out.  I can't tell yet if this works like the "flip" button, which just changes the values that the TX sends out over normal channels, or if the 0/1/2 in packet[12] actually have some effect on the drone directly.  The current version hardcodes it as 2, which seems to work just fine.

If you think it's worth re-mapping RTH / channel 7 to the extra 0x40 in packet[10] for the LEDs, that might be fun.  The rest one can certainly live without.  I might keep playing around with it on my own, of course.  But for now, this should satisfy anyone who just wants to fly one of these drones.  

Would you rather do it yourself, or should I type this up and submit a pull request?  I see that sub_protocol needs to be changed/defined/documented in a few places. Maybe it's safer if you did it.  :)

And if you'd like one, BTW, I bought a whole 10 of them on eBay for €9 plus shipping.  I'll send you one just for fun if you're not buried in el cheapo drones!  :)

pascallanger, 2020-10-30:
Can you test the latest version before I do a release?
I've added LED as a separated channel since it might also work on the JJRC345: 
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#JJRC345---71
You'll see that I've added your unknown flags as well just in case there is a new model outhere.

hexagon5un, 2020-10-30:
Great.  Will test tonight and let you know.

hexagon5un, 2020-10-30:
Looks great!  I can confirm that headless and LED work, and that the unknown buttons do the same thing that they do on the controller itself so far -- blink the LEDs once or twice, respectively. 

Thanks a ton for your help with this!

(Oh, and BTW: I ended up writing up a bit about this experience on Hackaday.  Hope you don't mind.  It'll come out tomorrow / Saturday Oct 31.  https://hackaday.com/?p=444626)

pascallanger, 2020-10-31:
The hackaday link does not work. FYI the protocol was already reversed with checksum, hopping, emulation and all that was already available. This is why it might appear really easy to you but it's not the case when starting from a blank sheet.

hexagon5un, 2020-10-31:
Yeah, totally!  I _do_ appreciate that it was a piece of cake b/c it's just a minor variant of another protocol that was already done. Doubly so b/c you recognized exactly which protocol it was a variant of.    

That was kinda the point of the writeup actually -- all that work that I didn't have to do b/c other folks have done the hard work.  https://hackaday.com/2020/10/31/scratching-that-itch should also work.

closed 2020-11-01 by pascallanger