pascallanger/DIY-Multiprotocol-TX-Module #462: Question about Frequency Tuning

HazzaHFPV, 2020-10-30:
Hi everyone,
I am well familiar with the procedure of frequency tuning, but I do have a question as I am getting inconsistent results.
I tuned several receivers running different FrSky protocols, each with different results, even on the same protocol.
To achieve good enough accuracy I bound the receivers, tuned the frequency, repeated the binding and tuned a second time.
Here are the 2 results per receiver:

Rx No. 1 FrSky XM+ with FrSkyX D16 LBT, tuning = +41, +43
Rx No. 2 FrSky XM+ with FrSkyX D16 LBT, tuning = -9, -11
Rx No. 3 FrSky XM+ with FrSkyX2 D16 LBT, tuning = -2, -3
Rx No. 4 SPI Receiver with FrSkyX D16, tuning = -27, -27

A spread from -27 up to +43 seems quite large in my eyes.
With each tuning I found the lowest (tune_min) and highest (tune_max) point at which the signal started to significantly drop and took the average (tune_median).
I wanted to compile the firmware with forced tuning values to make it easier in the future. Now it seems to me as if it is better to tune for each receiver and save the value for each model.
My question is: has anyone else noticed such inconsistencies in FrSky receivers or am I missing something in the tuning process?
Thanks for your help!

pascallanger, 2020-10-30:
Might be why people are saying the XM+ is crap...
I'm surprised for the SPI receiver as I thought betaflight was doing auto tunning. But may be you are using something else.

schwabe, 2020-10-30:
Well if the SPI receiver was not turned off, it might have autotuned to the 0 value on poweron and then stayed there.

pascallanger, 2020-10-30:
Usually the autotune is done at bind time. This is when you know that the TX is around and sending data continously to adjust the freq. But that could also be implemented differently...

HazzaHFPV, 2020-10-30:
> Might be why people are saying the XM+ is crap...

That might be it, I guess.
I wasn't aware of Betaflight's auto-tuning of the SPI receiver. I guess that makes them less suitable for tuning the frequency of a protocol, as they introduce a second variable on the receiver side?
Nevertheless, I now tried two different SPIs, one in an CarzybeeF3 Pro and in a CrazybeeF4 Pro v3.0 Flight Controller.
Both came out at around -27 without a lot of spread.
As of the XM+, each receiver by itself seems consistent, but with variation among them.

I will borrow some more FrSky receivers from a friend who uses R-XSRs and X4Rs and have a look what values those will
give me.

I also had a look at the HoTT protocol while I was at it. Tested three different receivers and they all
ended up at the same value: +26. +26, +25.5. Very consistent 👍 

Thank you for your input guys, much appreciated.

closed 2020-11-10 by HazzaHFPV