pascallanger/DIY-Multiprotocol-TX-Module #464: Cloning feature with several ID's

Mucki, 2020-11-05:
Hi together,
it would be nice if the FrSky cloning feature wouldnt be limited to one ID.
If it is possible it would be nice if you could use several Ids in dependance of the selected model in OpenTX.

That way you could clone models from different radios.

Best regards.

roscor61, 2020-11-21:
Hi Pascallanger,
I have the same request for different "D16 cloned", example five ID if is possible or five ID for any combination of D16, D8  cloned. 
I think that is very impressive feature to exchange model on field and for training sessions.
Thanks

Best Regards 
Rosario

pascallanger, 2020-12-15:
Will it's possible to do this on Multi, I won't know how to select the IDs on the OpenTX side.