pascallanger/DIY-Multiprotocol-TX-Module #468: No telemetry after 1.3.1.59

jchris001, 2020-11-12:
I don't understand why I have the message in my intern module, if I try to upload it with firmware above 1.3.1.59. I don't have the trouble with my external module. This one work with 1.3.1.77.

Thanks

benlye, 2020-11-12:
Which radio? What firmware is on the radio? Exactly what firmware file have you flashed to the internal module?

pascallanger, 2020-11-12:
Most likely related to: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/393

benlye, 2020-11-12:
> Most likely related to: #393

I was wondering that too, but curious which radio might have a C8 MCU in the internal module.

pascallanger, 2020-11-12:
> I was wondering that too, but curious which radio might have a C8 MCU in the internal module.

I agree

jchris001, 2020-11-13:
Sorry, it"s a T16 Hall with internal module upgraded since 3 months....

jchris001, 2020-11-13:
And I have no error with FTDI method or radio upgrade with firmware after 1.3.1.59. The firmware is on internal module, just it doesn't work ..I think It's not a problem with memory...

pascallanger, 2020-11-13:
Can you make a close and detailed picture of the STM32?
If you have a FTDI connection, can you connect it and load a debug firmware so we can see the module debug output?

jchris001, 2020-11-13:
[image]

I think it's a 64 kb !!! that's the problem. I bought it on jumper store !!!

benlye, 2020-11-13:
Yep, that's the 64KB part.  First time we've seen that on a T16 internal module.  I'd suggest opening a support case with Jumper and trying to return it.  

Otherwise you options:
* Stay on v1.3.1.53
* Compile your own FW for the 64KB board using the Arduino IDE - you'll have to limit the protocols to the ones you need in order to get it to fit.

I wonder how many more of these are out there?

benlye, 2020-11-14:
Seems like we figured out the root cause of this - it's the incorrect MCU.  You should let Jumoer know so that they know they have a manufacturing problem (at least [one other person](https://www.rcgroups.com/forums/showthread.php?3301299-Jumper-T16-Owner-s-Thread-%282-4G-16CH-OpenTX-Multi-protocol-Radio%29/page338#post45875687) has the same issue with a T16 internal module).

closed 2020-11-14 by benlye

jchris001, 2020-11-14:
It"s done...

3djc, 2020-12-03:
@jchris001 could you contact Jumper at support@jumper-rc.com mentionning the wrong chip on module ? They are investigating and need more details