pascallanger/DIY-Multiprotocol-TX-Module #469: Eachine E016H, unbinding

little-Beppo, 2020-11-13:
Hello, I'm new here.
I've already dealt with the topic of multi-protocols a little and now my question.
I bought the current E016H, but I cannot bind it with the DIY-Multiprotocol V1.3.1.69. I found the chip "FC298TX GS50XFN411" in the transmitter.
Is this the wrong version or the wrong protocol?
Please help me .
Thank you

little-Beppo, 2020-11-13:
[image]
that is the picture from the transmitter chip.
let me know if you need more information's.

benlye, 2020-11-13:
Are you trying the correct protocol?  I have an E016H and it works fine with the E01X protocol, sub-protocol E016H.

little-Beppo, 2020-11-14:
Yes I think so.
before I wrote the treat here, I asked goebish. He mean:
"There's not much I can do.
Your best bet is asking on the multi protocol thread for some guidance on dumping xn297 packets, but this is a unknown chip, no idea if it's xn297 compatible or not."
That's what I'm trying to do here now

little-Beppo, 2020-11-14:
are there instructions for reading out the log?

pascallanger, 2020-11-14:
Load a debug firmware, get a serial connection to the module and run the XN297Dump protocol in auto mode: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_XN297Ldump.md

benlye, 2020-11-14:
I opened mine up for comparison and, yep, it's a slightly different board with a different IC.

Here's the original, for reference...

[image]

[image]

little-Beppo, 2020-11-14:
Hello, 

First of all I want to say thank you for your support.
The next thing is, my English is not as good as it should be, so the request again

I'm not sure if I got that right.
should I load another FW into the 4-in-1 module? In order to generate a dump file with it?
Is it possible to use the USB interface on the back of the module to load it into the module with the Arduino? I've already updated the boot loader. Or is it possible with the falsh-multitool, cause i get this DFU mode not working with the Arduino.
[image]

benlye, 2020-11-14:
Use Flash Multi to flash this firmware file to your module:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/download/v1.3.1.77/multi-stm-xn297dump-usbdebug-v1.3.1.77.bin

You then need to use the Serial Monitor in Flash Multi to see the debug output from the module.

After you flash the module:
# Flash the module using Flash Multi
# Unplug the module from the computer
# Turn the radio on and configure a model using the XN297 Dump protocol (protocol number 63)
# Plug the module into the computer
# Open the Serial Monitor window and look at the output

little-Beppo, 2020-11-14:
Thanks for your support, today I'm busy. I'll try it tomorrow.

little-Beppo, 2020-11-15:
Hello benlye,
i am sorry to say, i is not working correct, so i was flashing the dump FW but i can't get the tool in the Monitor Mode.
[image]
The red LED is flashing fast the whole time.
Whats my mistake?

pascallanger, 2020-11-15:
The module is in bootloader not in normal mode... Unplug the USB cable, power on the radio wait that the module is powered then plug the USB cable. At that stage a com port should appear and then open the monitor.

little-Beppo, 2020-11-15:
i think there is the USB-Driver missing.

pascallanger, 2020-11-15:
The fact that the module fast blinks indicate that it is in bootloader because it has been powered with USB plugged in first.

little-Beppo, 2020-11-15:
ok, now its working.
should I now turn on the drone and read out?

little-Beppo, 2020-11-15:
so here is the result.
I'm not sure what the log file should tell me?
[Dump_.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5542159/Dump_.log)
[image]

pascallanger, 2020-11-15:
Power on the E016H TX, exit from bind (usually throttle up down), leave the TX untouch.
On the radio select the XN297Dump protocol, sub_proto Auto and watch the screen if there is any packets being received.

pascallanger, 2020-11-15:
> so here is the result.
> I'm not sure what the log file should tell me?

This does show that no packet has been received. Was the TX powered on out of bind mode?

little-Beppo, 2020-11-15:
no i'm so poor, i powered the copter.

little-Beppo, 2020-11-15:
now i switched on the remote control of the copter and activated it in normal mode.
Then the dump started, but it looks like no connection can be made here either.
[Dump_tx.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5542187/Dump_tx.log)
[image]

little-Beppo, 2020-11-15:
is it possible that there is a problem with the 9XD that xjt is flashed as an EU version?

pascallanger, 2020-11-15:
> is it possible that there is a problem with the 9XD that xjt is flashed as an EU version?

This comment makes no sense

pascallanger, 2020-11-15:
> now i switched on the remote control of the copter and activated it in normal mode.
> Then the dump started, but it looks like no connection can be made here either.

Probably not XN297 compatible chip. Need a SDR to investigate further.
If you want you can send me the TX + quad for further investigation otherwise it's a dead end.

little-Beppo, 2020-11-15:
Send only the TX or also the copter? so i need your address.

little-Beppo, 2020-11-15:
is it possible that there is a problem with the 9XD that xjt is flashed as an EU version?

"This comment makes no sense"

    that's probably true, was just one thought

little-Beppo, 2020-11-15:
yes i can send you the TX + quad, let me know your address.

pascallanger, 2020-11-15:
> yes i can send you the TX + quad, let me know your address.

Contact me on the emain address used for the donate button or by MP on rcgroups.

pascallanger, 2020-12-15:
I've finally received an E016H.
Here is the phase of the RF signal captured using SDR for a bind packet:
[image]

This is not a XN297 core, neither a NRF from the packet layout.
The RF component is doing GFSK @250Kbps. The signal is not clean which is a bit of a worry or at least it's not fully what I'm used to see with GFSK...
The good news is that scramble is not enabled which gives some hope to reverse this protocol.
So, it sends a bunch of 55 or AA as a preamble followed by some data.
I've decoded the signal manually since my normal script has a hard time to figure out the bits (ref comment above):
E7 | E7 | E7 | E7 | 5C | A0 | 81 | C9 | B0 | 4C | 4C | 4C | 4C | 0 | 0 | 0 | 64 | 69
ID | ID | ID | ID | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | CRC | CRC
First, there is a catch since it's missing 1 bit at the end for the CRC but that could be my human mistake reading the 0s and 1s. It could also be that there is a control field which is usually the payload length, pid and ack/no ack flag which is making the payload usually shifted by a couple of bits.
Second I don't know what the convention is, so the bytes could be inverted (E7 becomes 18) and also swapped since I've used MSB first (A0 becomes 05).
The first 4 bytes are interesting since they could be the 4 bytes of the bind address E7 E7 E7 E7 (note E7 is the same if you read it with MSB or LSB).
Using these 4 bytes (or 18 18 18 18) I should be able to get the NRF to listen to the message and get live readings.
More to come...

pascallanger, 2020-12-16:
The NRF has also a hard time to receive the signal properly, there are loads of errors in the dump...
Here is one line which I think is accurate (I might be biaised by the fact that this is what I've deoded previously ;-) ):
5C A0 81 C9 B0 4C 4C 4C 4C 00 00 00 64 D2 <- the CRC is 64 D2 where I was reading 64 69 which is strange since I've verified and can't find the same result but anyway...
Like I was thinking there is a control field with a pid which counts on 2 bits 0,1,2,3,0,1... :
50,54,58,5C: xxxxyyzz where yy is the pid. Also the pid is not contained in the CRC since the CRC is constant...

Going from there I've tried to apply on the payload the standard CRC16_CCIT_ZERO which has a polynom of 0x1021 and bingo!!!
The CRC of 0xA0 0x81 0xC9 0xB0 0x4C 0x4C 0x4C 0x4C 0x00 0x00 0x00 gives 0x64D2 !!!!

More to come...

pascallanger, 2020-12-16:
A wild guess would be that the address used in normal mode is going to be: A0 81 C9 B0.
4C could be the RF channel if only one is used...
But this is a tomorrow thing as it is 1:30am currently so time to go to bed.

little-Beppo, 2020-12-16:
Thanks Pascal for the news.
Do you think that a solution with the DIY Multiprotocol is possible?

pascallanger, 2020-12-17:
Not sure yet, it will depend how well the quad is going to receive the data from the module. I'm going to use the cc2500 to emulate it so we can always try to tune the signal a little.

I also would like to see what your remote is sending. I'll post later a code on GitHub that you can launch the same way you've done the previous dump attempts.

pascallanger, 2020-12-17:
The bytes in the payload seem to have their bits reversed and shifted by 1 bit based on the stick values.
The address for bind and normal packets is the same E7 E7 E7 E7. 

The bind packet is 11 bytes long sent on RF channel 5
0A | 02 | 27 | 1B | 64 | 64 | 64 | 64 | 00 | 00 | 00
--  | --  | -- |  -- |    -- | -- |   -- | -- |   -- | -- | -- 
ID?  | Bind?  | ID |  ID | T | R | E | A | flag1 | flag2 | flag3 

The normal packet is 11 bytes long sent on RF channel 44
0A | 20 | 27 | 1B | 64 | 64 | 64 | 64 | 00 | 00 | 00
--  | --  | -- |  -- |    -- | -- |   -- | -- |   -- | -- | -- 
ID?  | Normal?  | ID |  ID | T | R | E | A | Flag1 | Flag2 | Flag3 

T, R, E and A are 0x00..0x64..0xC8.
Flag1: 0x01=Flip, 0x02=Headless, 0x04=One Key Return
Flag2: Speed control 0x00:low, 0x01:medium, 0x02:high
Flag3: 0x01=TakeOff/Land (momentary switch), 0x04=Emergeny Stop (momentary switch)

pascallanger, 2020-12-17:
I've been able to fly the E016H with my TX16S ;-)

goebish, 2020-12-17:
Well played !

pascallanger, 2020-12-17:
Multi v1.3.1.87 has support for the E016H v2. You manually need to build from source to test it out.
It is **important** to tune the frequency since mine doesn't even bind at 0. On my module and this quad I need to set the RF freq to +80.
**Only 1 ID is available** so you can't have multiple quads flying at the same time (for now?).
The protocol details are located here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#E016H---80
Interestingly, this quad does not require a centered throttle. You can power it on with throttle at -100%. When you move throttle up the motor will arm. If you keep pushing it will lift off. To land simply lower throttle and it will auto shut off when touching the ground.

pascallanger, 2020-12-17:
@little-Beppo
Using the latest version on GitHub, can you do a dump of your remote?
Notes:
- the quad does not have to be on
- use the XN297Dump protocol as you've done some time ago but use the sub protocol NRF
- power on the radio, you should see one line appearing "P: XX XX..."
- put the radio in normal mode (throttle up/down) and use the Spectrum analyzer under System to find out which frequency the TX is using

little-Beppo, 2020-12-17:
Hello Pascal,
thank you for your great work, will test it today or tomorrow.

pascallanger, 2020-12-17:
@little-Beppo please consider the above. This is to add more than 1 ID...

little-Beppo, 2020-12-17:
Ok, i will do the dump now, is the version v 1.3.1.77 ok?

pascallanger, 2020-12-17:
> Ok, i will do the dump now, is the version v 1.3.1.77 ok?

No you need 1.3.1.87

little-Beppo, 2020-12-17:
where can i get this version?

pascallanger, 2020-12-17:
Here on Master but you need to compile it yourself...

little-Beppo, 2020-12-17:
ok i have no idea how to do that. Can you help me with that?

pascallanger, 2020-12-17:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md

little-Beppo, 2020-12-17:
I want to try, but it will take a while.
I found the ino for the dump at master but can't download it, sorry haven't been around that long and don't understand the processes here so well.

little-Beppo, 2020-12-17:
an error appears when compiling, am I doing something wrong?
[image]

pascallanger, 2020-12-17:
@little-Beppo try again with the latest code

little-Beppo, 2020-12-17:
sorry I'm too stupid, now the file is too big and brings an error message again.
how can I reduce the size without deleting important things.
[image]

little-Beppo, 2020-12-17:
now i get the module flashed, but i cant make a serial dump, cause there is no XN297 Dump setting for the TX and also no COM-Port for the  serial monitor?
Do you have any idea whats my mistake?
[image]

pascallanger, 2020-12-17:
> now i get the module flashed, but i cant make a serial dump, cause there is no XN297 Dump setting for the TX and also no COM-Port for the serial monitor?
> Do you have any idea whats my mistake?

As you can see on your picture, you have a standard build not a debug one: "Serial Debug Enabled: False"
On Arduino, Tools-> Debug Option -> then you have to choose the debug type you need for your module. If you have a iRangeX+ then select "Native USB debugging" otherwise you most likely want to select "Serial/FTDI debugging".
Pascal

pascallanger, 2020-12-18:
@little-Beppo I think you can forget the dump since I've kind of found the rule for the ID vs Freq:
Freq=(ID1 + ID2)%32 + 42
But there are invalid values which still needs to be found. For example, freqs 72 and 73 are not allowed, ID1=0 is not allowed.
I'm going to poke around to see if I can spot other invalid values like ID2=0 or ID1=FF or ID2=FF or...

little-Beppo, 2020-12-18:
yes had tested the E016H2v protocol, did not work,
Later today I'll try another dump, maybe it will help you to identify the gaps in the ID

pascallanger, 2020-12-18:
> yes had tested the E016H2v protocol, did not work,

The protocol works. As I said you must tune the frequency to even get a bind. Try to set "RF Freq. fine tune" to +40 or -40. On my module nothing works unless I set it to +40, +80 being the middle value.

little-Beppo, 2020-12-18:
Yes, you are right. It works, i have done my first flight with the TX setting below.
[image]

pascallanger, 2020-12-18:
Great! You still need to tune it since 40 just gets the bind working. Mine ended to +80.

Please test the new version 1.3.1.91 which integrates support for full IDs. Please test and report.

little-Beppo, 2020-12-18:
got some problems during the compiling.

Arduino: 1.8.12 (Windows 10), Board: "Multi X-in-1 STM32F103CB (128KB), Enabled, Serial/FTDI Debugging"

c:/users/dumbo/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: C:\Users\DUMBO\AppData\Local\Temp\arduino_build_798342/Multiprotocol.ino.elf section `.text' will not fit in region `rom'

c:/users/dumbo/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: region `rom' overflowed by 16344 bytes

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board Multi X-in-1 STM32F103CB (128KB).

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

do you have any idea?

kilrah, 2020-12-18:
You must not enable serial debugging for flying.

pascallanger, 2020-12-18:
Yes it means that the code does not fit... Disable debug which will free up some space.

little-Beppo, 2020-12-18:
Okay thanks

little-Beppo, 2020-12-18:
So I've now done some test work with the version 1.3.1.91:
RF Freq. fine tune:         /       Result:
       +80                          /    bind ok, bad trim in Ail and Ele, engine speed for flying ok
       +40                          /   bind ok, verybad trim in Ail and Ele, engine speed for flying ok
           0                          /    no bind possible
        -40                          /    bind ok,, engine speed to slow for flying 
        -80                          /    bind ok,, engine speed to slow for flying 

in my opinion version 1.3.1.89 was better at +40.
what did you change?

[9 more replies]

closed 2020-12-18 by pascallanger