pascallanger/DIY-Multiprotocol-TX-Module #473: AFHDS 2A loose Binding with 64 kB Board selected

jesterhead82, 2020-11-28:
Hello, 

I am using a self compiled version for a STM32F103C8 MCU (the unsupported one).

I can sucessfuly bind a Flysky IA6 RX (AFHDS 2A) using latest version (.77), using both board definitions (64 KB and 128KB, for 128 KB I reduced protocols so Im below the C8 MCU´s limit, also I disabled flash size check).

When I restart TX using 128 kB Board definition the binding will survive thr Power-Off.

When Im using the same configuration using the correct 64kB Board defition I need to rebind each time I restart the TX (only when TX gets restarted, restarting RX is not an issue for either board definition).

benlye, 2020-11-29:
@pascallanger am I right in thinking that AFHDS 2A writes something to the EEPROM for each RX bound? I'm wonder if there's a bug in the EEPROM handling on the 64KB board definition.

I don't have any AFHDS 2A RXs to test with but I'll see if I can understand what's happening.

pascallanger, 2020-11-29:
@benlye Yep it writes the RX ID to EEPROM. The comment above seems to indicate that the ID is not saved or retrieved correctly. I don't think it's only AFHDS2A related. You can bind a module in FrSky RX mode while being in debug mode. You'll see the write and read.

benlye, 2020-11-29:
I'll look into it.  When I did the 64KB board I did check that the module ID was being written properly so I just assumed that other operations were working too.  I'll try it out with an FrSky RX and see what is happening.

benlye, 2020-11-29:
I erased the module and flashed it and a new ID was generated and stored in the EEPROM:
```Multiprotocol version: 1.3.1.77
Module Flash size: 64KB
Protocol selection switch reads as 0
Generated ID from STM32 UUID
Module Id: 73739e23
Init complete
Protocol selected: 15, sub proto 0, rxnum 0, option 0
Proto=FrSky X, nbr_sub=6, Sub=D16   , Opt=2
```

On a subsequent startup the ID was read: 
```
Multiprotocol version: 1.3.1.77
Module Flash size: 64KB
Protocol selection switch reads as 0
Read ID from EEPROM
Module Id: 73739e23
Init complete
Protocol selected: 15, sub proto 0, rxnum 0, option 0
Proto=FrSky X, nbr_sub=6, Sub=D16   , Opt=2
```

So at least that part is working...

benlye, 2020-11-29:
64KB module as FrSky RX, binding:
```
FRSKY_RX_TUNE_START
Protocol selected: 55, sub proto 0, rxnum 6, option 0
Proto=FrSkyRX, nbr_sub=2, Sub=RX    , Opt=2
FRSKY_RX_TUNE_LOW
FRSKY_RX_TUNE_HIGH
FRSKY_RX_TUNE_HIGH
Bind complete
format=1, addr[3]=39, addr[2]=86, addr[1]=02, rx_num=06, tune=-10
ch: 05 47 89 CB 22 64 A6 E8 3F 81 C3 1A 5C 9E E0 37 79 BB 12 54 96 D8 2F 71 B3 0A 4C 8E D0 27 69 AB 02 44 86 C8 1F 61 A3 E5 3C 7E C0 17 59 9B DD
FRSKY_RX_DATA
```

Switching to another model and then back to the RX model:
```
0 pps
Bad frame RX
No input signal
Protocol selected: 15, sub proto 0, rxnum 5, option 0
Proto=FrSky X, nbr_sub=6, Sub=D16   , Opt=2
0 pps
0 pps
Bad frame RX
No input signal
format=1, addr[3]=39, addr[2]=86, addr[1]=02, rx_num=06, tune=-10
ch: 05 47 89 CB 22 64 A6 E8 3F 81 C3 1A 5C 9E E0 37 79 BB 12 54 96 D8 2F 71 B3 0A 4C 8E D0 27 69 AB 02 44 86 C8 1F 61 A3 E5 3C 7E C0 17 59 9B DD
Protocol selected: 55, sub proto 0, rxnum 6, option 0
Proto=FrSkyRX, nbr_sub=2, Sub=RX    , Opt=2
0 pps
```

Can I assume from this that the EEPROM was written and then read?

benlye, 2020-11-29:
@jesterhead82 can you put the 64KB firmware on your module, bind your RX, then use Flash Multi to read the module and save the backup (including the EEPROM), and post it here?  I can read the EEPROM data and see if it shows anything interesting.

Unfortunately we won't be able to compare it to your module running the 128KB firmware because we can only read 64KB out of your module via the USB port.

jesterhead82, 2020-11-30:
Hello,

I attached the backup.

[Backup module.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5614434/Backup.module.zip)
[Multi-Flash - writing.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5614436/Multi-Flash.-.writing.txt)
[Multi-Flash - Reading.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5614437/Multi-Flash.-.Reading.txt)

benlye, 2020-11-30:
There's something strange going on in that data.  I'm thinking that the 128KB flash has left data behind in the 64KB EEPROM section and it has not been re-initialized so just looks like garbage.

Can you use Flash Multi to erase the module (including the  EEPROM), then flash it with the 64KB firmware again, re-do the bind and see if the bind is retained now?  

If the bind is still not retained, save the backup again and re-send it and I'll have another look.

benlye, 2020-11-30:
@pascallanger I'm wondering if there is a way to init the entire EEPROM section from within the firmware if, for example, the global ID has to be generated.

Otherwise we'll just have to tell people to do a full erase of the module first if they are going to switch from firmware compiled for the 128KB board to firmware compiled for the 64KB board.  

I wouldn't want to do it automatically, but I could add a menu option in the IDE so that the user can choose to preserve or erase the EEPROM section (for both STM32 boards).

jesterhead82, 2020-11-30:
You are right, erasing both and flashing again the 64 kB board fixed it, sorry for the inconveniences.

closed 2020-11-30 by jesterhead82

benlye, 2020-11-30:
> You are right, erasing both and flashing again the 64 kB board fixed it, sorry for the inconveniences.

Glad you're sorted. We'll have to think about ways to prevent this from happening in the future if we can.

benlye, 2020-11-30:
@pascallanger I've had a look through [the docs](https://www.st.com/resource/en/application_note/dm00311483-eeprom-emulation-techniques-and-software-for-stm32-microcontrollers-stmicroelectronics.pdf) and it seems like it would be a good idea to call `EE_init` ([link](https://github.com/benlye/DIY-Multiprotocol-TX-Module-Boards/blob/bcc4d8b740269173edf65fb5a5348145cebccf3a/source/stm32/libraries/EEPROM/EEPROM.cpp#L298)) at startup to avoid / correct problems like this.

I'll play around and see if it does what I think it will.