pascallanger/DIY-Multiprotocol-TX-Module #476: XK450 Motor schaltet nach Landung nicht ab

Githubbachelor, 2020-12-05:
XK450 Motor schaltet nach Landung nicht ab. Alles ist soweit OK. Außer dass nach der Landung der Motor nicht abschaltet. Die original Anlage von dem XK450 macht das.
2. Die Bindung muss vor jeder Nutzung erneut vorgenommen werden, Feintuning wurde komplett durchgescheckt.
3. Motorausschalter CH7 geht nicht. Auch bindet oder Aktiviert er nicht die Bindung über den Throttle/Gas Knüppel, wie original Anlage macht.

Das Problem mit den Motoren auf CH4 hast du ja grandios erledigt. Vielleicht hast du eine Idee warum der Motor nicht ausgeht nach Landung. Damit er Abschaltet muss die Anlage ausgeschaltet/Horus  ausgeschaltet werden. Der XK450 läuft dann noch ca. 8 Sekunden - sollte aber nicht so sein.

Trotzdem vielen Dank Pascal, dass man das Ding schon mal fliegen kann. Vielleicht hast du eine Chance die restlichen Dinge zu erledigen, das wäre suuuupeerr!
Gruß Michael

pascallanger, 2020-12-05:
Please use English to describe your issue.
I don't have this model so I need a good description of what's happening. Without this there is no way I can help....
One thing which puzzles me is that I have 2 persons that have done test flights and have reported that everything was fine.

pascallanger, 2020-12-05:
The bind is not done through throttle like most of the Multi protocols but through the use of "Bind".
To use bind on channel, check the documentation: https://www.multi-module.org/using-the-module/binding#bind-on-channel

Emergency stop of the motors is done through CH7 as explained in the doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#xk---62

Githubbachelor, 2020-12-05:
Than you for your help pascal

closed 2020-12-05 by pascallanger