pascallanger/DIY-Multiprotocol-TX-Module #477: Multiprotocol board update with STM32F405RG

bbc25185, 2020-12-08:
Hello.
Given the shortage of flash memory now present, i wanted to know how much it might be of interest if I try to make an updated version of the board using a STM32F405RG instead of the STM32F103CB.
This would bring as much as 1Mb of flash memory and, therefore, new features.
An example is the removal of the mechanical selector in favor of an OLED display with menus and buttons.
Obviously I would try, as far as possible, to be as similar as possible both as components and as connections in order to facilitate the passage of the firmware on the new platform.

What are you saying? Do I try working on it?

pascallanger, 2020-12-15:
Hi, We've decided to go for a STM32F303RCT6. 256KB and can be upgraded later to 512KB as needed.
As you mentionned I've added an I2C OLED display with 3 keys for the navigation. But also an external flash to store information and other things.
If you are willing to participate in the development of that new board you are welcome.
Pascal

bbc25185, 2020-12-17:
I normally use EasyEda to make the schematics and PCBs of the electronic boards.
What do you use?

richardclli, 2021-02-21:
For opensource project, I think it is better to use KiCad, since it is opensource as well.

MRC3742, 2021-03-04:
> If you are willing to participate in the development of that new board you are welcome.
> Pascal

Is this being discussed or developed elsewhere?
Is it currently a DIY project or are prototypes available?

Thanks - Rich

richardclli, 2021-03-05:
@pascallanger How may I participate?  Any links or direction?

pascallanger, 2021-03-05:
I haven't started looking at it yet... I know... I'm doing quite some changes right now to the code but I'll get back to it soon.

pierrotm777, 2021-11-03:
Hello, i use Kicad for opensource development as OpenAVRc shield v2.0, v2.1 and v2.2.
If needed, i can help for the design.
Pierre

pierrotm777, 2023-10-22:
Hello Pascal,
Do you have some news about the add of a OLED screen.
Several Elrs module has this option and it's a very good solution for non optentx/edgetx transmitter.