pascallanger/DIY-Multiprotocol-TX-Module #478: Radiomaster Multiprotocol Assan Range.

Flypao, 2020-12-09:
Hello,
I use several Assan receivers but I noticed that the range compared to the original tx is at least half ... with protocols like frsky there are no problems .... why?!?
Is there a solution to have the range as the original tx Assan?!?
Thanks.

pascallanger, 2020-12-09:
I would expect this type of comment with a none genuine Nrf24l01 RF component. Normally Radiomaster uses good components so that shouldn't be the case. Can you confirm that you are using the internal module that they provide?

Flypao, 2020-12-09:
Yes, I confirm that I use the original internal module, I also updated the firmware to the latest version 1.3.1.78. After I nearly broke a plane, I did some tests and I noticed the difference in range. With his original Assan X8F Tx that I installed in a Futaba the range is double. I do not think of a failure of the radio frequency part since with a Rx FrSky X8R the range is regular!
Thanks.

pascallanger, 2020-12-10:
And again this looks like the Nrf24l01 is at fault here.

Flypao, 2020-12-10:
There is a solution?

pascallanger, 2020-12-11:
If you have access to another multi module I could write a firmware that test the nrf24l01 to see if this is the genuine part.

Flypao, 2020-12-12:
Thank you for your availability but currently I don't have another multimodule, possibly I can replace the internal one with a new one?!? This is the image of the multimodule mounted inside the radio but the model is not written on it!
[image]

pascallanger, 2020-12-15:
I'm not sure how to move forward. Replacing the internal module is an option but a better solution might be to purchase an external module (iRangeX+).

Flypao, 2020-12-15:
Hi, I talked to an American friend on facebook and he too has the same problems with the rx Assan and tx Radiomaster. I could put the external module, as you recommended, but then are you sure that I solve the range with Assan emulation?!? Do I lose the possibility of Receiver Match using the external module?!? Thanks for your help.

pascallanger, 2020-12-15:
I can't guarantee anything. I'm not the manufacturer of your module nor I'm being paid on what I'm doing.
I don't have an Assan receiver to do any testing. But I can tell that this protocol has not been touched for years.
Ask the question on rcgroups, you might have other users perspective. I know people with large gliders that have been reporting to use the Assan protocol without an issue.
An external module or internal module is the same.
Options:
 - purchase an external iRangeX+ module and test
 - send me an Assan receiver which I will be able to test on various modules and address of is related to some modules or if there is a general issue
 - throw away your RXs and go for another product not using the nrf24l01
Pascal

Flypao, 2020-12-15:
I'm sorry if I was misunderstood but I absolutely didn't want to give you any responsibility or guarantee ... it was a way to ask if you had had direct experience with the Assan receivers! Thank you very much for your willingness to answer me, mine were just questions that I shared with you! Unfortunately I have no experience with multi-protocol modules! Since, sooner or later I will have to change the receivers, for now, the simplest and cheapest solution I think is to use the original Assan module as an external module!

closed 2020-12-15 by pascallanger